lvgl/lv_binding_micropython #4: Rename this repo to lv_binding_micropython

kisvegabor, 2019-01-27:
@amirgon 
We were talking about having all binding here is not a good idea put Python and Micropython might work.

Now I see the project structure of `lv_mpy` and it definitely would be clearer is this repo would be only about micropython.

Do you agree with renaming this repo to ` lv_binding_micropython`? Would it cause trouble for you?

Renaming `lv_mpy` to `micropython_unix` would be reasonable too but at least won't break anything.

kisvegabor, 2019-01-27:
If `lv_bindings` were added to a project as submodule then adding Python (and other bindings) would be an unnecessary overhead. For example, if LittlevGL would be added to the "original" micropython repo, I'm sure they would prefer to don't have the other, unrelated bindings in it.

> different bindings may need to be aligned with specific versions of lvgl reflected in the submodule

Let's say a new lvgl version is released and you want to update the bindings for it. So you update the lvgl submodule to the new version but it might break other's bindings.
 
> Please do not rename lv_mpy to micropython_unix. I'm planning to add ESP32 support on the same repo so it will not be a unix-only example.

Awesome! Can't wait to try it!

kisvegabor, 2019-01-28:
@amirgon 
An other thing: it would be easier if you had write access to this repo. You know it the best and I assume you would open an issue if there is something to discuss. Can send an invitation?

amirgon, 2019-01-28:
@kisvegabor Sure.  Are referring to `lv_bindings`, `lv_mpy` or both?

kisvegabor, 2019-01-28:
Even both. Ok?

amirgon, 2019-01-28:
> Even both. Ok?

Sure.

> In light of this `lv_mpy` should be renamed too.

> So instead of `lv_mpy` even `micropython` can work as it's really an extension of the original micropython. (It has `lv_...` prefix now because it was in my personal account earlier)

So - `lv_micropython` and `lv_binding_micropython`?

kisvegabor, 2019-01-29:
I've sent the invitations.

For the names, I meant `lv_binding_micropython` and simply `micropython`. Now, none of the projects (where lvgl is a submodule) has `lv_...` prefix.

amirgon, 2019-01-29:
> For the names, I meant `lv_binding_micropython` and simply `micropython`. Now, none of the projects (where lvgl is a submodule) has `lv_...` prefix.

@kisvegabor I'm not sure we should have a project named `micropython`. This is ambiguity with Micropython project itself...  
Unless you think in the future this could be merged back to Micropython upstream.  
Is it an "example project" to show how to use lvgl with Micropython, or a Micropython fork for whoever wants lvgl with Micropython?

amirgon, 2019-01-29:
> I don't think it will be merged directly to the upstream. I suppose micropython would adopt lvgl and the binding as they really want to see it in their project.

More likely they would prefer a PR. But let's wait and see if there is enough interest in this, after this comes as part of an official lvgl release.

> What do you think, if things go well, the `lv_mpy` repo might even become obsolete and the upstream micropython should be used instead?

Maybe, too early to say.  
Micropython already has several forks that would probably never merge back to Micropython. Notable examples are [circuitpython](https://github.com/adafruit/circuitpython) and [MicroPython_ESP32_psRAM_LoBo](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo), so lv_mpy might become such a fork. I'm not familiar personally with Micropython maintainers (@dpgeorge and @pfalcon I think) and I'm not sure what would be their position regarding lvgl as micropython library.

kisvegabor, 2019-01-29:
Okay, we will see :)

I'll rename the repos in a minute as we discussed.

kisvegabor, 2019-01-29:
Done:
https://github.com/littlevgl/lv_micropython
https://github.com/littlevgl/lv_binding_micropython

closed 2019-02-01 by amirgon

pfalcon, 2019-02-01:
Hi guys, just a note that I got a @mention here, but so far didn't have a chance to respond (I have a habit to read up the existing discussion/content before writing responses, and so far didn't have a chance to do that). I hope to look into this over weekend, and have some suggestions to share in mind.

In the meantime, thanks for your work, it's definitely cool and appreciated!

pfalcon, 2019-02-02:
> I'm not familiar personally with Micropython maintainers (@dpgeorge and @pfalcon I think)

I should start with saying that I'm no longer a maintainer, so all I write represents my opinion, and I don't influence what can be merged into MicroPython mainline and what now.

pfalcon, 2019-02-02:
So, regarding naming: as was said, "mpy" tells nothing to outsiders. And for people familiar with MicroPython, "mpy" is the bytecode (vs source) module (effectively, MicroPython uses .mpy where CPython uses .pyc).

The usual contraction of MicroPython is "upy". Well, "uPy". (Where "u" is a pretty common way to "render" Greek "mu" letter in Latin script). That doesn't say much to outsiders either.

Summing up, I'd recommend to use "micropython" to ring a bell to the widest audience. I'd recommend against using "mpy".

IMHO, if the name "lv_binding_micropython" is long, it's because of the "binding" part. I wouldn't use it.

pfalcon, 2019-02-02:
> Micropython already has several forks that would probably never merge back to Micropython. Notable examples are circuitpython and MicroPython_ESP32_psRAM_LoBo, so lv_mpy might become such a fork.

So, you want to wrap a GUI lib and you make a fork of language, and plan for it to be a permanent fork? Well, that's pretty sad prophecy. Any fork like that can be only temporary, and may end only in 2 things:

* Upstream acquires enough machinery to let you stop forking it. (That doesn't mean it merges your stuff - more likely, it grows more modular, so you don't need to fork it to use with your stuff).
* You're getting tired on spending time on useless maintenance of your fork.

And there's exactly an effort to allow to maintain 3rd-party, external modules for MicroPython, and integrate them in the build process: https://github.com/micropython/micropython/pull/4195 .

Playing with and leveraging that would be the best suggestion I can give. littlegv would be of course a biggish project to start with that feature, but we really need people to give it a try for real projects, and it will be fully worth for this project. Any questions or suggestions, please share them via that ticket.

amirgon, 2019-02-02:
> So, you want to wrap a GUI lib and you make a fork of language, and plan for it to be a permanent fork?

@pfalcon, originally @kisvegabor and me were wondering if there would be an interest from the Micropython maintainers and community to integrate lvgl as a Micropython library in the Micropython mainline.  
From your answer it seems that you think think the answer is no, and a better approach would be to register lvgl as an external library.  
My current plans are to design `lv_micropython_bindings` in such a way that it would be relatively easy to integrate it with a Micropython fork, be it the original Micropython, CircuitPython or any other fork which is close enough.  

[`lv_micropython`](https://github.com/littlevgl/lv_micropython) is a Micropython fork built as an **example project** to show how to integrate `lv_micropython_bindings` with Micropython so that any Micropython fork user who is interested in lvgl could do it with his own Micropython fork. It was not meant to be another long-term Micropython fork which requires (as you put it) useless maintenance effort.   

So I think micropython/micropython#4195 is a good initiative. If it is merged with Micropython upstream and other forks (such as CircuitPython) it could be a way to achieve the same thing in a more standard and organized way.

pfalcon, 2019-02-03:
> originally @kisvegabor and me were wondering if there would be an interest from the Micropython maintainers and community to integrate lvgl as a Micropython library in the Micropython mainline.

lvgl is absolutely fantastic, and everyone wants to be able to use it with micropython. I myself looked at it some time ago, but it didn't go beyond a few [typo fixes](https://github.com/littlevgl/lvgl/commits?author=pfalcon) then. It's great to see you working on it. But we can't merge every possible module into MicroPython mainline. There are (or should be) thousands of them, and they can't all live in MicroPython's master repo. And if all can't live, then exceptions for single one don't make sense either - it's still unneeded maintenance burden. Instead, we should solve a problem of how let people maintain their modules on their own, while being able to integrate into MicroPython build easy (eventually, allow dynamically installable/downloadable modules). Just think about that: PyQt is absolutely great project, but doesn't and can't possibly live in CPython repository, and that's not needed. We want to achieve the same situation with MicroPython.

> It was not meant to be another long-term Micropython fork which requires (as you put it) useless maintenance effort.

Great to hear.

> So I think micropython/micropython#4195 is a good initiative. If it is merged with Micropython upstream

Right. To emphasize that point, if you just wait until it's merged, the wait may be too long. Please consider giving it try right now. If it works, post your success story - that would be great motivation to merge it ASAP. If it doesn't, share your problems/questions in that ticket, so they can be addressed (or at least known), and we actually have a progress along the vision described above.