lvgl/lv_binding_micropython #5: Port and Driver separation

amirgon, 2019-02-01:
`lv_binding_micropython` provides the script for generating a Micropython module for accessing lvgl, but does not yet handle:

- How lvgl Micropython module is registered and initialized on a Micropython Port (port in the sense of architecture such as "unix" or "esp32")
- How display and input drivers are registered to lvgl.

Currently the [`lv_micropython`](https://github.com/littlevgl/lv_micropython) repo contains an example of using `lv_binding_micropython` on the unix port. The unix  port itself registers the binding, initializes it and registers display and input drivers.
Problems are:

- It's difficult to scale `lv_micropython` to support other architectures other than unix
- It's harder to add lvgl support to other Micropython forks.
- No generic way to let the Micropython user select which driver he would like to use for display and input.

So I plan to:

- Change the directory structure of `lv_binding_micropython`: add `drivers` and `ports` directories that would contain port-specific and driver specific code
- Allow the user to register the display and input drivers in Micropython. Theoretically these drivers could be "pure python" but I don't think it's a good idea due to performance aspects.  
I would like, however, to let the user select and register a driver on the python script.

kisvegabor, 2019-02-01:
I can't say anything than all sounds great! :)

amirgon, 2019-02-05:
Display and input drivers are now registered in the Micropython script.  

- Added SDL micropython module to be used when registering SDL drivers
- Removed `lv_mp_init` function as all initialization are done on the script
- Added example of how to register drivers in  [`advanced_demo.py`](https://github.com/littlevgl/lv_binding_micropython/blob/a8e3b27bb181e34b4af1967fbc689f14c948a3ad/gen/advanced_demo.py#L91)
- Changed directory structure. Added port and drv directrories
- Updated [`lv_micropython`](https://github.com/littlevgl/lv_micropython/commit/bd7e5f16741104294d787a2e1c40d9d2217e75fa)

Closing.

closed 2019-02-05 by amirgon