lvgl/lv_binding_micropython #6: Bug: sometimes lv_init get stuck

amirgon, 2019-02-05:
Probably related to gc somehow. Is there a gc root missing?
Problem is intermittent. Same build sometimes works and sometimes doesn't.
Gets stuck in infinite loop on `lv_font.c:59`.

Callstack:

```
(gdb) bt
#0  lv_font_add (child=0x785ee0 <lv_font_dejavu_10_cyrillic>, parent=0x785ca0 <lv_font_dejavu_10_latin_sup>) at ../../lib/lv_bindings/lvgl/lv_misc/lv_font.c:59
#1  0x0000000000496d37 in lv_font_builtin_init () at ../../lib/lv_bindings/lvgl/lv_fonts/lv_font_builtin.c:55
#2  0x00000000004972b9 in lv_font_init () at ../../lib/lv_bindings/lvgl/lv_misc/lv_font.c:46
#3  0x00000000004a6331 in lv_init () at ../../lib/lv_bindings/lvgl/lv_core/lv_obj.c:83
#4  0x00000000004ce253 in mp_lv_init (n_args=0, args=0x7fffffffd7e8) at build/lvgl/lv_mpy.c:19327
#5  0x00000000004316f8 in fun_builtin_var_call (self_in=0x550960 <mp_lv_init_obj>, n_args=0, n_kw=0, args=0x7fffffffd7e8) at ../../py/objfun.c:127
#6  0x00000000004264ed in mp_call_function_n_kw (fun_in=0x550960 <mp_lv_init_obj>, n_args=0, n_kw=0, args=0x7fffffffd7e8) at ../../py/runtime.c:628
#7  0x0000000000426586 in mp_call_method_n_kw (n_args=0, n_kw=0, args=0x7fffffffd7d8) at ../../py/runtime.c:644
#8  0x000000000044a143 in mp_execute_bytecode (code_state=0x7fffffffd7b0, inject_exc=<optimized out>) at ../../py/vm.c:1012
#9  0x0000000000431970 in fun_bc_call (self_in=0x7ffff15c9ac0, n_args=1, n_kw=0, args=0x7fffffffda28) at ../../py/objfun.c:287
#10 0x00000000004264ed in mp_call_function_n_kw (fun_in=0x7ffff15c9ac0, n_args=1, n_kw=0, args=0x7fffffffda28) at ../../py/runtime.c:628
#11 0x0000000000426586 in mp_call_method_n_kw (n_args=0, n_kw=0, args=0x7fffffffda20) at ../../py/runtime.c:644
#12 0x000000000044a143 in mp_execute_bytecode (code_state=0x7fffffffd9f0, inject_exc=<optimized out>) at ../../py/vm.c:1012
#13 0x0000000000431970 in fun_bc_call (self_in=0x7ffff15c6880, n_args=0, n_kw=0, args=0x0) at ../../py/objfun.c:287
#14 0x00000000004264ed in mp_call_function_n_kw (fun_in=0x7ffff15c6880, n_args=0, n_kw=0, args=0x0) at ../../py/runtime.c:628
#15 0x0000000000426407 in mp_call_function_0 (fun=0x7ffff15c6880) at ../../py/runtime.c:602
#16 0x0000000000457cd3 in execute_from_lexer (source_kind=2, source=0x7fffffffdc00, input_kind=MP_PARSE_FILE_INPUT, is_repl=true) at main.c:144
#17 0x0000000000457f12 in do_repl () at main.c:258
#18 0x0000000000458dc1 in main_ (argc=1, argv=0x7fffffffde98) at main.c:637
#19 0x0000000000458372 in main (argc=1, argv=0x7fffffffde98) at main.c:418
```

kisvegabor, 2019-02-06:
This part of the code just creates a simple linked list from the related fonts. Printing `parent->next_page` in `lv_font_add` might help to what's going on.

Font declarations are not using `malloc` so I don't think it's related to GC.

amirgon, 2019-02-06:
It happens rarely, didn't find a way to reproduce it in a predictable way, so it's hard to debug.  
Looks like some memory overrun??
No root cause yet.

kisvegabor, 2019-02-07:
> Looks like some memory overrun??

Probably yes. I suggest leaving `printf`s in `lv_font_add` to see the values once the bug occurs.

amirgon, 2019-02-07:
Found it!
When `lv_init` is called the second time, it gets stuck in an infinite loop.  

@kisvegabor 
I understand that `lv_init` should be called only once.
But if someone calls it twice, by mistake, don't you think there should be a more "sane" behavior such as either doing nothing or reporting some error?  
When working with REPL on Micropython this could happen more, as someone types commands interactively.

closed 2019-02-07 by amirgon

kisvegabor, 2019-02-08:
Awesome! :+1: