lvgl/lv_binding_micropython #7: Add ESP32 port + drivers

amirgon, 2019-02-05:
Add ESP32 port + ILI9341 driver + Resistive Touch driver, and an example on [`lv_micropython`](https://github.com/littlevgl/lv_micropython) of integrating it all into a Micropython project.

amirgon, 2019-02-13:
### Update
Added support for ESP32 ILI9341.
Also updated [`lv_micropython`](https://github.com/littlevgl/lv_micropython) project to build and register it.  
Added usage examples and build command line to [`README.md`](https://github.com/littlevgl/lv_micropython/blob/master/README.md).  

Next - resistive touch driver.

kisvegabor, 2019-02-15:
I tried to test the nex version on ESP32. It compiled well using the instructions from the Readme. However, I didn't fond how to flash the compiled binary to the ESP32.

Can you help with that?

amirgon, 2019-02-15:
> I tried to test the new version on ESP32. It compiled well using the instructions from the Readme. However, I didn't find how to flash the compiled binary to the ESP32.
> 
> Can you help with that?

Yes.
I build and flash it with:

```
make -C ports/esp32/ ESPIDF=~/esp/projects/lv_mpy_deps/esp-idf -j4 LV_CFLAGS="-DLV_HOR_RES=240 -DLV_VER_RES=320 -DLV_COLOR_DEPTH=16" deploy
```

The `deploy` target at the end uses esptool to flash it through USB.  
You need esptool installed (I think it comes with the standard esp-idf installation, or you can install it separately).
You will also need to have user permissions to access the `/dev/ttyUSB0` device. On Linux, you can enable this by adding your user to the dialout group, and rebooting or logging out and in again.

In general, just follow the instructions in [ESP32 README](https://github.com/micropython/micropython/blob/master/ports/esp32/README.md).

kisvegabor, 2019-02-15:
I see, thank you!

Somehow I couldn't compile it again. But when I changed the IDF version accorfing to the build log it compiled. 

```
> make -C ports/esp32/ ESPIDF=~/esp/esp-idf LV_CFLAGS="-DLV_HOR_RES=320 -DLV_VER_RES=240 -DLV_COLOR_DEPTH=16" 

Current git hash:   44b97b7fe7e0f2612de467541b92e93abece5c87
Supported git hash: 30545f4cccec7460634b656d278782dd7151098e
```

I registered the display as you described in the README and it  was working well! 
**Congratulation!** :clap: :clap: :clap: 

[image]

amirgon, 2019-02-15:
> I registered the display as you described in the README and it was working well!
> **Congratulation!**

Great!! :smile: Thanks for trying it out.

amirgon, 2019-02-18:
Added resistive touch driver.
Updated `advanced_demo.py` example.

Closing.

closed 2019-02-18 by amirgon