lvgl/lv_binding_micropython #10: Add generic input device driver

amirgon, 2019-02-25:
Currently, supported input device drivers are:

- SDL mouse
- Raw resistive touch

I would like to add a new **generic** input device, which can be implemented in Micropython.  
This means that the user will be responsible of updating the input data from his Python script.  

(related: https://forum.micropython.org/viewtopic.php?p=34334#p34334)

kisvegabor, 2019-02-25:
Sounds great and very easy to use!

amirgon, 2019-03-14:
Added a generic input device.
See [lvindev_example.py](https://github.com/littlevgl/lv_binding_micropython/blob/master/driver/generic/lvindev_example.py) for an example of how to use it.  

On v6.0 it will become obsolete once generic callbacks are supported, since a generic callback could be used instead of the ad-hoc indev `read` callback in this generic input device driver.

Closing.

closed 2019-03-14 by amirgon