lvgl/lv_binding_micropython #11: Build error for STM32 port due to `free()` macro

BrendanSimon, 2019-02-25:
There is a build error when trying to build for an STM32 port (`BUILD=OLIMEX_E407`).

```console
$ make -C ports/stm32/ BOARD=OLIMEX_E407
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
GEN build-OLIMEX_E407/genhdr/qstr.i.last
../../lib/lv_bindings/lvgl/lv_misc/lv_fs.c:452:46: error: macro "free" passed 2 arguments, but takes just 1
         res = drv->free(&total_tmp, &free_tmp);
                                              ^
^Cmake: *** Deleting file `build-OLIMEX_E407/genhdr/qstr.i.last'
make: *** [build-OLIMEX_E407/genhdr/qstr.i.last] Interrupt: 2
```

I think the issues is due to the `free()` macro defined in `mpcconfigport.h`

```console
$ find . -name "*.h" | xargs grep --color "define *free("
./ports/stm32/mpconfigport.h:#define free(p) m_free(p)

```

amirgon, 2019-02-25:
Hi @BrendanSimon,
The `free` macro is not used directly in `lv_fs.c`.
Could you try undefine it on `lv_fs.c`?
Something like:

```c
#ifdef free
#undef free
#endif
```

Another option, set `USE_LV_FILESYSTEM` to 0 on `lv_conf.h` if you are not using it.

kisvegabor, 2019-02-25:
I think the compiler somehow thinks `drv->free()` refers to `free()`.

@amirgon 
I'm not sure if it helps but in `dev-6.0` I renamed `drv->free` to `drv->free_space`.

embeddedt, 2019-02-25:
@kisvegabor It's because in `mpconfigport.h` there is a global define:

```c
#define free(p) m_free(p)
```

The preprocessor will check for any use of `free()`, even when prefixed by `drv->`. This causes the error.

kisvegabor, 2019-02-25:
@embeddedt 
It seems reasonable. Then my fix should solve it. Unfortunately, it can't go to `master` registering driver registration might use `drv->free`. Anyway, it's a good lesson to avoid these typical names.

amirgon, 2019-02-25:
> It seems reasonable. Then my fix should solve it. Unfortunately, it can't go to `master` registering driver registration might use `drv->free`.

@kisvegabor I think that adding `#undef free` on `lv_fs.c` could solve this, as I suggested above.  
If @BrendanSimon confirms it solves the problem, would it be reasonable to commit this to lvgl `master`?

embeddedt, 2019-02-25:
@amirgon No. I respectfully disagree. I don't think we should be changing global macros like that. Two wrongs don't make a right.

It would be better to raise an issue on the Micropython side about how this affects external modules.

kisvegabor, 2019-02-25:
I agree with @embeddedt, it'd be hacky.

What about replacing 
```c
#define free(p) m_free(p)
```

with

```c
static inline void free(void *p) 
{
   m_free(p);
}
```

It seems to be equivalent from Micropython's point of view if `m_free` is visible in `mpconfigport.h`.

embeddedt, 2019-02-25:
@kisvegabor When including `stdlib.h` the compiler might complain that the prototype no longer matches, since `free` is now `static`.

amirgon, 2019-02-25:
> @amirgon No. I respectfully disagree. I don't think we should be changing global macros like that. Two wrongs don't make a right.
> 
> It would be better to raise an issue on the Micropython side about how this affects external modules.

@kisvegabor @embeddedt I see your point.
Although the change I suggested is local to `lv_fs.c` and will not affect anything else, I agree it's hacky.  

However relying on Micropython PR for solving this has its own problems:

- It sometime takes lots of time for Micropython PRs to be accepted. They don't tend to accept changes that are meant to solve very specific problem in a very specific library (but who knows?)
- Even if it is accepted, it will take time for this fix to propagate to other forks. Take CircuitPython for example, they only take major Micropython releases when they come out.

So I wonder if there could be some other acceptable solution which does not require Micropython PR.

amirgon, 2019-02-27:
Here is [Micropython's maintainer suggestion](https://github.com/micropython/micropython/issues/4559#issuecomment-467657336) for this problem:

> In the meantime, a workaround is to #undef the macros before they are used in different contexts.

@kisvegabor @embeddedt, if you don't want to do this, as you mentioned above, then I suggest fixing it in the meanwhile on `lv_micropython` repository, so that at least those who use `lv_micropython` directly (instead of using their port with `lv_binding_micropython`) will not have this problem.

kisvegabor, 2019-02-27:
@amirgon 

As @embeddedt has pointed out the use of `static inline` can cause issues if `stdlib` is also included.

So it seems to me the only working method is `undef`. In light of that,  IMO we can use `undef` to solve it. @embeddedt, in lack of a better solution, do you agree with that?

Although the root of the issue is in Micropython, LittlevGL should be more robust against these issues. So if you find other common "names" in lvgl please report.

BrendanSimon, 2019-02-27:
Use of common names like `free` is fraught with danger.  It should be resolved at the Micropython end, however I agree that is also a pragmatic solution to change the name in the LittleGL code to be able to solve the problem (in case Micropython don't want to change it for one reason or another).

kisvegabor, 2019-02-27:
@BrendanSimon The problem is we can't change this name in LittlevGL `master` because it wouldn't be backward compatible. (I already changed in `dev-6.0`)

embeddedt, 2019-02-27:
@kisvegabor @amirgon @BrendanSimon Can we wrap the `#undef free` in an `#ifdef` that checks to see if we are in the Micropython context? That would be safer.

amirgon, 2019-02-27:
> Can we wrap the `#undef free` in an `#ifdef` that checks to see if we are in the Micropython context? That would be safer.

@embeddedt We can, but I don't see how it makes anything safer.  
Let's say that some other non-micropython header file defines `free`. It would cause the same problem! So why limit the solution to micropython?

embeddedt, 2019-02-27:
@amirgon I'd like to avoid changing the global scope as much as possible. If the fix can't go on the Micropython side, then we should put the hack in LittlevGL but restrict it to only apply to Micropython. I don't think we should try to solve other `free` problems that aren't known to exist yet.

kisvegabor, 2019-02-28:
@amirgon Is there a global `define` which we can use?

amirgon, 2019-02-28:
> Is there a global `define` which we can use?

@kisvegabor There probably is, but first, I still don't understand the need for it.  
I would be happy if you could explain this **by an example**.   

So - suppose we don't wrap the `#undef free` with `#ifdef` to check if we are in micropython context, and we put the `#undef free` on `lv_fs.c`. What bad thing could happen? Could you give a theoretical example of how this would have a bad effect, when not in micropython context?

kisvegabor, 2019-03-01:
@amirgon 
I don't expect any side effect from simply `#undef free` but maybe @embeddedt can tell any. I think the reason behind is to reduce the scope of the hack to minimize the probability of issue due to it. There can be other non-trivial cases which can't be foreseen.

amirgon, 2019-03-01:
@kisvegabor @embeddedt  
If anything, I think that wrapping `#undef free` with `#ifdef` to check if we are in micropython context is **less safe**. If, for example, someone defines `free` from some H file unrelated to micropython, you'd get the same problem.    

However, if you insist on this - we can add a wrapping `#ifdef`.  
I didn't find a suitable define in [`py.mk`](https://github.com/littlevgl/lv_micropython/blob/master/py/py.mk), but we can add our own by setting `CFLAGS_MOD`, like [here](https://github.com/littlevgl/lv_micropython/blob/7589796a3c2e807f49ab04e8e80cb0684a1c8537/py/py.mk#L130).  
If we change this on `lv_micropython` repo, it will take effect there, but anyone who is porting lvgl to his micropython fork would have to add this `CFLAGS_MOD` addition to his makefile as well (in addition to the other lvgl build rules there).

kisvegabor, 2019-03-01:
@amirgon @embeddedt 

Okay, all that being said I vote for simply adding the `#undef` without checking the Micropython environment. 

As @amirgon  said:
> If, for example, someone defines `free` from some H file unrelated to micropython, you'd get the same problem

If someone had a `#define free xxx` then it would already cause issues for him too (assuming `USE_LV_FILESYSTEM  1`) but nobody reported a similar issue so far. So I believe we won't break existing codes.

(This fix won't be required in `v6.0` as the function has renamed)

amirgon, 2019-03-02:
> Okay, all that being said I vote for simply adding the `#undef` without checking the Micropython environment.

@embeddedt What do you think? I would like to know your opinion before I send a PR for this.

embeddedt, 2019-03-02:
@amirgon Let's just add the `#undef` without any `#ifdef` wrappers.

amirgon, 2019-03-02:
@BrendanSimon littlevgl/lvgl#922 was merged to lvgl main (thanks @embeddedt!) and should fix this problem.  

If there is still some problem, please let me know.
Closing.

closed 2019-03-02 by amirgon

amirgon, 2019-03-03:
Forgot to update git sub modules... I've updated them just now so [lv_micropython](https://github.com/littlevgl/lv_micropython) contains this workaround.

BrendanSimon, 2019-03-18:
Didn't work for me at first, but I fell victim of the stale submodules too.
After updating the submodules the build got further.  It crapped out but not due to the `free` macro issue.

```console
arm-none-eabi-gcc -I../../lib/lv_bindings/lvgl -I../../lib/lv_bindings/include -I. -I../.. -Ibuild-OLIMEX_E407 -I../../lib/cmsis/inc -I../../lib/stm32lib/CMSIS/STM32F4xx/Include/ -I../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc -Iusbdev/core/inc -Iusbdev/class/inc -I../../lib/lwip/src/include -Ilwip_inc -Wall -Wpointer-arith -Werror -std=gnu99 -nostdlib -DFFCONF_H=\"lib/oofatfs/ffconf.h\"   -Wno-unused-function  -DSTM32F407xx -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mtune=cortex-m4 -mcpu=cortex-m4 -Os -DNDEBUG -Iboards/OLIMEX_E407 -DSTM32_HAL_H='<stm32f4xx_hal.h>' -DMICROPY_HW_VTOR=0x08000000 -DMICROPY_FLOAT_IMPL=MICROPY_FLOAT_IMPL_FLOAT -fsingle-precision-constant -Wdouble-promotion -fdata-sections -ffunction-sections -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -c -MD -o build-OLIMEX_E407/mphalport.o mphalport.c
mphalport.c: In function 'mp_hal_gpio_clock_enable':
mphalport.c:119:33: error: 'RCC_AHB1ENR_GPIOAEN_Pos' undeclared (first use in this function); did you mean 'RCC_AHB1ENR_GPIOAEN'?
     #define AHBxENR_GPIOAEN_Pos RCC_AHB1ENR_GPIOAEN_Pos
                                 ^~~~~~~~~~~~~~~~~~~~~~~
mphalport.c:129:27: note: in expansion of macro 'AHBxENR_GPIOAEN_Pos'
     RCC->AHBxENR |= 1 << (AHBxENR_GPIOAEN_Pos + gpio_idx);
                           ^~~~~~~~~~~~~~~~~~~
mphalport.c:119:33: note: each undeclared identifier is reported only once for each function it appears in
     #define AHBxENR_GPIOAEN_Pos RCC_AHB1ENR_GPIOAEN_Pos
                                 ^~~~~~~~~~~~~~~~~~~~~~~
mphalport.c:129:27: note: in expansion of macro 'AHBxENR_GPIOAEN_Pos'
     RCC->AHBxENR |= 1 << (AHBxENR_GPIOAEN_Pos + gpio_idx);
                           ^~~~~~~~~~~~~~~~~~~
make: *** [build-OLIMEX_E407/mphalport.o] Error 1
```

embeddedt, 2019-03-18:
@BrendanSimon I think that's a problem with the STM32 port on Micropython's side, not LittlevGL.

amirgon, 2019-03-26:
Fixed on micropython upstream by micropython/micropython@781947a.