lvgl/lv_binding_micropython #12: Can not support mouse event on Mac os

fanday, 2019-02-27:
It seems that mouse could not work on Mac os.any ideas? Thanks.

amirgon, 2019-02-27:
@fanday  I assume you tried to build and run the unix port of `lv_micropython`, is that correct?  
Is the display working ok?   

The SDL driver for mouse and display is the same driver used on [lv_drivers](https://github.com/littlevgl/lv_drivers) repo.  
Did you try to run the [pc simulator](https://docs.littlevgl.com/#PC-simulator)? Could you confirm it runs correctly on Mac OS?  

@kisvegabor - any idea why the SDL driver wouldn't work on Mac?

embeddedt, 2019-02-27:
@amirgon @fanday I think there is a setting in `lv_drv_conf.h` for using SDL on Mac OS.

amirgon, 2019-02-27:
@embeddedt Did you mean [USE_EVDEV](https://github.com/littlevgl/lv_binding_micropython/blob/0ac20982d4dbe5939600236a6786a40219abd444/driver/SDL/lv_drv_conf.h#L205)? There is no mention of Mac or OSX...

embeddedt, 2019-02-27:
@amirgon I think I got mixed up... I was referring to this but it looks like it's defined automatically:

https://github.com/littlevgl/lv_drivers/blob/b9825c196924275073402637f056e25362086c9d/display/monitor.c#L34-L38

embeddedt, 2019-02-27:
@amirgon  Additionally,  this needs to be used in the `while(1)` loop:

```c

#if defined(__APPLE__) && defined(TARGET_OS_MAC)
# if __APPLE__ && TARGET_OS_MAC
#define SDL_APPLE
# endif
#endif

...

        #ifdef SDL_APPLE
            SDL_Event event;
            
            while(SDL_PollEvent(&event)) {
                #if USE_MOUSE != 0
                    mouse_handler(&event);
                #endif

                #if USE_KEYBOARD
                    keyboard_handler(&event);
                #endif

                #if USE_MOUSEWHEEL != 0
                    mousewheel_handler(&event);
                #endif
            }
        #endif
```

fanday, 2019-02-28:
> @fanday I assume you tried to build and run the unix port of `lv_micropython`, is that correct?
> Is the display working ok?
> 
> The SDL driver for mouse and display is the same driver used on [lv_drivers](https://github.com/littlevgl/lv_drivers) repo.
> Did you try to run the [pc simulator](https://docs.littlevgl.com/#PC-simulator)? Could you confirm it runs correctly on Mac OS?
>  @amirgon  the LCD works fine, only the mouse event not work.BTW, the littlevgl C demo works fine on mac os.
> @kisvegabor - any idea why the SDL driver wouldn't work on Mac?

fanday, 2019-02-28:
> @fanday I assume you tried to build and run the unix port of `lv_micropython`, is that correct?
> Is the display working ok?
>
> The SDL driver for mouse and display is the same driver used on [...]
@amirgon the LCD works fine when use mpy, only the mouse event not work.BTW, the littlevgl pc simulator works fine on mac os.

fanday, 2019-02-28:
> @amirgon Additionally, this needs to be used in the `while(1)` loop:
>
> ```c
> #if defined(__APPLE__) && defined(TARGET_OS_MAC)
> # if __APPLE__ && TARGET_OS_MAC
> #define SDL_APPLE
> # endif
> #endif [...]

these code could not enable in the SDL refresh thread, I have try to enable it and it crash the app.

embeddedt, 2019-02-28:
@fanday The code should be put underneath the call to `lv_task_handler`.

kisvegabor, 2019-02-28:
For some reason (not clear why) the event handling needs to be called fro the main thread on OSX. Like here: https://github.com/littlevgl/pc_simulator_sdl_eclipse/blob/master/main.c#L81

@amirgon Is it possible to do something similar in mpy?

amirgon, 2019-02-28:
> For some reason (not clear why) the event handling needs to be called fro the main thread on OSX. Like here: https://github.com/littlevgl/pc_simulator_sdl_eclipse/blob/master/main.c#L81
> 
> @amirgon Is it possible to do something similar in mpy?

@kisvegabor the event handling is done on the same thread the micropython is running on.  
I assume micropython is running on the main thread on OSX, I'm not sure about this.  

on [modSDL.c](https://github.com/littlevgl/lv_binding_micropython/blob/65b56e1e9d02a333c6a7bc9b8b5986d01aaf413b/driver/SDL/modSDL.c#L15):

```c
STATIC int tick_thread(void * data)
{
    (void)data;

    while(1) {
        SDL_Delay(1);   /*Sleep for 1 millisecond*/
        lv_tick_inc(1); /*Tell LittelvGL that 1 milliseconds were elapsed*/
        mp_sched_schedule((mp_obj_t)&mp_lv_task_handler_obj, mp_const_none);
    }

    return 0;
}
```

Here, `tick_thread` is running on a different thread, but `mp_sched_schedule` schedules the actual event handling **back on the micropython thread**.

amirgon, 2019-02-28:
@kisvegabor The problem is probably [here](https://github.com/littlevgl/lv_binding_micropython/blob/65b56e1e9d02a333c6a7bc9b8b5986d01aaf413b/driver/SDL/SDL_monitor.c#L79):

```c
void monitor_init(void)
{
    /*OSX needs to initialize SDL here*/
#ifdef MONITOR_APPLE
    monitor_sdl_init();
#endif

    SDL_CreateThread(monitor_sdl_refr_thread, "sdl_refr", NULL);
    while(sdl_inited == false); /*Wait until 'sdl_refr' initializes the SDL*/
}
``` 

Then `monitor_sdl_refr_thread` calls `monitor_sdl_refr_core` in a loop.
Why not call `monitor_sdl_refr_core` periodically as lv_task, or directly from 
`mp_lv_task_handler` instead of from its own thread?

fanday, 2019-02-28:
> @fanday The code should be put underneath the call to `lv_task_handler`.

@embeddedt  The code work fine now on Mac OS, thanks.

amirgon, 2019-02-28:
> @embeddedt The code work fine now on Mac OS, thanks.

@fanday Could you summarize what is the change you did that solved your problem?

embeddedt, 2019-02-28:
@amirgon He added the code I suggested here: https://github.com/littlevgl/lv_binding_micropython/issues/12#issuecomment-468075846

It needed to go under the call to `lv_task_handler` just like in the PC simulator.

amirgon, 2019-02-28:
> @amirgon He added the code I suggested here: [#12 (comment)](https://github.com/littlevgl/lv_binding_micropython/issues/12#issuecomment-468075846)
> 
> It needed to go under the call to `lv_task_handler` just like in the PC simulator.

Thanks @embeddedt!

I've submitted a change to SDL driver so the event loop runs on the same thread as Micropython. This should fix both this issue and issues related to concurrency, since Micropython Bindings is design to run lvgl and the display driver [on the same thread](https://github.com/littlevgl/lv_binding_micropython/blob/master/README.md#concurrency) as micropython.  

@kisvegabor I don't know much about SDL, so I would appreciate if you could review 77b0c9f26.

@fanday I don't have access to a Mac machine. I **think** this solves your issue. I would appreciate if you could verify it. If it doesn't please re-open the issue.

Closing.

closed 2019-02-28 by amirgon

kisvegabor, 2019-03-01:
@amirgon 

It should be fine if SDL is inited in the same thread as `monitor_sdl_refr_core()`.

I'm considering to restructure the PC simulator project this way (init SDL in `main()` and add `monitor_sdl_refr_core` to the main `while(1)`).