lvgl/lv_binding_micropython #13: Support for regular python.

traverseda, 2019-03-01:
It would be nice if this was pip-installable under normal python. I don't really understand the differences between a micropython C binding and a normal python C binding, would that be viable?

amirgon, 2019-03-01:
@traverseda Micropython and python [are different](https://github.com/micropython/micropython/wiki/Differences). It's not the same code base, and not even the same design. Binding for Micropython does not provide binding for Python, so you can't use it for running LittlevGL in Python.  

Actually, there was some work on **Python** bindings by @rreilink, see [pylvgl](https://github.com/rreilink/pylvgl), but I guess the work there is not complete yet.  

If you want to continue this discussion, I suggest doing that on [LittlevGL forum](https://forum.littlevgl.com/).

closed 2019-03-01 by amirgon

traverseda, 2019-03-01:
Thanks for letting me know.