lvgl/lv_binding_micropython #14: AttributeError: 'bytes' object has no attribute 'encode'

traverseda, 2019-03-04:
Using the https://github.com/littlevgl/lv_micropython make files

I'm presuming this is because my python command points to python3? It would be good to explicitly set a python version.

```
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
LVGL-GEN build/lvgl/lv_mpy.c
make[1]: Entering directory '/home/traverseda/code/lv_mpy/mpy-cross'
Use make V=1 or set BUILD_VERBOSE in your environment to increase build verbosity.
LVGL-GEN build/lvgl/lv_mpy.c
Traceback (most recent call last):
  File "../../lib/lv_bindings/gen/gen_mpy.py", line 1028, in <module>
    gen_obj(enum_name)
  File "../../lib/lv_bindings/gen/gen_mpy.py", line 1018, in gen_obj
    locals_dict_entries = ",\n    ".join(gen_obj_methods(obj_name)),
  File "../../lib/lv_bindings/gen/gen_mpy.py", line 958, in gen_obj_methods
    format(enum_member = get_enum_member_name(enum_memebr_name), enum_member_value = get_enum_value(obj_name, enum_memebr_name)) for enum_memebr_name in get_enum_members(obj_name)]
  File "../../lib/lv_bindings/gen/gen_mpy.py", line 958, in <listcomp>
    format(enum_member = get_enum_member_name(enum_memebr_name), enum_member_value = get_enum_value(obj_name, enum_memebr_name)) for enum_memebr_name in get_enum_members(obj_name)]
  File "../../lib/lv_bindings/gen/gen_mpy.py", line 188, in get_enum_value
    numstr = struct.pack('<i', enum[enum_member_name]).encode('string_escape')
AttributeError: 'bytes' object has no attribute 'encode'
make: *** [../../py/py.mk:136: build/lvgl/lv_mpy.c] Error 1
make: *** Deleting file 'build/lvgl/lv_mpy.c'
make: *** Waiting for unfinished jobs....
Traceback (most recent call last):
  File "../lib/lv_bindings/gen/gen_mpy.py", line 1028, in <module>
    gen_obj(enum_name)
  File "../lib/lv_bindings/gen/gen_mpy.py", line 1018, in gen_obj
    locals_dict_entries = ",\n    ".join(gen_obj_methods(obj_name)),
  File "../lib/lv_bindings/gen/gen_mpy.py", line 958, in gen_obj_methods
    format(enum_member = get_enum_member_name(enum_memebr_name), enum_member_value = get_enum_value(obj_name, enum_memebr_name)) for enum_memebr_name in get_enum_members(obj_name)]
  File "../lib/lv_bindings/gen/gen_mpy.py", line 958, in <listcomp>
    format(enum_member = get_enum_member_name(enum_memebr_name), enum_member_value = get_enum_value(obj_name, enum_memebr_name)) for enum_memebr_name in get_enum_members(obj_name)]
  File "../lib/lv_bindings/gen/gen_mpy.py", line 188, in get_enum_value
    numstr = struct.pack('<i', enum[enum_member_name]).encode('string_escape')
AttributeError: 'bytes' object has no attribute 'encode'
make[1]: *** [../py/py.mk:136: build/lvgl/lv_mpy.c] Error 1
make[1]: *** Deleting file 'build/lvgl/lv_mpy.c'
make: *** [../../py/mkrules.mk:108: ../../mpy-cross/mpy-cross] Error 2
make[1]: Leaving directory '/home/traverseda/code/lv_mpy/mpy-cross'
```

amirgon, 2019-03-04:
Thank you for reporting this!  
You are right. I'm using Python 2.7.  
I will fix this, to either be compatible with Python 3, or require Python 2.  

In the meanwhile as a workaround, you can add `PYTHON=...` argument to your make command, and set it to point to Python 2.7.

amirgon, 2019-03-05:
Fixed on a56dd03.
Closing.

closed 2019-03-05 by amirgon