lvgl/lv_binding_micropython #15: advanced_demo.py closes immediatly

traverseda, 2019-03-04:
I'm not sure if if supposed to be entering a loop or what, but when I run the basic example it closes immediately with no error output.

Not sure if this would be a better issue here or at https://github.com/littlevgl/lv_micropython

amirgon, 2019-03-05:
@traverseda `advanced_demo.py` is not supposed to close immediately.  
It is supposed to display an interactive GUI with two pages, a few buttons etc.  
It is working for me on both unix and ESP32 port, so I'm not sure what the problem is on your side.  

It would help if you could provide more information:

- What micropython port are you using?
- How did you build it? What make command did you use?
- Were there any warnings or other problems during the build?
- Did you get the micropython prompt? Did you try to use it?
- How did you load `advanced_demo.py` on your port? On the unix port the simplest thing is to copy it, click <kbd>CTRL</kbd>+<kbd>e</kbd>, paste the script, then click <kbd>CTRL</kbd>+<kbd>d</kbd> to run the script you pasted.  
On ESP32 it's better to first write the script to the file system then import it from there.

fanday, 2019-03-05:
> I'm not sure if if supposed to be entering a loop or what, but when I run the basic example it closes immediately with no error output.
> 
> Not sure if this would be a better issue here or at https://github.com/littlevgl/lv_micropython

@traverseda  Try to add the while True: pass at the end of file.

amirgon, 2019-03-05:
> @traverseda Try to add the while True: pass at the end of file.

@fanday I don't think this is the issue. The REPL (interactive prompt) should remain open and responsive to user input. If it closes, there is some problem.  
Also worth running it with gdb and see if there was some exception thrown, invalid memory access etc.

fanday, 2019-03-05:
> @fanday I don't think this is the issue. The REPL (interactive prompt) should remain open and responsive to user input. If it closes, there is some problem.
> Also worth running it with gdb and see if [...]

@amirgon Somebody always try run the script as this:./micropython advanced_demo.py.And it seems nothing on the screen.

amirgon, 2019-03-05:
> @amirgon Somebody always try run the script as this:./micropython advanced_demo.py.And it seems nothing on the screen.

@fanday You can't run the script like that, it will not work. It should run in micropython with `lvgl` module, not as a standalone Python script.  
If someone tries to run it like this, he would get `import error` immediately since `lvgl` module is missing on regular Python.

fanday, 2019-03-05:
> @fanday You can't run the script like that, it will not work. It should run in micropython with `lvgl` module, not as a standalone Python script.
> If someone tries to run it like this, he would get [...]

@amirgon Yes,you are right.I can run it after build the lv_mpy.

traverseda, 2019-03-05:
I can confirm that running the repl and typing `import advanced_demo` works fine, but trying to run it like a normal python script using `micropython advanced_demo.py` causes it to close immediately.

amirgon, 2019-03-05:
> @traverseda Try to add the while True: pass at the end of file.

@fanday you were right.
The REPL is not run by default when a script is given as a parameter.
Fixed on 50ebf70.

Closing.

closed 2019-03-05 by amirgon