lvgl/lv_binding_micropython #16: How about add a linux framebuffer driver to lv_mpy?

fanday, 2019-03-05:
Is it possible to add a linux frambuffer driver to lv_mpy instead of SDL?Thanks.

amirgon, 2019-03-05:
@fanday  
Yes.  
Should be easy to do that based on https://github.com/littlevgl/linux_frame_buffer

fanday, 2019-03-05:
> @fanday
> Yes.
> Should be easy to do that based on https://github.com/littlevgl/linux_frame_buffer
> 
> But what about input device? How do you get the mouse events?

@amirgon I would like to run lv_mpy on arm linux,and get touch event with touch screen.

amirgon, 2019-03-05:
> I would like to run lv_mpy on arm linux,and get touch event from touch screen.

@fanday That's great!  

You would need an input device driver as well.
There are two options:

- Write the indev driver in C.
- Write the indev driver in micropython. I've just added [an example](https://github.com/littlevgl/lv_binding_micropython/blob/master/driver/generic/lvindev_example.py) for this yesterday, if you want to have a look. (I'm thinking of changing it a little so this would not be the final way of doing it, but it would get you a sense of how to do it)

amirgon, 2019-03-14:
Added framebuffer driver.
See [fb_test.py](https://github.com/littlevgl/lv_binding_micropython/blob/master/driver/linux/fb_test.py) for a simple example how to use it.

Closing.

closed 2019-03-14 by amirgon