lvgl/lv_binding_micropython #18: Duplicate constant Enum attributes

rreilink, 2019-03-10:
We have chosen to map e.g. LV_BTN_STYLE_xxx enum constants to btn.STYLE.xxx and LV_IMGBTN_STYLE_xxx to imgbtn.STYLE.xxx . However, this causes an issue since imgbtn inherits from btn. As seen in the generated code:
```c
STATIC const mp_rom_map_elem_t imgbtn_locals_dict_table[] = {
...
    { MP_ROM_QSTR(MP_QSTR_STYLE), MP_ROM_PTR(&mp_LV_BTN_STYLE_type) },
    { MP_ROM_QSTR(MP_QSTR_STYLE), MP_ROM_PTR(&mp_LV_IMGBTN_STYLE_type) 
```

This issue occurs on several other places. lv_roller has even three enums: `LV_PAGE_STYLE`, `LV_DDLIST_STYLE` and LV_ROLLER_STYLE`.

I see a few different options to resolve this:
1. keep the `BTN_` prefix so one would access the enum values as `btn.BTN_STYLE.xxx`
2. do not propagate enums to inherited object classes, so `btn.STYLE.xxx` refers to `LV_BTN_STYLE_xxx` whereas `imgbtn.STYLE.xxx` refers to `LV_IMGBTN_STYLE.xxx`
3. do propagate to inherited object classes, but let the derived class take precedence in case of duplicates
4. Do not associate enums to object classes but make them module global, so one would refer to `lvgl.BTN_STYLE.xxx`

Each have their drawbacks. 
1. seems a bit duplicate/verbose; 
2 violates the idea of inheritance (which will be an issue in regular Python bindings which do already use true inheritance); 
3. is confusing: self.STYLE.xxx would refer to LV_BTN_STYLE_xxx if self is a btn but to LV_IMG_BTN_xxx if self is an imgbtn; such that code which works for a btn would not work for an imgbtn, even though imgbtn inherits from btn.
4 adds more items to the module global namespace.

I think 2 and 3 would be confusing and prone to creating bugs, so I would suggest option 1 or 4 (or maybe there are other solutions).

amirgon, 2019-03-12:
@rreilink You are right, there is a problem here.  
It is not limited to enums, it's also relevant to functions structs and globals, in case an object and its parent have the same function/struct/global name. Even if this does not happen today, it might happen in the future and we would want to be immuned to this scenario.

How about adding an indirection level for accessing the parent member?  

For example, for a `my_imgbtn` object of type `imgbtn`:
`my_imgbtn.STYLE` would be the `STYLE` enum of the inherited imbtn, while `my_imgbtn.btn.STYLE` would be the style enum of the parent btn. This can easily be done in micropython by defining `btn` as an attribute of `my_imgbtn` object.  
We get another benefit from this. It will provide a simple way to "cast" `imgbtn` to a `btn`, since the `btn` attribute of `imgbtn` would return a `btn` object **of type `btn`** that can be used as any other btn.

I think that we should still provide access to the parent members directly from the child when there is no ambiguity (when the same enum/function name does not exist on both parent and child), otherwise we would have to write: `my_imgbtn.btn.cont.obj.set_x` instead of `my_imgbtn.set_x`, for example.  
It makes more sense to have `my_imgbtn.set_x` when `set_x` only exists on the parent `obj` and not in any of its children, especially since this ambiguity is rare.

rreilink, 2019-03-12:
I just ran a check on the code (I checked it programmatically so I wouldn't miss anything) and the only issue is with the .STYLE. The other duplicates are `set_xx`, `get_xx`, `focus` and `clean`; those are clearly ment as duplicates since they override behavior for derived classes.

I think we should *not* use some attribute access like `.btn.cont.obj` to access parent members. The whole idea of inheritance is that one can change the behavior by providing a member with the same name as a parent. If a derived class provides a method of the same name as a parent, it should do a similar thing or extend the behavior of the parent as appropriate for that particular class.

> Even if this does not happen today, it might happen in the future and we would want to be immuned to this scenario.

I would view it the other way around: if a clash occurs, it could be a sign that something is not clear / not in order. If both a parent and a child implement `do_xxx` and that does something different for the two objects, it would surely be confusing. Since we now have bindings to true object-oriented languages, future lvgl adaptations could be implemented with those in mind.

My conclusion: only the enum constants need to be solved. 

By the way: in 'regular' Python, I have implemented true inheritance, so the `super()` function can be used to explicitly access methods of an ancestor. However as I understood it true inheritance cannot be implemented in MicroPython?

kisvegabor, 2019-03-13:
I wasn't fully concious when named those functions which overwrites the ancestors' behavior. However, it would be great and imporatnt to be more consequent with this. 

The API of the obejcts are designt to use only `lv_type_...` and `lv_obj_...` with them. For example, you shouldn't use `lv_btnm_set_map()` directly on a keyboard.  I wasn't sure there are no naming conflicts with in common words like `width` but @rreilink confirmed that luckily it's fine. Anyway a review on API function names with this in mind might be required.

The STYLES are really different for every obejct type but here only the "top level" obejct type's styles should be used.

rreilink, 2019-03-13:
> The API of the objects are designed to use only `lv_type_...` and `lv_obj_...` with them

Ah, that clarifies things. It would mean that in (Micro)Python, every object should inherit just from `obj` and not from any other object. That way, all functions that are not defined for that specific object, will not be present (i.e. there will be no `my_kb.set_map(..)` any more).

> The STYLES are really different for every object type but here only the "top level" object type's styles should be used.

If we only derive from `lv_obj`, the clashes on the STYLES will disappear as only the STYLE for that particular object will be present.

@kisvegabor, @amirgon do you agree to mark the issue as invalid and open a new one that all objects should derive only from `lv_obj`? Or rename this issue? If we agree, I will do the same for regular Python.

amirgon, 2019-03-13:
> By the way: in 'regular' Python, I have implemented true inheritance, so the `super()` function can be used to explicitly access methods of an ancestor. However as I understood it true inheritance cannot be implemented in MicroPython?

In Micropython there is a distinction between "native" classes and user defined "non-native" classes. While inheritance is fully implemented for pure python user defined "non-native" classes, native classes (that are defined in C) have some restrictions on them. More info on micropython/micropython#1159 and https://forum.micropython.org/viewtopic.php?f=3&t=5772&p=33166#p33166.

> Ah, that clarifies things. It would mean that in (Micro)Python, every object should inherit just from `obj` and not from any other object. That way, all functions that are not defined for that specific object, will not be present (i.e. there will be no `my_kb.set_map(..)` any more).

I'm not sure I understand.  
Doesn't that defeat the whole idea of inheritance? For example, why define both `lv_cont_set_fit` and `lv_btn_set_fit` that do the same thing instead of calling `lv_cont_set_fit` on a `btn` object which is a child of `cont`? @kisvegabor How does it makes sense to repeat all parent functions for every child?   
Of course, if it's designed that way, it makes things simpler for the bindings as @rreilink suggests.  

If you are sure it makes sense to inherit only from `obj`, I'll close this issue and open a new one.

kisvegabor, 2019-03-14:
> Doesn't that defeat the whole idea of inheritance? For example, why define both lv_cont_set_fit and lv_btn_set_fit that do the same thing instead of calling lv_cont_set_fit on a btn object which is a child of cont? 

It makes things clearer.  Without adding all parent functions to the child the user would need to use the ancestor's API functions which:
- makes the API more complicated (the user need to check the ancesto's functions too)
- the code itself becomes messy (E.g. using `lv_obj_`, `lv_cont_` and `lv_btn_` when creating a button) 
- some ancestor function doesn't make sense with the derived object. E.g. `switch` is derived from `slider`. You could use `lv_slider_set_value()` but it doesn't make any sense for `switch`. Therefore switch has `lv_sw_on/off` which uses `lv_slider_set_value`.

amirgon, 2019-03-14:
Continue this on #19.

Closing.

closed 2019-03-14 by amirgon