lvgl/lv_binding_micropython #20: `RCC_AHB1ENR_GPIOAEN_Pos` undefined error for stm32 port

BrendanSimon, 2019-03-18:
I get the following error when trying to build for an stm32 port (`OLIMEX_E407`).

Looks like the `CMSIS/STM32F4xx` header files are not up to date, as `RCC_AHB1ENR_GPIOAEN_Pos` is not defined in the `lv_micropython` sources, but is defined in the upstream `micropython` sources.

I updated the submodules with `git submodule update --remote -- recursive` but that didn't resolve the issue.

```console
arm-none-eabi-gcc -I../../lib/lv_bindings/lvgl -I../../lib/lv_bindings/include -I. -I../.. -Ibuild-OLIMEX_E407 -I../../lib/cmsis/inc -I../../lib/stm32lib/CMSIS/STM32F4xx/Include/ -I../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc -Iusbdev/core/inc -Iusbdev/class/inc -I../../lib/lwip/src/include -Ilwip_inc -Wall -Wpointer-arith -Werror -std=gnu99 -nostdlib -DFFCONF_H=\"lib/oofatfs/ffconf.h\"   -Wno-unused-function  -DSTM32F407xx -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mtune=cortex-m4 -mcpu=cortex-m4 -Os -DNDEBUG -Iboards/OLIMEX_E407 -DSTM32_HAL_H='<stm32f4xx_hal.h>' -DMICROPY_HW_VTOR=0x08000000 -DMICROPY_FLOAT_IMPL=MICROPY_FLOAT_IMPL_FLOAT -fsingle-precision-constant -Wdouble-promotion -fdata-sections -ffunction-sections -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -c -MD -o build-OLIMEX_E407/mphalport.o mphalport.c
mphalport.c: In function 'mp_hal_gpio_clock_enable':
mphalport.c:119:33: error: 'RCC_AHB1ENR_GPIOAEN_Pos' undeclared (first use in this function); did you mean 'RCC_AHB1ENR_GPIOAEN'?
     #define AHBxENR_GPIOAEN_Pos RCC_AHB1ENR_GPIOAEN_Pos
                                 ^~~~~~~~~~~~~~~~~~~~~~~
mphalport.c:129:27: note: in expansion of macro 'AHBxENR_GPIOAEN_Pos'
     RCC->AHBxENR |= 1 << (AHBxENR_GPIOAEN_Pos + gpio_idx);
                           ^~~~~~~~~~~~~~~~~~~
mphalport.c:119:33: note: each undeclared identifier is reported only once for each function it appears in
     #define AHBxENR_GPIOAEN_Pos RCC_AHB1ENR_GPIOAEN_Pos
                                 ^~~~~~~~~~~~~~~~~~~~~~~
mphalport.c:129:27: note: in expansion of macro 'AHBxENR_GPIOAEN_Pos'
     RCC->AHBxENR |= 1 << (AHBxENR_GPIOAEN_Pos + gpio_idx);
                           ^~~~~~~~~~~~~~~~~~~
make: *** [build-OLIMEX_E407/mphalport.o] Error 1
```

BrendanSimon, 2019-03-18:
As a hack, I copied all the STM32F4 CMSIS files and also the HAL files from micropython to lv_micorpython.  I got passed the errors above, but now have issues with ...

```console
arm-none-eabi-gcc -I../../lib/lv_bindings/lvgl -I../../lib/lv_bindings/include -I. -I../.. -Ibuild-OLIMEX_E407 -I../../lib/cmsis/inc -I../../lib/stm32lib/CMSIS/STM32F4xx/Include/ -I../../lib/stm32lib/STM32F4xx_HAL_Driver/Inc -Iusbdev/core/inc -Iusbdev/class/inc -I../../lib/lwip/src/include -Ilwip_inc -Wall -Wpointer-arith -Werror -std=gnu99 -nostdlib -DFFCONF_H=\"lib/oofatfs/ffconf.h\"   -Wno-unused-function  -DSTM32F407xx -mthumb -mfpu=fpv4-sp-d16 -mfloat-abi=hard -mtune=cortex-m4 -mcpu=cortex-m4 -Os -DNDEBUG -Iboards/OLIMEX_E407 -DSTM32_HAL_H='<stm32f4xx_hal.h>' -DMICROPY_HW_VTOR=0x08000000 -DMICROPY_FLOAT_IMPL=MICROPY_FLOAT_IMPL_FLOAT -fsingle-precision-constant -Wdouble-promotion -fdata-sections -ffunction-sections -DMICROPY_QSTR_EXTRA_POOL=mp_qstr_frozen_const_pool -DMICROPY_MODULE_FROZEN_MPY -c -MD -o build-OLIMEX_E407/lib/lv_bindings/lvgl/lv_core/lv_obj.o ../../lib/lv_bindings/lvgl/lv_core/lv_obj.c
In file included from ../../lib/lv_bindings/lvgl/lv_core/lv_obj.c:25:
../../lib/lv_bindings/lvgl/lv_core/lv_obj.c: In function 'lv_init':
../../py/mpstate.h:258:40: error: 'mp_state_vm_t' {aka 'struct _mp_state_vm_t'} has no member named '_lv_def_scr'
 #define MP_STATE_VM(x) (mp_state_ctx.vm.x)
                                        ^
./mpconfigport.h:270:23: note: in expansion of macro 'MP_STATE_VM'
 #define MP_STATE_PORT MP_STATE_VM
                       ^~~~~~~~~~~
../../lib/lv_bindings/lvgl/../../lv_conf.h:34:26: note: in expansion of macro 'MP_STATE_PORT'
 #   define LV_GC_ROOT(x) MP_STATE_PORT(x)
                          ^~~~~~~~~~~~~
../../lib/lv_bindings/lvgl/lv_core/lv_obj.c:71:5: note: in expansion of macro 'LV_GC_ROOT'
     LV_GC_ROOT(_lv_def_scr) = NULL;
     ^~~~~~~~~~
...
```

BrendanSimon, 2019-03-18:
Looks like copying the STM32F4 CMSIS and HAL files is not required.  My submodules were not checked out with the correct branch.  I don't know if it's git or sourcetree, but I noticed in sourcetree window for the submodules didn't have the correct branches checked - so I checked them out and things were ok.

I'm still stuck with the above errors though.  I even blew away my repo and cloned everything from scratch, but still get the errors above.

BrendanSimon, 2019-03-18:
Should this issue be closed and start a new issue for the above errors?

embeddedt, 2019-03-18:
The previous errors weren't a binding issue, but these ones appear to be, so let's leave this topic open.

I think @amirgon will be able to help you.

amirgon, 2019-03-18:
@BrendanSimon there are two issues here:

1. The original `RCC_AHB1ENR_GPIOAEN_Pos` problem. Is there some problem with a git submodule?  
Please try to clone a clean repo with `git clone --recurse-submodules ...` and let me know if you can reproduce the problem.

2. `mp_state_ctx` issue.  
Currently I've modified only the `unix` and `esp32` ports to support micropython, but other ports (such as `stm32`) can be easily added. This is the reason for the error you get - missing lvgl gc root pointers.  
Please edit `mpconfigport.h` on your port:
    - Include `"lib/lv_bindings/lvgl/lv_misc/lv_gc.h"`, like is done [here](https://github.com/littlevgl/lv_micropython/blob/c57ba5fb2de366707a95b53007e76d3d005b51c0/ports/esp32/mpconfigport.h#L223) (`MICROPY_PY_LVGL` is defined [here](https://github.com/littlevgl/lv_micropython/blob/c57ba5fb2de366707a95b53007e76d3d005b51c0/ports/esp32/mpconfigport.h#L56))
    - Add `LV_ROOTS` to `MICROPY_PORT_ROOT_POINTERS`, like is done [here](https://github.com/littlevgl/lv_micropython/blob/c57ba5fb2de366707a95b53007e76d3d005b51c0/ports/esp32/mpconfigport.h#L229).  
The purpose of this change is to add lvgl root pointers to gc.
    - Add `lvgl` module to micropython built in modules, (so you could import it in python), as done [here](https://github.com/littlevgl/lv_micropython/blob/c57ba5fb2de366707a95b53007e76d3d005b51c0/ports/esp32/mpconfigport.h#L215).
   - You would probably need display/input drivers. They can be added as micropython modules as well, like [here](https://github.com/littlevgl/lv_micropython/blob/c57ba5fb2de366707a95b53007e76d3d005b51c0/ports/esp32/mpconfigport.h#L216-L218).
    - Module needs some extern declaration like [this](https://github.com/littlevgl/lv_micropython/blob/c57ba5fb2de366707a95b53007e76d3d005b51c0/ports/esp32/mpconfigport.h#L182-L185).

You also need to add Makefile rules to build your display/input drivers, like [this](https://github.com/littlevgl/lv_micropython/blob/c57ba5fb2de366707a95b53007e76d3d005b51c0/ports/esp32/Makefile#L201-L204).  

Here is the [complete diff](https://github.com/micropython/micropython/compare/v1.10...littlevgl:master#diff-77cbec69bd2e6d723afa755b9d99031bR201) between micropython (v1.10) and lv_micropython.  
Most of it is common to all ports. Per port, only `mpconfigport.h` and `Makefile` contain important changes.

closed 2019-04-21 by stale[bot]