lvgl/lv_binding_micropython #21: Add XPT2046 input device driver

timstegeman, 2019-03-18:
Can you please add support in the ESP32 port for the XPT2046 touch panel controller?

amirgon, 2019-03-19:
@timstegeman Yes I can add it, but I cannot test it with a controller.  
Will you be able to test it?

timstegeman, 2019-03-19:
Yes, of course I can test it.

amirgon, 2019-03-20:
Ok.  
I've added an **untested** xpt2046 driver, based on [lvgl esp32 demo](https://github.com/littlevgl/esp32_ili9431/blob/master/drv/xpt2046.c), assuming it works ok.  
I assume you are using ILI9341, and I assume you use the xpt2046 with the same SPI device.  

So, in order to use the xpt2046 driver in micropython:

- First import ILI9341 driver, initialize and register it, as explained [here](https://github.com/littlevgl/lv_micropython/blob/master/README.md). 
Something like this:

```python
import lvgl as lv
lv.init()
import lvesp32
import ILI9341 as ili
d = ili.display(miso=5, mosi=18, clk=19, cs=13, dc=12, rst=4, backlight=2)
d.init()
disp_drv = lv.disp_drv_t()
lv.disp_drv_init(disp_drv)
disp_drv.disp_flush = d.flush
disp_drv.disp_fill = d.fill
lv.disp_drv_register(disp_drv)
```

- Then import xpt2046 driver, initialize and register it. Something like this:

```python
import xpt2046
touch = xpt2046.xpt2046(d.get_spi_device())
touch.init()
indev_drv = lv.indev_drv_t()
lv.indev_drv_init(indev_drv) 
indev_drv.type = lv.INDEV_TYPE.POINTER;
indev_drv.read = touch.read;
lv.indev_drv_register(indev_drv);
```

Pay attention to this line: `touch = xpt2046.xpt2046(d.get_spi_device())`  
Here we create an instance of the xpt2046 driver.
The first (required) argument on this constructor is the spi device from the display driver, but you can add additional arguments after it.

Addition (optional, kw args) arguments are:
- `cs` : xpt2046 chip select pin number
- `irq` : xpt2046 irq pin number
-  `x_min`, `y_min`, `x_max`, `y_max` : integer parameters for touch panel calibration
- `x_inv`, `y_inv`, `xy_swap`: boolean parameters for inversing the position, if needed.  

All these optional arguments have [default values](https://github.com/littlevgl/lv_binding_micropython/blob/58e6db390808cb6c02c6301765c00abf93171ad6/driver/esp32/modxpt2046.c#L78-L86), but they would probably need to change according to your needs.  

@timstegeman please give it a try, and let me know if it works for you.

timstegeman, 2019-03-21:
I'm sorry, I can't get it to work.

`E (103386) spi_master: spi_device_queue_trans(780): SPI half duplex mode does not support using DMA with both MOSI and MISO phases.`

The same hardware setup works with this example https://blog.littlevgl.com/2019-01-31/esp32

amirgon, 2019-03-21:
@timstegeman I **think** this may be related to an [esp32 limitaion](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#write-and-read-phases):
> Half duplex transactions with both read and write phases are not supported when using DMA. See [Known Issues](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#spi-known-issues) for details and workarounds.  

I'm not sure why we see this on `lv_micropython` project and not on `esp32_ili9431` project, which you confirmed is working with your hardware.

Could you try a few things?

- First, lower the clock speed to 10mhz (default is 40mhz). On the python code, add the argument `mhz=10` to `ili.display` constructor.
- Try to switch it from half to full duplex (on [this line](https://github.com/littlevgl/lv_binding_micropython/blob/58e6db390808cb6c02c6301765c00abf93171ad6/driver/esp32/modILI9341.c#L188) replace `SPI_DEVICE_HALFDUPLEX` by 0).
- Try to Disable DMA (on [this line](https://github.com/littlevgl/lv_binding_micropython/blob/58e6db390808cb6c02c6301765c00abf93171ad6/driver/esp32/modILI9341.c#L204) change the last argument of `spi_bus_initialize` from 1 to 0).

timstegeman, 2019-03-23:
I'm getting there. The full duplex get's rid of the error message, but it now looks like the CS of the display is still activated when reading the touchpad. After this weekend I can continue testing.

amirgon, 2019-03-23:
@timstegeman I just realized one big difference between `esp32_ili9431` and `lv_micropython` drivers:  
`esp32_ili9431` uses two **different SPI devices** for display and touch, [HSPI_HOST for display](https://github.com/littlevgl/esp32_ili9431/blob/8f991f486d909b03ea7054f5721126759edf6f8d/drv/disp_spi.c#L70) and [VSPI_HOST for touch](https://github.com/littlevgl/esp32_ili9431/blob/8f991f486d909b03ea7054f5721126759edf6f8d/drv/tp_spi.c#L63).  
On `lv_micropython` I tried to use the same SPI device. I thought that reusing the same SPI device for both slaves is a good idea, instead of wasting SPI device, HW pins etc.  

Anyway, I think I have a problem there because I need to call `spi_bus_add_device` separately for each SPI slave, and select each `cs` individually (another option is to set `cs` manually). This also may explain the `cs` issue you are experiencing.  

I'll fix that and let you know.

amirgon, 2019-03-24:
@timstegeman  I pushed a fix (202e988).
In this fix, same SPI bus is used by both ili9341 and xpt2046, but two different slave device are registered.
`xpt2046.xpt2046` constructor no longer needs the `d.get_spi_device() argument`. But it can receive two additional optional arguments: 

- `mhz` (default it `20`, on both ili9341 and xpt2046)
- `spihost` (default is `HSPI_HOST`, on both ili9341 and xpt2046)

I kept `flags=SPI_DEVICE_HALFDUPLEX` hoping that now when these are two seperate spi devices, the `E (103386) spi_master: spi_device_queue_trans(780)` error will not happen. If it still happens, you can try changing it to full duplex (0) instead, or disabling DMA.  

Please let me know if you make some progress.

timstegeman, 2019-03-27:
I have made some progress. If I set the clockspeed to 10 Mhz and set `flags=SPI_DEVICE_HALFDUPLEX` to `flags=0` I'll get some coordinates. Now I just need to write some calibration function in MicroPython.

amirgon, 2019-03-27:
> I have made some progress. If I set the clockspeed to 10 Mhz and set `flags=SPI_DEVICE_HALFDUPLEX` to `flags=0` I'll get some coordinates. Now I just need to write some calibration function in MicroPython.

Good to hear! You also see the display, right?  

You can calibrate the coordinates by setting `x_min`, `y_min`, `x_max`, `y_max`.
Did you set it to 10Mhz only for xpt2046? It sounds a bit slow for the display. I'm actually not sure what happens if you set different speeds for two devices on the same SPI bus.

gersch07, 2019-03-27:
Dear timstegeman, dear amirgon,

Congratulations to this amazing work! I got really excited about using micropython and littlevgl together to create a touch UI!
I'm currently trying to follow your steps (on a freshly-flashed ESP32, ILI9341 and XPT2046). Would you mind sharing your wiring and the "init.display"/"xpt2046.xpt2046" code with me?

Since I didn't get the "spihost" command working, I've tried it with: 
d = ili.display(mhz=10, mosi=13, miso=12, clk=14, cs=16, dc=17, rst=4, backlight=2) and
touch = xpt2046.xpt2046(d.get_spi_device(),  irq=32, cs=33)

Unfortunately, all I get is black vertical stripes appearing on the screen while touching it...

amirgon, 2019-03-27:
@gersch07 
> touch = xpt2046.xpt2046(d.get_spi_device(), irq=32, cs=33)

You don't need to pass `d.get_spi_device()` after my [last fix](https://github.com/littlevgl/lv_binding_micropython/issues/21#issuecomment-476014944) (did you pull my last fix?).  
Instead, pass `mhz` and `spihost`, if defaults are not good enough for you.
I believe @timstegeman also changed the C code to full duplex (`flags=0`).

gersch07, 2019-03-27:
@amirgon 
Thank you, yes, I've pulled your last fix and also adopted @timstegeman 's `flags=0` before compiling. 
Could you let me know which pins of the display/touchscreen to connect for the default `HSPI_HOST`? I'd expect "SCK" to be on pin 14, "MISO" on 12, "MOSI" on 13 and "SS"(cs?) on 15. I guess, the remaining ones (dc,rst) still have to be set manually? 
Would I need something like `ili.display(mhz=10, spihost=HSPI_HOST, rst=16,dc=17)` for the display and `touch = xpt2046.xpt2046(mhz=10, spihost=HSPI_HOST,cs=4,irq=32)` for the touchscreen?
Sorry for my stupid questions, I'm a bit confused...

amirgon, 2019-03-27:
@gersch07 

> Could you let me know which pins of the display/touchscreen to connect for the default `HSPI_HOST`? I'd expect "SCK" to be on pin 14, "MISO" on 12, "MOSI" on 13 and "SS"(cs?) on 15. I guess, the remaining ones (dc,rst) still have to be set manually?

That depends on your board. What board are you using? Do you have its schematic? You should look at how the SPI ports (miso, mosi, sck, cs, irq) are connected between the display/touch controllers and your ESP32 MCU.  
HSPI_HOST can be used with any IO pins and signals are routed by the [GPIO matrix](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#gpio-matrix-and-iomux).

gersch07, 2019-03-28:
@amirgon 
Sorry, my bad, I had somehow flashed the old compiled version without your last fix... now I can run the display and register the xpt2046 driver with `spihost=1` without any error popping up. Is there any easy way to check if I connected the touchscreen correctly, like reading out coordinates of the last pressed position (maybe using `lvgl.indev_get_point`)? Thank you for your help!

amirgon, 2019-03-29:
@gersch07 Yes, you can check this with `indev_get_point`.  
Here is an example that displays a label with current input coordinates:

```python

# Assuming you registered and initialized display and input devices!

# Create a screen and a label 

scr = lv.obj()
l = lv.label(scr)
l.set_text('---')
l.align(scr, lv.ALIGN.CENTER, 0, 0)
lv.scr_load(scr)

# Update the label according to input device coordinates

def update_label():
    p = lv.point_t()
    lv.indev_get_point(lv.indev_next(lv.indev_t.cast(None)), p)
    l.set_text('%s:%s' % (p.x,p.y))

if __name__ == '__main__':
    while True:
        update_label()

```

I've added a small fix that allows casting a struct to null (`lv.indev_t.cast(None)`), so please **make sure you pull the latest version of lv_binding_micropython**.

gersch07, 2019-03-31:
@amirgon 
Wow, thanks a lot for your fix! I got the touchscreen now to spit out coordinates, using 
```
d = ili.display(spihost=1,mosi=13,miso=12,clk=14,cs=15,dc=17,rst=4,backlight=2)
touch = xpt2046.xpt2046(spihost=1,mhz=10,cs=19,irq=18)
```
and connecting "T_DIN" to MISO and "T_DO" to MOSI. However, the code only works if xpt2046 is initialized with mhz=10 and ILI9341 without specifying it. And even then, it seems that the coordinates I receive (to some degree) scatter around and are not exactly repeatable, so I guess a calibration will be difficult. I can observe a similar behaviour with the C-based demo app, where touch events appear as red dots on the screen. If you're interested, I could upload a screenshot of that.

@timstegeman 
Did you experience the same issues?

By the way, the display I'm using is from Amazon and labelled as "TJCTM24028-SPI v1.1".

amirgon, 2019-03-31:
> However, the code only works if xpt2046 is initialized with mhz=10 and ILI9341 without specifying it

@gersch07 When you don't configure `mhz` on ILI9341, [the default freq is 20mhz](https://github.com/littlevgl/lv_binding_micropython/blob/cd2636275d3436b07d2db46f81151b86d153b699/driver/esp32/modILI9341.c#L118). You can set `mhz` parameter on the `display` ctor to 20mhz and see that it behaves the same as omitting it.  
You can try setting the display to up to 26mhz which is the maximum when using full duplex with the gpio matrix.  
If you need higher freq than this, it's possible to reach 40mhz when configuring it to half duplex and disabling DMA (but then we would need different SPI bus for the display and for the touch) and up to 80mhz when using the IOMUX pins instead of the GPIO matrix. You can find more information on the [ESP32 docs](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#timing-considerations).

> it seems that the coordinates I receive (to some degree) scatter around and are not exactly repeatable, so I guess a calibration will be difficult.

I would expect that clicking on the same spot on the touch pad would yield the same results, more or less. If the results are non deterministic - this is not related to calibration.  

> I can observe a similar behaviour with the C-based demo app, where touch events appear as red dots on the screen.

The micropython ESP32 drivers are based on the ESP32 drivers on [esp32_ili9341](https://github.com/littlevgl/esp32_ili9341). When you refer to the "C-based demo app", do you mean the [esp32_ili9341 repo](https://github.com/littlevgl/esp32_ili9341)? (If not, could you try it out?) If esp32_ili9341 repo works fine with your display/touch, it would give me a good hint where to look for the problem.  
But if the same problem happens with esp32_ili9341 repo, it's worth opening an issue there.

gersch07, 2019-04-02:
@amirgon 
I just wanted to give a quick update: after some fiddling around with the clock parameter, I found that I can get quite reproducible and reasonable coordinates when driving the XPT2046 at 1 MHz (the display runs at default settings). I guess that setting could fix also my issue in the [esp32_ili9341](https://github.com/littlevgl/esp32_ili9341) repository I had initially tried.

amirgon, 2019-04-02:
> @amirgon
> I just wanted to give a quick update: after some fiddling around with the clock parameter, I found that I can get quite reproducible and reasonable coordinates when driving the XPT2046 at 1 [...]

Thanks for the update @gersch07.
After you have something good enough, please let me know what I should fix in the repo, or just send a PR if you want! (I know I should at least set `flags=0` and change XPT2046 default mhz to 1)

closed 2019-04-30 by stale[bot]

gersch07, 2019-08-11:
@amirgon:
First of all, thanks for keeping this repo updated and sorry for reopening this relatively old issue! 
After some time of absence I pulled the most recent version and found that your code above (to update a label with the current touch position) doesn't work anymore. As in LVGL 5.3, before compiling I had to set `flags=0` to avoid SPI_HALF_DUPLEX errors; in contrast to LVGL 5.3, where I can get reasonable coordinates with identical wiring, I'm now only getting "-1:-1" all the time. Has there been any major change in syntax (instead of `indev_next` -> `indev_get_next`)? Thanks for your help!