lvgl/lv_binding_micropython #22: How to bind lvgl to an custom stm32 board?

turoksama, 2019-03-24:
Hi,
  Thanks for your great work for the project first!
  And I am new to micropython, learn to know how it works these days.
  I noticed /unix(your example) is under /ports directory, and I think it is a mutually exclusive selection between them(please correct me if I am wrong) when porting.
  So, if I have to bind lvgl to my custom stm32 board, the unix port will not work?
  What should i do? Please help me out.
  Thanks!

amirgon, 2019-03-24:
Hi @turoksama!

> I noticed /unix(your example) is under /ports directory, and I think it is a mutually exclusive selection between them(please correct me if I am wrong) when porting.

Not exactly. Most of micropython code is common to all ports, including the binding to lvgl. All the port-specific code, configuration and makefile are under `ports` directory.  

When you want to build micropython, you decide which port to build according to the `make` parameters. For example, `make -C ports/unix/` would build the unix port, while `make -C ports/esp32/` would build the esp32 port ([a few more parameters](https://github.com/littlevgl/lv_micropython#for-esp32-port) are needed for the esp32 port).  

> So, if I have to bind lvgl to my custom stm32 board, the unix port will not work?

Micropython binding for lvgl can be ported to any platform supported by micropython. I currently ported it only to "unix" and "esp32" platforms, just because these are the hardware platforms I use.  
**You can port it to stm32, if you want, and I'll help you with that**. I cannot do it myself because I'm not using stm32.   
If you port it to stm32, please send a PR, so other people could use it too!  

Just recently I discussed with @BrendanSimon on #20 about porting the binding to stm32, and explained how to port lv_micropython to stm32 on [this post](https://github.com/littlevgl/lv_binding_micropython/issues/20#issuecomment-473926869). Please have a look. Why don't you try to implement it and let me know how it is going?  
Most of the work, I think, would be the display and input drivers. But if you already have Littlevgl C drivers, it should be quite simple to wrap them as a micropython modules and use them with micropython bindings. I can also help you with that.

turoksama, 2019-03-24:
@amirgon 
Hi,
 That's very kind of you. I am very much appreciate with that.
 I will learn #20 first and give it a shot, when it is done or something goes wrong, I would love to share!
 Thank you.

turoksama, 2019-03-26:
Hi,
 When I tried to make a default PYBV10, using ac6 sw4stm32 under win10, it failed like this:

...
GEN build-PYBV10/genhdr/qstr.split
../../py/mkrules.mk:78: recipe for target 'build-PYBV10/genhdr/qstr.split' failed
process_begin: CreateProcess(NULL, touch build-PYBV10/genhdr/qstr.split, ...) failed.
make (e=2): 系统找不到指定的文件。

make: *** [build-PYBV10/genhdr/qstr.split] Error 2

What's that suppose to be?

amirgon, 2019-03-26:
@turoksama This doesn't seem to be related to lvgl bindings. It may be some environment issue with win10 when trying to build micropython (maybe some tool is not installed correctly etc. )  
My dev machine is Linux, so I never tried building it on windows.

I have a suggestion.  
Why don't you try building [upsteam micropython](https://github.com/micropython/micropython/tree/v1.10) first?  (the link points to micropython upstream v1.10 release, which is the latest stable release)
Once you build and run pure micropython, we can try making the next step and add the micropython lvgl binding.  

Also have a look at this: https://forum.micropython.org/viewtopic.php?t=1691  
If you are still having trouble building it on Windows, I suggest you try building it on Linux (you could use Linux VM on your windows machine).

turoksama, 2019-03-27:
Hi,
 I did what you said, now i am using ubuntu in VM , but something bad emerged:
```
MPY modules/lcd160cr_test.py
Traceback (most recent call last):
  File "modules/lcd160cr_test.py", line 1
SyntaxError: invalid syntax
../../py/mkrules.mk:116: recipe for target 'build-PYBV10/frozen_mpy/lcd160cr_test.mpy' failed
make: *** [build-PYBV10/frozen_mpy/lcd160cr_test.mpy] Error 1
```

 It is something about frozen py files. Why?

 When I modified the **FROZEN_MPY_DIR ?= modules** to blank FROZEN_MPY_DIR ?=
 report errors:
```
build-PYBV10/py/qstr.o: In function `qstr_init':
qstr.c:(.text.qstr_init+0x14): undefined reference to `mp_qstr_frozen_const_pool'
build-PYBV10/py/qstr.o: In function `qstr_pool_info':
qstr.c:(.text.qstr_pool_info+0x74): undefined reference to `mp_qstr_frozen_const_pool'
build-PYBV10/py/qstr.o: In function `qstr_dump_data':
qstr.c:(.text.qstr_dump_data+0x44): undefined reference to `mp_qstr_frozen_const_pool'
build-PYBV10/py/builtinhelp.o: In function `mp_builtin_help':
builtinhelp.c:(.text.mp_builtin_help+0x180): undefined reference to `mp_frozen_mpy_names'
build-PYBV10/py/frozenmod.o: In function `mp_frozen_stat':
frozenmod.c:(.text.mp_frozen_stat+0x40): undefined reference to `mp_frozen_mpy_names'
build-PYBV10/py/frozenmod.o: In function `mp_find_frozen_module':
frozenmod.c:(.text.mp_find_frozen_module+0x50): undefined reference to `mp_frozen_mpy_names'
frozenmod.c:(.text.mp_find_frozen_module+0x54): undefined reference to `mp_frozen_mpy_content'
Makefile:499: recipe for target 'build-PYBV10/firmware.elf' failed
make: *** [build-PYBV10/firmware.elf] Error 1
```
It looks like these mpy symbols are missing because of excluding from build.
Well, I think it may be a switch some where to exclude from building these mpy files?

amirgon, 2019-03-27:
> I did what you said, now i am using ubuntu in VM

That good! I'm also using Ubuntu (Mint, actually).

> But not with lv_binding , I think I have replaced the whole folder with each other, and then something bad emerged

What do you mean "replaced the whole folder"? Could you try cloning a fresh repo instead?

> It is something about frozen py files. Why?

I don't think I did any change that should affect frozen py files.  
Please have a look at the [complete diff](https://github.com/micropython/micropython/compare/v1.10...littlevgl:master#diff-77cbec69bd2e6d723afa755b9d99031bR201) between micropython (v1.10) and lv_micropython.  
As you can see, only 15 files has changed, most of them on `ports/unix` and `ports/esp32` folders, and none of the changes is related to frozen py files in any way.  

I'm interested in seeing the changes you did on `port/stm32` directory, as explained on #20. Could you send them? Maybe something went wrong there?

**If you are sure this is not some local problem**, please let me know exactly how you build lv_micropython (which ARM toolchain do you use, what `make` command did you use etc.) and I'll try building it on my side. I don't have the stm32 board so I can't run anything, but I can still make sure it builds ok.

turoksama, 2019-03-28:
Hi there,
 Thanks for your help. I would rather love to use git clone cmd than download individually.
 I finally successfully cloned this fresh repo in several hours, due to the network barrier, it really sucks!!! You can't imagine these pains.
 And it(unix port) builds with no error and works fine on ubuntu. 
 I have a question here: how to run a py scritpt file? I don't think it is a good idea to CTRL+E paste codes every time. I used the following cmd but only comes with a flash and then gone: **./micropython test.py**
 Here is my py file.
[test.py.zip](https://github.com/littlevgl/lv_binding_micropython/files/3018498/test.py.zip)

 Well, the following steps should be done in my future work:
 1.add my own modules(c files)
 2.add my own frozen py functions
 3.tweak etc...

amirgon, 2019-03-28:
> I used the following cmd but only comes with a flash and then gone: **./micropython test.py**

See #15. You probably need some infinite loop at the end, like in 50ebf70.

> Well, the following steps should be done in my future work:

@turoksama , Even if you have the unix port working, it will take some more work to support stm32.  
Do you plan to use the stm32 board with micropython+littlevgl? Which display/input hardware are you using with it?

turoksama, 2019-03-29:
@amirgon 
Hmm, that's right, it works fine now.
I plan to use PYB board with some custom modifications, so it may be easier to approach.
Due to the limitation of the mcu(STM32f405Rx), i think a small resolution spi lcd with xpt2046 resistor touch should be fine for the first time.

amirgon, 2019-03-29:
@turoksama You would need a driver for xpt2046 and for your display (you didn't specify which) on stm32.  
Drivers are specific for both the hardware (display/touch) and architecture (stm32).  
Today we have unix and ESP32 drivers, but they will not work on stm32.  

So -

- First make sure you have working lvgl C drivers. Check them and make sure that they work correctly with lvgl (pure lvgl, without micropython) on your stm32 board.  
You must have C driver for the display, but for the input device (touch pad) you can have either C driver or pure micropython driver.
- To use lvgl C drivers on micropython, we must wrap them as micropython module. I can help you with that it you want.

turoksama, 2019-03-30:
Yes, I will do that to make it work in the first place. 
If it does not work I would need your help. So, I will leave this issue open for a while.
That is very kind of you, thanks a lot!

turoksama, 2019-03-31:
@amirgon 
Hi there,
When I dig into the porting issues, noticed these lines and not sure what is used for, would you please help me out?Thanks!

- 1. ports/esp32/Makefile
line 200:
```
ifeq ($(MICROPY_PY_LVGL), 1)
LIB_SRC_C += \
lib/lv_bindings/driver/esp32/**modlvesp32.c**   \   **//does it have to begin with 'mod'**
lib/lv_bindings/driver/esp32/modILI9341.c   \
lib/lv_bindings/driver/esp32/modxpt2046.c   \
lib/lv_bindings/driver/esp32/modrtch.c
endif
```

- 2. ports/esp32/mpconfigport.h
line 182:
```
extern const struct _mp_obj_module_t mp_module_lvgl;    **//remove 'mod' here**
extern const struct _mp_obj_module_t mp_module_lvesp32;
extern const struct _mp_obj_module_t mp_module_ILI9341;
extern const struct _mp_obj_module_t mp_module_xpt2046;
extern const struct _mp_obj_module_t mp_module_rtch;
```

line 216:
```
{ MP_OBJ_NEW_QSTR(MP_QSTR_lvgl), (mp_obj_t)&mp_module_lvgl }, \     
{ MP_OBJ_NEW_QSTR(MP_QSTR_lvesp32), (mp_obj_t)&mp_module_lvesp32 }, \
{ MP_OBJ_NEW_QSTR(MP_QSTR_ILI9341), (mp_obj_t)&mp_module_ILI9341 }, \
{ MP_OBJ_NEW_QSTR(MP_QSTR_xpt2046), (mp_obj_t)&mp_module_xpt2046 }, \
{ MP_OBJ_NEW_QSTR(MP_QSTR_rtch), (mp_obj_t)&mp_module_rtch }, \
```

- 3. py.mk:

128  LVGL_PP = $(BUILD)/lvgl/lvgl.pp.c
**I do not know what it is used for.**

129  LVGL_MPY = $(BUILD)/lvgl/lv_mpy.c
**this file is used to export all lvgl functions name str in upy?**

[image]

As i can see, the third one generates what the first & second do, so, can i just use that tool to do the job instead?

And why not add lvgl as an extmod instead of lib?

amirgon, 2019-03-31:
@turoksama 
> @amirgon
> Hi there,
> When I dig into the porting issues, noticed these lines and not sure what is used for, would you please help me out?Thanks!
>
> * 1. ports/esp32/Makefile
>      line 200:
>
> ```
> ifeq [...]

It doesn't have to begin with "mod" it's just a convention.
These are the additional **files** that should be built as part of the ESP32 make process.

> * 1. ports/esp32/mpconfigport.h
>      line 182:
> 
> ```
> extern const struct _mp_obj_module_t mp_module_lvgl;    **//remove 'mod' here**
> extern const struct _mp_obj_module_t mp_module_lvesp32;
> extern const struct _mp_obj_module_t mp_module_ILI9341;
> extern const struct _mp_obj_module_t mp_module_xpt2046;
> extern const struct _mp_obj_module_t mp_module_rtch;
> ```
> line 216:
> 
> ```
> { MP_OBJ_NEW_QSTR(MP_QSTR_lvgl), (mp_obj_t)&mp_module_lvgl }, \     
> { MP_OBJ_NEW_QSTR(MP_QSTR_lvesp32), (mp_obj_t)&mp_module_lvesp32 }, \
> { MP_OBJ_NEW_QSTR(MP_QSTR_ILI9341), (mp_obj_t)&mp_module_ILI9341 }, \
> { MP_OBJ_NEW_QSTR(MP_QSTR_xpt2046), (mp_obj_t)&mp_module_xpt2046 }, \
> { MP_OBJ_NEW_QSTR(MP_QSTR_rtch), (mp_obj_t)&mp_module_rtch }, \
> ```

On `mpconfigport.h`, we refer to the **exported symbols** that were exported from the new C files we build. For example, the file `modlvesp32.c` [exports `mp_module_lvesp32`](https://github.com/littlevgl/lv_binding_micropython/blob/cd2636275d3436b07d2db46f81151b86d153b699/driver/esp32/modlvesp32.c#L64), so we add `modlvesp32.c` to the makefile, but extern the exported symbol `mp_module_lvesp32` and add it as a new micropython module on `mpconfigport.h`.  
`mp_module_lvgl` is exported by the generated `lv_mpy.c` that is generated from lvgl headers.

> * 1. py.mk:
> 
> 128 LVGL_PP = $(BUILD)/lvgl/lvgl.pp.c
> **I do not know what it is used for.**
> 
> 129 LVGL_MPY = $(BUILD)/lvgl/lv_mpy.c
> **this file is used to export all lvgl functions name str in upy?**
> 

You don't really need to change anything in `py.mk`, it's common to all ports. The gmake rules there do the following:
- Preprocess lvgl header files
- Feed the preprocessed lvgl header file to the script that create `lv_mpy.c` which is lvgl micropython module C file.
- Add `lv_mpy.c` to the build

> [image]
> 
> As i can see, the third one generates what the first & second do, so, can i just use that tool to do the job instead?
> 

I don't understand the question.

> And why not add lvgl as an extmod instead of lib?

It's possible. Currently I added it as a library, to make it easier to port other forks of micropython (minimal change to common files), but `mp_module_lvgl` could be added as a module in other ways.

turoksama, 2019-03-31:
@amirgon 
Sorry about that, i meant that, can we use this step
[image]

to do the step 1&2 instead?
[image]

If yes, I think it is much easier and remain c source files unchanged.

turoksama, 2019-03-31:
@amirgon 

I can't find any rtos related header files, but they all mentioned:

lib/lv_bindings/driver/esp32/modlvesp32.c
lib/lv_bindings/driver/esp32/modILI9341.c  
lib/lv_bindings/driver/esp32/modxpt2046.c  
lib/lv_bindings/driver/esp32/modrtch.c

amirgon, 2019-03-31:
> I can't find any rtos related header files, but they all mentioned:

@turoksama I don't understand your question.  
The files you listed are ESP32 drivers, and can only be compiled and used on the ESP32 port. You cannot use them on stm32.

turoksama, 2019-03-31:
No, I understand that.
My question is:
lvgl c source files and namespace(i mean qstr) are generated automatically by using cyparser, right?
So, can i use cyparser to gen as lvgl do instead of writing manually?

amirgon, 2019-03-31:
> namespace(i mean QSTR etc needed by upy) are generated from lvgl c source files automatically by using cyparser, right?

The lvgl micropython module is generated automatically from lvgl header files using the `gen_mpy.py` script, which uses `pycparser` internally.
The generated lvgl micropython module is common to all ports and can be used by stm32 as well.

> So, can i use cyparser to gen as lvgl do instead of writing manually?

@turoksama, please explain, instead of writing manually what? Are you talking about the few lines you need to manually add to the makefille and `mpconfigport.h`? Or do you refer to something else?

turoksama, 2019-03-31:
Well, my bad. ...instead of writing micropython module source style manually.
looks like below:
```
mp_obj_t mp_machine_spi_make_new(const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *args) {
    // check the id argument, if given
    if (n_args > 0) {
        if (args[0] != MP_OBJ_NEW_SMALL_INT(-1)) {
            #if defined(MICROPY_PY_MACHINE_SPI_MAKE_NEW)
            // dispatch to port-specific constructor
            extern mp_obj_t MICROPY_PY_MACHINE_SPI_MAKE_NEW(const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *all_args);
            return MICROPY_PY_MACHINE_SPI_MAKE_NEW(type, n_args, n_kw, args);
            #else
            mp_raise_ValueError("invalid SPI peripheral");
            #endif
        }
        --n_args;
        ++args;
    }

    // software SPI
    return mp_machine_soft_spi_make_new(type, n_args, n_kw, args);
}
```

something ```mp_obj_t,  const mp_obj_type_t *type, size_t n_args, size_t n_kw, const mp_obj_t *args) ```

They will not ever exist when we write ordinary codes without using micropython, right?

It(micropython) formats ordinary source code into micropython style(assume to call it like that), otherwise it won't work at all.

amirgon, 2019-03-31:
> Well, my bad. ...instead of writing micropython module source style manually.
> looks like below:

So you mean, instead of writing micropython **driver** module, right?
Currently `gen_mpy.py` is designed to wrap only lvgl API into micropython module, but not driver code.
So you can't use it today to convert your C driver into micropython module.   

But lvgl driver API is usually small and simple. For example, display driver only needs a function to create it with some parameters, init it, and expose `flush` and `fill` callbacks.  
So it's very easy to manually wrap lvgl driver in micropython module.  When you have working lvgl drivers for stm32 please let me know, and I'll help you wrap them for micropython.

turoksama, 2019-03-31:
Yes, almost there.
But what if I need to add other c source files as micropython module, and then what?
I don't think it is a good idea to re-write them. So, I would rather love to wrap them with scripts.

amirgon, 2019-03-31:
> But what if I need to add other c source files as micropython module, and then what?

What C module do you plan to add to micropython, other than lvgl library and lvgl drivers?  

I suggest going step by step:

- First let's run lvgl without micropython on stm32, with stm32 display and input drivers.
- Then let's wrap these drivers for micropython, and run lvgl+micropython on stm32
- Then, if you want, we can review some other C code you have and see what it would take to automatically convert it to micropython module with `gen_mpy.py`.

> I built the stm32 port with LVGL binding successfully just now, but unfortunatelly, its size is more than 512KB, what module should I exclude from building?

How much of this 512KB is lvgl library? could you check? How much is micropython?  
You can control what features and objects are included in lvgl library by editing `lv_config.h`. You can disable what you don't need in order to lower program size.

turoksama, 2019-03-31:
That's right.
Here is the map file for the test build:
[firmware.zip](https://github.com/littlevgl/lv_binding_micropython/files/3026664/firmware.zip)

amirgon, 2019-03-31:
> Here is the map file for the test build:

I've analyzed your map file.

|Files|Size (bytes)|Comments
|---|---|---|
|Total|576962|
|Micropython|200837|
|lvgl|228226|
|Other|147899| 
|lv_mpy.o|90092|Included in lvgl|
|frozen_mpy.o|14412|Included in Other|
|extmod|27063|Included in Other|
|emitnthumb.o|11258|Included in Micropython|

Here are some things you can try:

- Remove or reduce frozen modules and external modules in Micropython.
- Set `MICROPY_EMIT_THUMB` and `MICROPY_EMIT_INLINE_THUMB`  to 0 to disable `emitnthumb.o` (see [mpconfigport.h](https://github.com/littlevgl/lv_micropython/blob/75d612796de1ea0b94faa7955664da82a5679c04/ports/stm32/mpconfigport.h#L44))
- Disable some lvgl objects in `lv_config.h`
- Get 1MB flash instead of 512Mb flash for program memory.

138134 bytes are lvgl objects, fonts etc..
You can disable many of them on `lv_conf.h`.
When you disable objects from lvgl, `lv_mpy.o` will also get smaller.
Here is their split:

|Size (bytes)|lvgl object|
|---|---|
|9337 |  lv_fonts/lv_font_symbol_20.o|
|8897 |  lv_fonts/lv_font_dejavu_20.o|
|8229 |  lv_draw/lv_draw_rect.o|
|5581 |  lv_themes/lv_theme_material.o|
|5445 |  lv_objx/lv_ta.o|
|5023 |  lv_objx/lv_page.o|
|4412 |  lv_core/lv_obj.o|
|3810 |  lv_objx/lv_calendar.o|
|3610 |  lv_objx/lv_label.o|
|3588 |  lv_objx/lv_table.o|
|3537 |  lv_draw/lv_draw_img.o|
|3493 |  lv_objx/lv_btnm.o|
|3305 |  lv_objx/lv_tabview.o|
|3177 |  lv_objx/lv_chart.o|
|3149 |  lv_draw/lv_draw_line.o|
|2928 |  lv_objx/lv_ddlist.o|
|2849 |  lv_objx/lv_list.o|
|2656 |  lv_core/lv_indev.o|
|2443 |  lv_objx/lv_canvas.o|
|2389 |  lv_objx/lv_cont.o|
|2382 |  lv_objx/lv_btn.o|
|2362 |  lv_draw/lv_draw_vbasic.o|
|2249 |  lv_objx/lv_roller.o|
|2211 |  lv_objx/lv_tileview.o|
|2096 |  lv_objx/lv_kb.o|
|2075 |  lv_objx/lv_slider.o|
|1917 |  lv_core/lv_style.o|
|1887 |  lv_objx/lv_gauge.o|
|1788 |  lv_objx/lv_win.o|
|1732 |  lv_objx/lv_img.o|
|1685 |  lv_misc/lv_txt.o|
|1632 |  lv_core/lv_refr.o|
|1564 |  lv_core/lv_group.o|
|1430 |  lv_objx/lv_spinbox.o|
|1428 |  lv_objx/lv_bar.o lv_objx/lv_sw.o|
|1428 |  lv_objx/lv_bar.o lv_objx/lv_sw.o|
|1325 |  lv_objx/lv_lmeter.o|
|1321 |  lv_objx/lv_mbox.o|
|1250 |  lv_misc/lv_anim.o|
|1161 |  lv_draw/lv_draw_arc.o|
|1125 |  lv_objx/lv_cb.o|
|1105 |  lv_misc/lv_fs.o|
|1081 |  lv_objx/lv_preload.o lv_objx/lv_line.o|
|1081 |  lv_objx/lv_preload.o lv_objx/lv_line.o|
|1013 |  lv_misc/lv_ufs.o|
|957 |  lv_misc/lv_task.o|
|903 |  lv_draw/lv_draw_label.o|
|893 |  lv_objx/lv_arc.o|
|785 |  lv_draw/lv_draw_triangle.o|
|773 |  lv_objx/lv_led.o|
|705 |  lv_misc/lv_ll.o|
|697 |  lv_objx/lv_imgbtn.o|
|617 |  lv_hal/lv_hal_disp.o|
|571 |  lv_misc/lv_math.o|
|465 |  lv_misc/lv_area.o|
|419 |  lv_misc/lv_font.o|
|413 |  lv_misc/lv_color.o|
|373 |  lv_hal/lv_hal_indev.o|
|343 |  lv_draw/lv_draw.o|
|254 |  lv_misc/lv_mem.o|
|250 |  lv_core/lv_vdb.o|
|141 |  lv_misc/lv_circ.o|
|122 |  lv_hal/lv_hal_tick.o|
|93 |  lv_misc/lv_log.o|
|89 |  lv_fonts/lv_font_builtin.o|
|85 |  lv_themes/lv_theme.o|

turoksama, 2019-04-01:
Thanks for your help.

> Set MICROPY_EMIT_THUMB and MICROPY_EMIT_INLINE_THUMB to 0 to disable emitnthumb.o (see mpconfigport.h)

What is EMIT_THUMB used for?

I'll give a go in later time.

amirgon, 2019-04-01:
> What is EMIT_THUMB used for?

It is used to compile micropythoncode to ARM assembly. This is useful if you have micropython code that must run very fast. It does not affect lvgl binding since lvgl is implemented in C.  
When it is disabled micropythoncode code will always be interpreted instead of compiled.

> By the way, what software do you use to analyze map file? I used amap, I don't think it tells every details.

I have a script for that. I can share it with you later.

turoksama, 2019-04-01:
Oh, I see, that is the viper what they said.
Thank you for sharing.

turoksama, 2019-04-02:
@amirgon 
Thank you!

turoksama, 2019-05-13:
> What C module do you plan to add to micropython, other than lvgl library and lvgl drivers?
>
> I suggest going step by step:
>
> * First let's run lvgl without micropython on stm32, with stm32 display and [...]

@amirgon 
Hi there,

> * First let's run lvgl without micropython on stm32, with stm32 display and input drivers.

I followed your suggestion and finished the first step, would you like to help me with NEXT steps?
Here are my driver in C.
[lv_drv.zip](https://github.com/littlevgl/lv_binding_micropython/files/3173216/lv_drv.zip)

amirgon, 2019-05-13:
Hi @turoksama 
I'm glad to see you made some progress! :+1: 
Just to make sure:
- Were you able to run lvgl successfuly with your code on stm32 + ILI9341 + xpt2046 hardware?
- How did you solve the flash size problem you had? Did you get a larger flash or did you exclude something from the build?

I'll be happy to help you on the next steps.
First, let's commit the code you already have to `lv_binding_micropython/driver/stm32`.

I have a few comments about your code:
- All your code seems specific to ILI9341 and xpt2046 hardware, but the files have generic names such as `lcd.c` and `lv_drv_disp`. This can work if you only have specific display and input device, but is not scalable if we ever want to support additional display/input drivers.  
Therefore - I suggest renaming those files as well as C variables and functions with generic names (that are not defined `static` and are visible to the linker) to **hardware specific names**.  
- There seem to be some redundant files and functions, like `lcd_draw_line`, `lcd_fill_rect` which are not needed for the drivers.
- I noticed that all parameters are **defined as macros** on your code. For example, ports such as `cs`, `dc` etc. are defined on `ili9341.h` and touch ports in `xpt2046.h` as well as threshold and calibration values.  
In order to make this code generic and usable for other users, these parameters need to be set **on runtime** instead of compile-time. When the user initializes the display and touch device in micropython, he provides these parameters (SPI ports, for example) on his python script, and he doesn't need to re-compile the micropython C code when ports change.  
Therefore - I suggest changing all the relevant parameters to be runtime parameters (received as a function arguments) instead of defines.

Next steps are:
- Create a micropython module to wrap the drivers
- Register the module in micropython.

Both of these steps are simple and I will be happy to help you with them.  

Let's start by committing your code to `lv_binding_micropython/driver/stm32` after you fix the problems I mentioned above.
You are welcome to send a PR!

turoksama, 2019-05-15:
Hi,
 It is fully tested and works fine with LVGL.
 I have a question about hardware spi in micropython.
 for example, i want to use hardware spi:
  import pyb
  spi = SPI(1)
but when i use spi.xxx to send/recieve data, it failed and said: sck,miso,mosi is not defined, well then:
  spi = SPI(1,sck=Pin(xx),miso=...)
It use software spi instead!
  I don;t know why.

> *  All your code seems specific to ILI9341 and xpt2046 hardware, but the files have generic names such as lcd.c and lv_drv_disp. This can work if you only have specific display and input device, but is not scalable if we ever want to support additional display/input drivers.
Therefore - I suggest renaming those files as well as C variables and functions with generic names (that are not defined static and are visible to the linker) to hardware specific names.

I will modify these code into runtime and with generic interface like lcd.c, it's hardware independent.

> * There seem to be some redundant files and functions, like lcd_draw_line, lcd_fill_rect which are not needed for the drivers.

I need them for other use, so, it;s not specific to LVGL itself.

> * I noticed that all parameters are defined as macros on your code. For example, ports such as cs, dc etc. are defined on ili9341.h and touch ports in xpt2046.h as well as threshold and calibration values.
In order to make this code generic and usable for other users, these parameters need to be set on runtime instead of compile-time. When the user initializes the display and touch device in micropython, he provides these parameters (SPI ports, for example) on his python script, and he doesn't need to re-compile the micropython C code when ports change.
Therefore - I suggest changing all the relevant parameters to be runtime parameters (received as a function arguments) instead of defines.

I am not so sure how to do about this. would you help me? (not in micropython code style, i think it is not generic for other use.)

amirgon, 2019-05-15:
> It is fully tested and works fine with LVGL.

Great!

> I have a question about hardware spi in micropython.
> for example, i want to use hardware spi:
> import pyb

Micropython has different libraries for stm32 and other boards.  
I'm using ESP32, so I'm not familiar with stm32 SPI configuration.  
I suggest you try asking this question in some forum. You are using pyboard, right? You can send the question to micropython [pyboard forum](https://forum.micropython.org/viewforum.php?f=6)

> I need them for other use, so, it;s not specific to LVGL itself.

Even if you still want to use them with micropython, they should not be part of the display driver.   
Are you sure you can't do the same thing with lvgl API? Is it a functionality that is missing from lvgl?  
Moreover, you can't just write to screen directly when using lvgl, because lvgl will refresh it and delete your changes. You need to do that within the lvgl framework.  

I can suggest choosing one of the following options:

- Do what you need to do with existing lvgl functions
- If lvgl is missing some functionality, discuss this on lvgl forum and send a PR for the new functionality to lvgl.
- Otherwise, add this functionality as a separate micropython module, which is not part of the display driver. This is more difficult because it needs to work with lvgl without conflict.

> I am not so sure how to do about this. would you help me? (not in micropython code style, i think it is not generic for other use.)

Yes.
I'll give you an example how we do this on ESP32.
Usually we define a struct with all the relevant parameters. 

https://github.com/littlevgl/lv_binding_micropython/blob/c910795ac3a32991793187c22e3325b4c27dc80b/driver/esp32/modILI9341.c#L34-L47

In current version we keep a global variable for this struct. (In the next version we will be able to save these parameters per driver instance)

https://github.com/littlevgl/lv_binding_micropython/blob/c910795ac3a32991793187c22e3325b4c27dc80b/driver/esp32/modILI9341.c#L49-L51

When initializing the driver, we set values to each of the parameters.
On ESP32 this is done when creating the micropython object, I can help you write the micropython syntax for creating micropython object, so for now you can simply initialize them with constant values, and we'll change this later.

https://github.com/littlevgl/lv_binding_micropython/blob/c910795ac3a32991793187c22e3325b4c27dc80b/driver/esp32/modILI9341.c#L131-L142

On the SPI initialization code and on the driver implementation (flush,fill,map functions), we use the global variable we defined earlier to access these parameters:

https://github.com/littlevgl/lv_binding_micropython/blob/c910795ac3a32991793187c22e3325b4c27dc80b/driver/esp32/modILI9341.c#L367-L372

Then the function can access `self->some_parameters` to access the driver parameters.  

The point is that these functions **don't access the constant values** (your defines), but **runtime values** instead (the struct parameters).

Does it make things clearer?  
Please let me know if you need more help with it.

turoksama, 2019-05-15:
> * Micropython has different libraries for stm32 and other boards.
I'm using ESP32, so I'm not familiar with stm32 SPI configuration.
I suggest you try asking this question in some forum. You are using pyboard, right? You can send the question to micropython pyboard forum

I'll give it a go later.

> * Even if you still want to use them with micropython, they should not be part of the display driver.
Are you sure you can't do the same thing with lvgl API? Is it a functionality that is missing from lvgl?
Moreover, you can't just write to screen directly when using lvgl, because lvgl will refresh it and delete your changes. You need to do that within the lvgl framework.
I can suggest choosing one of the following options:
Do what you need to do with existing lvgl functions
If lvgl is missing some functionality, discuss this on lvgl forum and send a PR for the new functionality to lvgl.
Otherwise, add this functionality as a separate micropython module, which is not part of the display driver. This is more difficult because it needs to work with lvgl without conflict.

Well,they should be removed for LVGL driver, I understand and agree.

But, I insist to have these functions in my custom file, because users will need them without using LVGL sometimes, for test purpose or study purpose or any others.

It is clearer, I will read your examples for more. Thanks.

closed 2019-06-12 by stale[bot]

turoksama, 2019-06-19:
@amirgon 
Hi there,
 I had a situation about args of mpy recently, would you like to have a look at the issue?
Here I have a very strange behavior like this:
I ported an 18-bit TFT LCD to micropython as type of micropython, which i modified lcd.c to do that.
The color system is 16-bit(RGB565), a 3-byte rgb data can only be accept by the lcd , so, I have to use a buffer to store the expanded color bytes( RGB565 -> RGB888 ) , and then transfer.
For example, I defined a buffer for 8 lines of lcd: uint8_t buf[8*LCD_WIDTH*3].
I tested 16 lines of filling, tranfer twice will do.

Same driver codes in 2 software platforms:

In micropython run: 
the color of the first 8 lines is always not correct with values except 0x0000(BLACK), but the later is correct, for example: color 0xffff should be white, but the first 8 lines will show somewhat grey, the rest shows white. 

**The most tricky thing is the content of the buffer is left untouched again once caculated!**

In naked run: 
**the color of all lines is always correct!**

```
void lcd_dev_fill_rect_mono(uint16_t x0, uint16_t y0, uint16_t x1, uint16_t y1, uint16_t color)
{
	uint32_t size, i;
	uint16_t d, m;
	uint8_t  rgb_buf[8*LCD_WIDTH*3];

	size = (y1 - y0 + 1) * (x1 - x0 + 1);

	lcd_dev_set_window(x0, y0, x1, y1);
	lcd_dev_xfer_cmd(MONITOR_WRITE_PIXELS);

	uint8_t rgb[3];
	uint32_t t;
	rgb[0] = (color >> 8) & 0xF8;
	rgb[1] = (color >> 3) & 0xFC;
	rgb[2] = color << 3;
	for (i = 0; i < 8*LCD_WIDTH; i++)
	{
		rgb_buf[i * 3 + 0] = rgb[0];
		rgb_buf[i * 3 + 1] = rgb[1];
		rgb_buf[i * 3 + 2] = rgb[2];
	}
	d = size/(8*LCD_WIDTH);//block
	m = size%(8*LCD_WIDTH);//rest
	for(i = 0; i < d; i++)
	{
		lcd_dev_xfer_data(rgb_buf, 8*LCD_WIDTH*3);
	}
	lcd_dev_xfer_data(rgb_buf, m*3);
}

/// \method fill(colour)
///
/// Fill the screen with the given colour (0 or 1 for white or black).
///
/// This method writes to the hidden buffer.  Use `show()` to show the buffer.
STATIC mp_obj_t pyb_lcd_fill(size_t n_args, const mp_obj_t *args)
{
    pyb_lcd_obj_t *self = MP_OBJ_TO_PTR(args[0]);
    int x0 = mp_obj_get_int(args[1]);
    int y0 = mp_obj_get_int(args[2]);
    int x1 = mp_obj_get_int(args[3]);
    int y1 = mp_obj_get_int(args[4]);
	int c  = mp_obj_get_int(args[5]);

	//use an uint16_t variable does not help.
//	uint16_t ct;
//	ct = (uint16_t)c;

	/*Truncate the area to the screen*/
	int xt0 = x0 < 0 ? 0 : x0;
	int yt0 = y0 < 0 ? 0 : y0;
	int xt1 = x1 > self->width  ? self->width  : x1;
	int yt1 = y1 > self->height ? self->height : y1;

	/*Return if the area is out the screen*/
	lcd_dev_fill_rect_mono(xt0,yt0,xt1,yt1,c);

	//use an uint16_t variable does not help.
//	lcd_dev_fill_rect_mono(xt0,yt0,xt1,yt1,ct);

	return mp_const_none;
}
STATIC MP_DEFINE_CONST_FUN_OBJ_VAR_BETWEEN(pyb_lcd_fill_obj, 6, 6, pyb_lcd_fill);
```

Micropython test code: (LCD WIDTH = 320)
>>>pyb.lcd.fill(0,0,319,15,0xffff)
Result: Color of first 8 lines is not correct.

>>>pyb.lcd.fill(0,0,319,15,0x00ff)
Result: Color of first 8 lines is not correct.

>>>pyb.lcd.fill(0,0,319,15,0x000)
Result: Color of all lines is correct.

I noticed a configuration in /py/mpconfig.h, lines 108-110:
```
// A MicroPython object is a machine word having the following form:
//  - xxxx...xxx1 : a small int, bits 1 and above are the value
//  - xxxx...xx10 : a qstr, bits 2 and above are the value
//  - xxxx...xx00 : a pointer to an mp_obj_base_t (unless a fake object)
#define MICROPY_OBJ_REPR_A (0)

... ...

#ifndef MICROPY_OBJ_REPR
#define MICROPY_OBJ_REPR (MICROPY_OBJ_REPR_A)
#endif
```

Well, #define MICROPY_OBJ_REPR (MICROPY_OBJ_REPR_C) with no luck...

Stucked here without any help of debug. I don't know how to setup a develop env for mpy with eclipse, so there is almost no way to seeing what is happening inside.