lvgl/lv_binding_micropython #24: How to create a "set" of lv.point_t()

patrickomatik, 2019-06-19:
I'm a python novice, and I am struggling with C-binding use.
Translating the line example from the C docs, I want to draw a line through 5 points

`line1=lv.obj()
line1 = lv.line(lv.scr_act())
line1.set_points(line_points, 5)`

but I can't work out how to correctly instantiate the line_points object? I've tried various combinations of lists, dictionaries and arrays, and the closest I've got is an error "TypeError: can't convert 'lv_point_t' object to str implicitly" when using an array.

I'm really excited about the possibilities lvgl on micropython opens up, especially with the PC emulation. Thanks for a great project.

amirgon, 2019-06-19:
Hi @patrickomatik 
I'm happy to see you are interested in this project!

We have an online (web) simulator, [here is a script that draws a line](https://cdn.statically.io/gh/littlevgl/lv_micropython/c1d321174/ports/javascript/lvgl_editor.html?env=dev&script=https://gist.githubusercontent.com/amirgon/e0e0bf9358fa5c1677b3b338e4e0d8b4/raw/a8e45aa6d808fdbed1ee19349b6642baceaa6eb4/draw_line.py)

The interesting code is 

```python
l = lv.line(scr)
l.set_points([{'x':100,'y':100},{'x':150,'y':100},{'x':150,'y':150}],3)
```

Another option is to create `lv.point_t()` objects, set their `x` and `y` properties, and pass an array of these objects.

**The support for arrays was added on dev-6.0 branch, which will be released soon.**
You can try it [here](https://github.com/amirgon/lv_mpy/tree/dev-6.0/lib).

patrickomatik, 2019-06-19:
Awesome - thanks for the speedy response! Emscripten support rocks.

I've pulled the dev-6.0 branch, but make -C ports/unix fails due to missing the directory ../../lib/lv_bindings/lvgl/src

The src directory also doesn't seem to be present in the lvgl repo - do I need to do something specific to create it? Building the lv_binding main branch behaves as expected.

amirgon, 2019-06-19:
@patrickomatik  Did you run `git submodule update --recursive`?

embeddedt, 2019-06-19:
I think it should be `git submodule update --init --recursive`.

patrickomatik, 2019-06-19:
I cloned a fresh dev-6.0 repo using submodule recursive. @embeddedt 's suggestion seems to have done the trick, thanks!

patrickomatik, 2019-06-19:
Ooops
`GEN build/frozen_mpy.c
Traceback (most recent call last):
  File "../../tools/mpy-tool.py", line 875, in <module>
    main()
  File "../../tools/mpy-tool.py", line 863, in main
    raw_codes = [read_mpy(file) for file in args.files]
  File "../../tools/mpy-tool.py", line 717, in read_mpy
    raise Exception('incompatible .mpy version')
Exception: incompatible .mpy version
make: *** [../../py/mkrules.mk:119: build/frozen_mpy.c] Error 1
`

Suggestions please?

embeddedt, 2019-06-19:
@patrickomatik `make clean` in both the `mpy-cross` and `unix` directories.

patrickomatik, 2019-06-19:
Great. I then get an error mpy-cross executable does not exist, which I fixed by making that directory first.

But the build fails with many, many errors of which this is just the first:

In file included from build/lvgl/lv_mpy.c:20:
../../py/obj.h:638:45: note: expected ‘mp_uint_t’ {aka ‘long unsigned int’} but argument is of type ‘void *’
 mp_obj_t mp_obj_new_int_from_uint(mp_uint_t value);
                                   ~~~~~~~~~~^~~~~

embeddedt, 2019-06-19:
@patrickomatik Please clone a fresh, new repository using these exact instructions and see if it works. I just tested a fresh copy and it worked properly for me.

```c
git clone https://github.com/littlevgl/lv_micropython.git
cd lv_micropython
git checkout dev-6.0
git submodule update --init --recursive
cd mpy-cross
make -j1
cd ../ports/unix
make -j1
```

amirgon, 2019-06-19:
@embeddedt @patrickomatik I think https://github.com/littlevgl/lv_micropython.git is broken.  
I'll fix it soon.
Until then you can try using https://github.com/amirgon/lv_mpy/tree/dev-6.0

embeddedt, 2019-06-19:
> I think https://github.com/littlevgl/lv_micropython.git is broken.

Exactly what is broken? I can clone and build it just fine.

amirgon, 2019-06-19:
> Exactly what is broken? I can clone and build it just fine.

@embeddedt I meant the [dev-6.0 branch](https://github.com/littlevgl/lv_micropython/tree/dev-6.0). There's some mixup with submodule versions. I'm going to fix this soon when I make some commits and also align to latest dev-6.0 of lvgl itself.

amirgon, 2019-06-20:
@embeddedt,
Last night I tried merging my changes with your changes on dev-6.0 branch (your Emscripten port) and ran into trouble.  

- I noticed that you updated Micropython sources from upstream.  
I think it would be better to **rebase** instead of **merge** when you update Micropython upstream. It could make it much easier to understand what changed and differentiate our changes from upstream changes.
- Did you update to some tag/release, or to current `master` of Micropython upstream?  
- After you made your changes, did you try building some other targets besides Emscripten? (the unix port, esp32, stm32 ports?)

embeddedt, 2019-06-20:
> Did you update to some tag/release, or to current `master` of Micropython upstream?

I merged the current `master`. Is that a problem?

> After you made your changes, did you try building some other targets besides Emscripten?

Yes, I can build the 'unix` port just fine on my computer.

amirgon, 2019-06-20:
> Yes, I can build the `unix` port just fine on my computer.

@embeddedt I just tried cloning and building a clean version - it fails.
```
git clone https://github.com/littlevgl/lv_micropython.git
git submodule update --recursive --init
git checkout dev-6.0
cd ports/unix/
make DEBUG=1
```

Then I get
```
make: ../../mpy-cross/mpy-cross: Command not found
../../py/mkrules.mk:112: recipe for target 'build/frozen_mpy/upip.mpy' failed
```

Did I do something wrong? Could you try it on your side?

embeddedt, 2019-06-20:
You need to build the mpy-cross directory first.

amirgon, 2019-06-20:
> You need to build the mpy-cross directory first.

@embeddedt You are right, of course! I forgot this step.
Thanks!

amirgon, 2019-06-20:
> Great. I then get an error mpy-cross executable does not exist, which I fixed by making that directory first.
>
> But the build fails with many, many errors of which this is just the first:
>
> In file [...]

> @embeddedt @patrickomatik I think https://github.com/littlevgl/lv_micropython.git is broken.
> I'll fix it soon.
> Until then you can try using https://github.com/amirgon/lv_mpy/tree/dev-6.0

@patrickomatik **I just pushed a new version to dev-6.0**. Everything should work now.
Please try it out, and let me know if there are any problems.
(don't forget to build mpy-cross first, like I did :wink:)

embeddedt, 2019-06-20:
> don't forget to build mpy-cross first

Just for the record, the reason we were seeing the `mpy-cross` issues is because MicroPython changed the process on their end: https://github.com/micropython/micropython/issues/3255

amirgon, 2019-06-21:
Closing for now.
Please re-open if there is still a problem.

closed 2019-06-21 by amirgon