lvgl/lv_binding_micropython #25: How to pass icon file to list object?

patrickomatik, 2019-06-19:
Hi
Again, working from the C Docs: I'm trying to create a list using the python bindings
`list1 = lv.list(lv.scr_act());
list1.set_size(130, 170);
list1.align(lv.scr_act(),  lv.ALIGN.CENTER, 20, 40);
list1.add(lv.SYMBOL.CLOSE, "Close", lambda action: print("Click on close")or lv.RES.OK);
`
gives me 

SyntaxError: Cannot convert str to pointer!

which I think relates the reference to lv.SYMBOL.CLOSE. 

Trying various other ways of wrapping the parameter seems to suggest the method is expecting an object with a protocol buffer?

What is the expected object type? Am I missing an obvious trick to work this out myself?

Many thanks again

amirgon, 2019-06-20:
@patrickomatik It should work on dev-6.0 branch, but it doesn't [because of a bug](https://github.com/littlevgl/lvgl/issues/1036#issuecomment-504226690).

I'll let you know once this is solved.

amirgon, 2019-06-21:
@patrickomatik Solved on dev-6.0 branch.
Try this (on dev-6.0 branch):

```python
list1 = lv.list(lv.scr_act());
list1.set_size(130, 170);
list1.align(lv.scr_act(),  lv.ALIGN.CENTER, 20, 40);
btn_open = list1.add_btn(lv.SYMBOL.OK, "Open")
btn_open.set_event_cb(lambda obj,  event: event != lv.EVENT.CLICKED or print("Click on open"));
btn_close = list1.add_btn(lv.SYMBOL.CLOSE, "Close")
btn_close.set_event_cb(lambda obj,  event: event != lv.EVENT.CLICKED or print("Click on close"));
```

Closing this for now.
Feel free to re-open if you see any problem.

closed 2019-06-21 by amirgon