lvgl/lv_binding_micropython #26: Generic issues porting C examples

patrickomatik, 2019-06-19:
I've worked through the C samples, and created a python equivalent for each.
The majority work as expected; 

- references to images above is problematic wherever they are used (see my prev issue)
- some constants seem to be missing e.g. OPA_70, LV_COLOR_X (or I have the incorrect form)
- Message_box.add_btns fails with a typerror: object with buffer protocol required, passing a list of strings for the button pair
- I can't get the chart example to run without it seg-faulting (in the PC emulator)

The rest of the controls all seem to work very well, really impressive project.

Maybe these python code snippets might be useful for a wider audience? In which case you're very welcome to them!

amirgon, 2019-06-19:
> All against master; will repeat for dev-6.0 once I have built it out.

@patrickomatik There are a lot of changes on dev-6.0. I really recommend testing everything there. Many bugs were fixed, callbacks were redesigned completely, added support for arrays, added more conversions etc.  
I've added image support on dev-6.0 - [here is an example!](https://cdn.statically.io/gh/littlevgl/lv_micropython/c1d321174/ports/javascript/lvgl_editor.html?env=dev&script=https://gist.githubusercontent.com/amirgon/764a2f14f403d62d46ca61d7e0546042/raw/f963e607d73f96ba793caaf2287e4ea3f7728562/lv_img_example1.py)

And also chart with animation: [Example](https://cdn.statically.io/gh/littlevgl/lv_micropython/c1d321174/ports/javascript/lvgl_editor.html?env=dev&script=https://gist.githubusercontent.com/amirgon/7f2a0702932a8d08611241d10c1f0e57/raw/2fdda0882ad3372e8746f4d7af3c05fc5f815137/advanced_demo.py) (see the "Chart" tab)

> Maybe these python code snippets might be useful for a wider audience? In which case you're very welcome to them!

Yes I think they would be very useful!  Thanks!
But we will need to adapt them to v6.0.

@kisvegabor is working on the documentation of dev-6.0 and [we discussed](https://github.com/littlevgl/lvgl/issues/1099#issuecomment-502372439) about adding some python examples. This could be a good starting point!

Did you try running your examples on the web simulator? (Emscripten)

amirgon, 2019-06-21:
@patrickomatik 

> * references to images above is problematic wherever they are used (see my prev issue)

Images are supported on dev-6.0. You can either use them directly, or through image decoder.  
I also added PNG decoder support.
Here are two examples:
- [Displaying an image stored in an array](https://github.com/littlevgl/lv_binding_micropython/blob/43d80431758f7b2640164b76c315f277730606ce/examples/example1.py)
- [Decoding PNG, using the image decoder](https://github.com/littlevgl/lv_binding_micropython/blob/43d80431758f7b2640164b76c315f277730606ce/examples/example2.py)

> * some constants seem to be missing e.g. OPA_70, LV_COLOR_X (or I have the incorrect form)

In general, the Micropython binding converts enums, but not defines (because it works on preprocessed files).  
I did some work to convert defines to enums wherever this is possible. On dev-6.0 you can use `lv.OPA`, for example.  
Converting colors is more difficult because they can't be converted to enum. It's also less a priority right now (you can use `lv.color_hex` instead), but maybe I'll get to that in the future.

> * Message_box.add_btns fails with a typerror: object with buffer protocol required, passing a list of strings for the button pair

Should be ok on dev-6.0 branch

> * I can't get the chart example to run without it seg-faulting (in the PC emulator)

Please try this on dev-6.0. Let me know if it's seg-faulting, or if you see any other problem.

> Maybe these python code snippets might be useful for a wider audience? In which case you're very welcome to them!

If you have some working examples and you want to contribute them, we'll be happy to use them!  
You can open a PR and add them to the project.

---

Closing for now.
Please feel free to open if you see any problem on the dev-6.0 branch!

closed 2019-06-21 by amirgon