lvgl/lv_binding_micropython #27: Need help about TFT driver issue.

turoksama, 2019-06-20:
@amirgon 
Hi there,
 I had a situation about args of mpy(maybe) recently, any suggestions?
Here I have a very strange behavior like this:
I ported an 18-bit TFT LCD to micropython as type of micropython, which i modified lcd.c to do that.
The color system is 16-bit(RGB565), a 3-byte rgb data can only be accept by the lcd , so, I have to use a buffer to store the expanded color bytes( RGB565 -> RGB888 ) , and then transfer.
For example, I defined a buffer for 8 lines of lcd: uint8_t buf[8 *LCD_WIDTH *3].
I tested 16 lines of filling, tranfer twice will do.

Same driver codes in 2 software platforms:

In micropython run: 
the color of the first 8 lines is always incorrect with values except 0x0000(BLACK), but the later is correct, for example: color 0xffff should be white, but the first 8 lines will show somewhat grey, the rest shows white. 

**The most tricky thing is the content of the buffer is left untouched again once caculated!**

In naked run: 
**the color of all lines is always correct!**

```
void lcd_dev_fill_rect_mono(uint16_t x0, uint16_t y0, uint16_t x1, uint16_t y1, uint16_t color)
{
	uint32_t size, i;
	uint16_t d, m;
	uint8_t  rgb_buf[8*LCD_WIDTH*3];

	size = (y1 - y0 + 1) * (x1 - x0 + 1);

	lcd_dev_set_window(x0, y0, x1, y1);
	lcd_dev_xfer_cmd(MONITOR_WRITE_PIXELS);

	uint8_t rgb[3];
	rgb[0] = (color >> 8) & 0xF8;
	rgb[1] = (color >> 3) & 0xFC;
	rgb[2] = color << 3;
	for (i = 0; i < 8*LCD_WIDTH; i++)
	{
		rgb_buf[i * 3 + 0] = rgb[0];
		rgb_buf[i * 3 + 1] = rgb[1];
		rgb_buf[i * 3 + 2] = rgb[2];
	}
	d = size/(8*LCD_WIDTH);//block
	m = size%(8*LCD_WIDTH);//rest
	//repeated blocks
	for(i = 0; i < d; i++)
	{
		lcd_dev_xfer_data(rgb_buf, 8*LCD_WIDTH*3);
	}
	//rest
	lcd_dev_xfer_data(rgb_buf, m*3);
}

/// \method fill(colour)
///
/// Fill the screen with the given colour (0 or 1 for white or black).
///
/// This method writes to the hidden buffer.  Use `show()` to show the buffer.
STATIC mp_obj_t pyb_lcd_fill(size_t n_args, const mp_obj_t *args)
{
    pyb_lcd_obj_t *self = MP_OBJ_TO_PTR(args[0]);
    int x0 = mp_obj_get_int(args[1]);
    int y0 = mp_obj_get_int(args[2]);
    int x1 = mp_obj_get_int(args[3]);
    int y1 = mp_obj_get_int(args[4]);
	int c  = mp_obj_get_int(args[5]);

	//use an uint16_t variable does not help.
//	uint16_t ct;
//	ct = (uint16_t)c;

	/*Truncate the area to the screen*/
	int xt0 = x0 < 0 ? 0 : x0;
	int yt0 = y0 < 0 ? 0 : y0;
	int xt1 = x1 > self->width  ? self->width  : x1;
	int yt1 = y1 > self->height ? self->height : y1;

	/*Return if the area is out the screen*/
	lcd_dev_fill_rect_mono(xt0,yt0,xt1,yt1,c);

	//use an uint16_t variable does not help.
//	lcd_dev_fill_rect_mono(xt0,yt0,xt1,yt1,ct);

	return mp_const_none;
}
STATIC MP_DEFINE_CONST_FUN_OBJ_VAR_BETWEEN(pyb_lcd_fill_obj, 6, 6, pyb_lcd_fill);
```

Micropython test code: (LCD WIDTH = 320)
```>>>pyb.lcd.fill(0,0,319,15,0xffff)```
Result: Color of first 8 lines is incorrect.

```>>>pyb.lcd.fill(0,0,319,15,0x00ff)```
Result: Color of first 8 lines is incorrect.

```>>>pyb.lcd.fill(0,0,319,15,0x000)```
Result: Color of all lines is correct.

I noticed a configuration in /py/mpconfig.h, lines 108-110:
```
// A MicroPython object is a machine word having the following form:
//  - xxxx...xxx1 : a small int, bits 1 and above are the value
//  - xxxx...xx10 : a qstr, bits 2 and above are the value
//  - xxxx...xx00 : a pointer to an mp_obj_base_t (unless a fake object)
#define MICROPY_OBJ_REPR_A (0)

... ...

#ifndef MICROPY_OBJ_REPR
#define MICROPY_OBJ_REPR (MICROPY_OBJ_REPR_A)
#endif
```

Well, #define MICROPY_OBJ_REPR (MICROPY_OBJ_REPR_C) with no luck...

Stucked here without any help of debug. I don't know how to setup a develop env for mpy with eclipse, so there is almost no way to seeing what is happening inside.

amirgon, 2019-06-20:
@turoksama Once `pyb_lcd_fill` is called, it's all C. 
I don't think the problem is related to micropython at all.  

Can you print out stuff with printf?
Output would usually go through UART, depending on your platform.  

Please try the following:

- Add printf on `lcd_dev_fill_rect_mono` to print all the parameters it recieves. This will show you what parameters Micropython actually sends to your function, and you'll be able to see if they are equal to what you are sending on the python script.
- Try writing pure C code (without Micropython) that calls `lcd_dev_fill_rect_mono` with the exact same parameters. 

My guess is that you'll see the same problem on pure-C program, without any Micropython involved.

turoksama, 2019-06-21:
@amirgon 
Unfortunately, there is no such problem on pure-C program, without any Micropython involved.
I tested in pure-C, it worked fine, and then ported to micropython.
I will try to print all variables in micropython to check...

turoksama, 2019-06-21:
@amirgon 
It is something wrong with the UART too, I intend to send 5 args(10 bytes) by using uart, well, it only sends 3 bytes, what? seriously?

turoksama, 2019-06-23:
Add printf on ```lcd_dev_fill_rect_mono``` , all the parameters it recieves are correct. 
And actually sends to ```lcd_dev_fill_rect_mono ``` .

amirgon, 2019-06-23:
@turoksama Are you using this **with LittlevGL**?
You are not supposed to write directly to screen when using LittlevGL, you must go through LittlevGL high level routines.  
If you do that, both LittlevGL driver and your routine would access the screen and interfere each other.

turoksama, 2019-06-23:
Hello there,
I test the driver standalone before LVGL. This is very tricky with micropython...

amirgon, 2019-06-23:
@turoksama I don't see how Micropython is related here. 
You saw (with the `printf ` you added) that your `lcd_dev_fill_rect_mono` routine is called with the correct parameters, but doesn't do what it should. Micropython is not involved here, and you say LittlevGL neither.

turoksama, 2019-06-23:
Yes, I dont know why this happened, my test:
The driver works in standalone pure-C run,
doesn't work correctly in micropython, so, i don't know what is going on with mpy, or confilt with the driver.

amirgon, 2019-06-23:
@turoksama 
If the **exact** function call **with the same parameters** is working on pure-C but not working with Micropython, I would guess that **your function and Micropython might have some resource contention**. For example both try to access SPI, or you may be out of RAM, or maybe with Micropython your code was forced to use different type of RAM (non DMA-able for example).

Anyway, at this point I can suggest:
- Get a debugger hardware (JTAG) and debug your live system
- Send a question to the [Micropython Forum](https://forum.micropython.org/). Your problem seems unrelated to LittlevGL, and hopefully someone there could offer some help. You would definitely reach a wider audience by posting there instead of here.

turoksama, 2019-06-23:
Thanks for your help.
I tried to send a question to MPY forum, and doesnot get extra help.
I think a live debug should be helpful.

turoksama, 2019-06-26:
@amirgon 
Hello there
 A bug in STM SPI HAL with spi re-init in micropython is found and fixed.
If you re-init spi within micropython,  __HAL_SPI_ENABLE(&spi) should followed.
It seems missed configured in HAL_SPI_TANSMIT:
```
  /* Check if the SPI is already enabled */
  if((hspi->Instance->CR1 & SPI_CR1_SPE) != SPI_CR1_SPE)
  {
    /* Enable SPI peripheral */
    __HAL_SPI_ENABLE(hspi);
  }
```

closed 2019-06-26 by turoksama