lvgl/lv_binding_micropython #28: [dev-6.0] Fix ESP32 port

amirgon, 2019-06-24:
Multiple issues need to be fixed on the ESP32 port

- [ ] Fix build errors
- [ ] Fix indev driver
- [ ] Fix display driver

tianzhengtong, 2019-06-25:
[image]

amirgon, 2019-06-26:
@tianzhengtong  It's working now.

**You must register a display driver** before you try to creating an lvgl object (since lv_create_* would return NULL otherwise).

To build it for ESP32 + ILI9341:
`make -C ports/esp32 -j4 LV_CFLAGS="-DLV_COLOR_DEPTH=16"`

Here is how a display driver is registered:
https://github.com/littlevgl/lv_binding_micropython/blob/9bd17ae15595fb62375afa8284d1b23859ae0abc/examples/chart_anim_example.py#L40-L56

You can try running [chart_anim_example.py](https://github.com/littlevgl/lv_binding_micropython/blob/9bd17ae15595fb62375afa8284d1b23859ae0abc/examples/chart_anim_example.py)

amirgon, 2019-06-28:
@tianzhengtong I'm closing this for now.
If you experience more problems, please feel free to open an issue!

closed 2019-06-28 by amirgon