lvgl/lv_binding_micropython #32: lv_micropython build error

turoksama, 2019-06-29:
@amirgon 
Hi there,
I built dev_6.0 recently with errors like these:
```
In file included from ../../lib/lv_bindings/lvgl/src/lv_misc/lv_task.c:16:0:
../../lib/lv_bindings/lvgl/src/lv_misc/lv_task.c: In function 'lv_task_core_init':
../../py/mpstate.h:260:40: error: 'mp_state_vm_t {aka struct _mp_state_vm_t}' has no member named '_lv_task_ll'
 #define MP_STATE_VM(x) (mp_state_ctx.vm.x)
```

Is there anything wrong with gen_mpy.py perhaps?

embeddedt, 2019-06-29:
@turoksama Have you run `git pull` and `git submodule update --init --recursive`?

turoksama, 2019-06-29:
@embeddedt 
I used micropython latest version v1.11 combined with latest lvgl dev-6.0,  compaired with v5.3 and get errors like those below:

```
CC ../../lib/lv_bindings/lvgl/src/lv_themes/lv_theme_night.c
CC ../../lib/lv_bindings/lvgl/src/lv_themes/lv_theme_templ.c
CC ../../lib/lv_bindings/lvgl/src/lv_themes/lv_theme_default.c
CC ../../lib/lv_bindings/lvgl/src/lv_themes/lv_theme_mono.c
CC ../../lib/lv_bindings/lvgl/src/lv_themes/lv_theme_nemo.c
CC ../../lib/lv_bindings/lvgl/src/lv_themes/lv_theme_material.c
CC ../../lib/lv_bindings/lvgl/src/lv_themes/lv_theme_zen.c
CC ../../lib/lv_bindings/lvgl/src/lv_themes/lv_theme.c
CC ../../lib/lv_bindings/lvgl/src/lv_themes/lv_theme_alien.c
CC ../../lib/lv_bindings/lvgl/src/lv_misc/lv_task.c
CC ../../lib/lv_bindings/lvgl/src/lv_misc/lv_anim.c
CC ../../lib/lv_bindings/lvgl/src/lv_misc/lv_fs.c
CC ../../lib/lv_bindings/lvgl/src/lv_hal/lv_hal_disp.c
CC ../../lib/lv_bindings/lvgl/src/lv_hal/lv_hal_indev.c
CC ../../lib/lv_bindings/lvgl/src/lv_core/lv_refr.c
CC ../../lib/lv_bindings/lvgl/src/lv_core/lv_indev.c
CC ../../lib/lv_bindings/lvgl/src/lv_core/lv_obj.c
CC ../../lib/lv_bindings/lvgl/src/lv_core/lv_group.c
CC ../../lib/lv_bindings/lvgl/src/lv_core/lv_disp.c
CC ../../lib/lv_bindings/lvgl/src/lv_core/lv_style.c
CC ../../lib/lv_bindings/lvgl/src/lv_draw/lv_draw_arc.c
CC ../../lib/lv_bindings/lvgl/src/lv_draw/lv_img_cache.c
CC ../../lib/lv_bindings/lvgl/src/lv_draw/lv_draw_img.c
CC ../../lib/lv_bindings/lvgl/src/lv_draw/lv_img_decoder.c
CC ../../lib/lv_bindings/lvgl/src/lv_draw/lv_draw.c
CC ../../lib/lv_bindings/lvgl/src/lv_draw/lv_draw_basic.c
CC ../../lib/lv_bindings/lvgl/src/lv_draw/lv_draw_label.c
CC ../../lib/lv_bindings/lvgl/src/lv_draw/lv_draw_line.c
CC ../../lib/lv_bindings/lvgl/src/lv_draw/lv_draw_triangle.c
CC ../../lib/lv_bindings/lvgl/src/lv_draw/lv_draw_rect.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_sw.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_kb.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_arc.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_tileview.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_bar.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_gauge.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_calendar.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_roller.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_spinbox.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_ta.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_canvas.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_label.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_chart.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_img.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_slider.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_cont.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_cb.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_page.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_btnm.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_btn.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_tabview.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_ddlist.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_line.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_mbox.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_objx_templ.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_win.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_list.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_preload.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_lmeter.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_imgbtn.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_led.c
CC ../../lib/lv_bindings/lvgl/src/lv_objx/lv_table.c
CC ../../lib/lv_bindings/lvgl/src/lv_font/lv_font_fmt_txt.c
CC ../../lib/lv_bindings/lvgl/src/lv_font/lv_font_roboto_28.c
CC ../../lib/lv_bindings/lvgl/src/lv_font/lv_font_roboto_22.c
CC ../../lib/lv_bindings/lvgl/src/lv_font/lv_font_roboto_12.c
CC ../../lib/lv_bindings/lvgl/src/lv_font/lv_font_roboto_16.c
CC ../../lib/lv_bindings/lvgl/src/lv_font/lv_font.c
CC ../../lib/lv_bindings/lvgl/src/lv_font/lv_font_unscii_8.c
CC ../../lib/lv_bindings/driver/generic/modlvindev.c
CC build-eBRAINTINY2/lvgl/lv_mpy.c
CC build-eBRAINTINY2/lodepng/lodepng.c
CC build-eBRAINTINY2/lodepng/mp_lodepng.c
CC build-eBRAINTINY2/pins_eBRAINTINY2.c
CC build-eBRAINTINY2/frozen_mpy.c
In file included from ../../py/runtime.h:29:0,
                 from build-eBRAINTINY2/lvgl/lv_mpy.c:22:
build-eBRAINTINY2/lvgl/lv_mpy.c: In function 'lv_log_print_g_cb_t_print_cb_callback':
../../py/mpstate.h:260:40: error: 'mp_state_vm_t {aka struct _mp_state_vm_t}' has no member named 'mp_lv_user_data'
 #define MP_STATE_VM(x) (mp_state_ctx.vm.x)
                                        ^
./mpconfigport.h:292:23: note: in expansion of macro 'MP_STATE_VM'
 #define MP_STATE_PORT MP_STATE_VM
                       ^~~~~~~~~~~
build-eBRAINTINY2/lvgl/lv_mpy.c:21300:59: note: in expansion of macro 'MP_STATE_PORT'
     mp_obj_t callbacks = get_callback_dict_from_user_data(MP_STATE_PORT(mp_lv_user_data));
                                                           ^~~~~~~~~~~~~
build-eBRAINTINY2/lvgl/lv_mpy.c: In function 'mp_lv_log_register_print_cb':
../../py/mpstate.h:260:40: error: 'mp_state_vm_t {aka struct _mp_state_vm_t}' has no member named 'mp_lv_user_data'
 #define MP_STATE_VM(x) (mp_state_ctx.vm.x)
                                        ^
./mpconfigport.h:292:23: note: in expansion of macro 'MP_STATE_VM'
 #define MP_STATE_PORT MP_STATE_VM
                       ^~~~~~~~~~~
build-eBRAINTINY2/lvgl/lv_mpy.c:21313:125: note: in expansion of macro 'MP_STATE_PORT'
     void *print_cb = mp_lv_callback(args[0], &lv_log_print_g_cb_t_print_cb_callback, MP_QSTR_lv_log_print_g_cb_t_print_cb, &MP_STATE_PORT(mp_lv_user_data));
                                                                                                                             ^~~~~~~~~~~~~
make: *** [../../py/mkrules.mk:47：build-eBRAINTINY2/build-eBRAINTINY2/lvgl/lv_mpy.o] error 1
make: *** waiting for unfinished jobs....

```

It seems lvgl-build passed...

turoksama, 2019-06-29:
Well, it seems that I missed one line here in mpconfigport.h:

```
#define MICROPY_PORT_ROOT_POINTERS \
    LV_ROOTS \
    **void *mp_lv_user_data; \**
    const char *readline_hist[8]; \
    mp_obj_t machine_pin_irq_handler[40]; \
    struct _machine_timer_obj_t *machine_timer_obj_head; \

```
It is all fine now.

amirgon, 2019-06-29:
> It is all fine now.

Good to hear the problem is solved!
Closing.

closed 2019-06-29 by amirgon