lvgl/lv_binding_micropython #33: maybe MP_STATE_PORT should replaced with LV_GC_ROOT?

Neutree, 2019-07-02:
maybe `MP_STATE_PORT` should replaced with `LV_GC_ROOT` in generated file `lv_mpy.c`

for somone don't want to place `LV_ROOTS` in `mp_state_vm_t`, the struct is too big!

amirgon, 2019-07-02:
@Neutree Please explain what you mean.  

MP_STATE_PORT is the way to access a global variable which is a gc root. (see `MICROPY_PORT_ROOT_POINTERS`)  

We must place `LV_ROOTS` in `mp_state_vm_t` beacuse gc must be aware of lvgl gc roots.
LV_ROOTS is around 100 bytes total - is that big?

Neutree, 2019-07-02:
I just mean we defined `LV_GC_ROOT(x)` in `lv_confg.h` 
```
#  define LV_GC_ROOT(x) MP_STATE_PORT(x)`)
```
and we should use `LV_GC_ROOT` instead of `MP_STATE_PORT` to get lv gc root,

It's not a big problem, thanks for your rapidly reply, feel free to close this issue

amirgon, 2019-07-27:
`lv_binding_micropython` can be used to create Micopython modules for other modules than lvgl. (used today for lodepng and esp-idf). In such case there might be no `LV_GC_ROOT`, but we still need `MP_STATE_PORT` to access `mp_lv_user_data` on `MICROPY_PORT_ROOT_POINTERS`.   

Maybe naming should be changed to remove `lv_` when converting other modules than lvgl.    

Closing.

closed 2019-07-27 by amirgon