lvgl/lv_binding_micropython #34: Missing file after merge

amirgon, 2019-07-02:
@embeddedt Please have a look at commit 6574e2960

```
$ git show 6574e2960
commit 6574e29602a9d6016f018a92bcc188616f524ae4
Merge: 7ead886 63da850
Author: Themba Dube <embeddedthemba@gmail.com>
Date:   Tue Jun 4 12:09:54 2019 -0400

    Merge Emscripten patches into dev-6.0

```

On this commit, you merged two parents, and also **deleted a file**

```
$ git diff 63da850 6574e2960 -- driver/esp32/modxpt2046.c
diff --git a/driver/esp32/modxpt2046.c b/driver/esp32/modxpt2046.c
deleted file mode 100644
index ff829e2..0000000
--- a/driver/esp32/modxpt2046.c
+++ /dev/null
```

This file, `modxpt2046.c` is now missing.  
Any idea how this happened?

If this is a mistake - please don't rush into fixing it.  
I would like to make sure that there are no additional problems that I didn't identify yet.

embeddedt, 2019-07-02:
Strange. I never intended to delete that file.

amirgon, 2019-07-02:
> Strange. I never intended to delete that file. I also don't see how I could have deleted a file while resolving merge conflicts (if there were any).

@embeddedt Maybe we can try to re-run the merge, to see if there are other problems.
Here is some explanation about how to do it: https://superuser.com/a/1219061/1041458

embeddedt, 2019-07-02:
@amirgon By using the `git log -m --name-status` command we can see exactly what the merge changed/added/deleted without having to re-run it. It shows that only the `driver/esp32/modxpt2046.c` was deleted.

```
commit 6574e29602a9d6016f018a92bcc188616f524ae4 (from 7ead88632eff06ea268e6ba830496dad93ccd067)
Merge: 7ead886 63da850
Author: Themba Dube <embeddedthemba@gmail.com>
Date:   Tue Jun 4 12:09:54 2019 -0400

    Merge Emscripten patches into dev-6.0

M       driver/SDL/SDL_monitor.c
M       driver/SDL/modSDL.c

commit 6574e29602a9d6016f018a92bcc188616f524ae4 (from 63da8504ab134d2452238a3bdc4e71f8805be833)
Merge: 7ead886 63da850
Author: Themba Dube <embeddedthemba@gmail.com>
Date:   Tue Jun 4 12:09:54 2019 -0400

    Merge Emscripten patches into dev-6.0

M       .gitignore
M       README.md
M       driver/SDL/SDL_monitor.c
M       driver/SDL/SDL_monitor.h
M       driver/SDL/SDL_mouse.c
M       driver/SDL/SDL_mouse.h
M       driver/SDL/lv_drv_conf.h
M       driver/SDL/modSDL.c
M       driver/esp32/modILI9341.c
D       driver/esp32/modxpt2046.c
M       driver/generic/modlvindev.c
M       driver/include/common.h
M       driver/linux/modfb.c
R073    gen/advanced_demo.py    examples/advanced_demo.py
A       examples/example1.py
M       gen/gen_mpy.py
M       gen/lv_mpy_example.c
M       lvgl
M       pycparser
```

amirgon, 2019-07-02:
@embeddedt Re-running the merge is the simplest way to restore a file with its commit history.

embeddedt, 2019-07-02:
@amirgon The problem is that I think we have more commits added on both branches. I don't understand how I can redo the merge while preserving future commit history, since that's not how Git works.

amirgon, 2019-07-02:
> I don't understand how I can redo the merge while preserving future commit history

@embeddedt Did you take a look at https://superuser.com/a/1219061/1041458 ?
I think they show there exactly what we need.

embeddedt, 2019-07-02:
@amirgon Yes, but that page shows the case where the merge was just completed, with **no further commits.**

If I tried to run those instructions now, wouldn't I end up discarding all commits past `Merge Emscripten patches into dev-6.0`?

amirgon, 2019-07-02:
> If I tried to run those instructions now, wouldn't I end up discarding all commits past `Merge Emscripten patches into dev-6.0`?

@embeddedt I don't think so, git is smarter than this. After you commit to `merge_fix` branch, git knows it's in the history according to its parents. When you try to merge `merge_fix` to master, I don't think it would discard newer commits.  

We can try and see! no harm is done until you "push" something. You can always delete your local repo and start over.

embeddedt, 2019-07-02:
@amirgon I see. I was misunderstanding the general approach.

I think it worked properly. Can you please check the results on the `merge_fix` branch? If everything's fine I will merge to `master`.

amirgon, 2019-07-03:
> I think it worked properly. Can you please check the results on the `merge_fix` branch? If everything's fine I will merge to `master`.

@embeddedt I think this is ok - you can merge it.

embeddedt, 2019-07-03:
@amirgon I've merged it. The issue should be solved now.

Still not sure exactly how I managed to delete it, but my apologies for the inconvenience.

closed 2019-07-03 by embeddedt