lvgl/lv_binding_micropython #35: Should arm linux use this or pylvgl?

XQDD, 2019-07-04:
If yes, which one need  I port, Unix or minimal?

amirgon, 2019-07-04:
> Should arm linux use this or pylvgl?  

@XQDD - That depends if you want to use Python or Micropython. These are two different Python interpreters. You should select the one that best suits your needs.   

btw, lvgl latest version (v6.0) is supported on Micropython (lv_micropython), however, I'm not sure if it's fully supported on pylvgl. Perhaps @rreilink could provide more information on that.

> If lv_binding_micropython, which one need I port, Unix or minimal?

You should use a port that matches your architecture.  
You can use the [unix port](https://github.com/micropython/micropython#the-unix-version) if you are using Linux on either x86, ARM, MIPS etc.  
On bare metal you should use some other port.
 
This question is more related to Micropython than LittlevGL, I suggest you consult with Micropython docs and forum.

XQDD, 2019-07-04:
ok, I got it. Thanks.

closed 2019-07-04 by XQDD