lvgl/lv_binding_micropython #36: How to compile espidf module?

sEpt0r, 2019-07-18:
I tried to use the pure python display driver ili9341.py, but I received the error `ImportError: no module named 'espidf'`.
Then I tried to make esp32 with this module `driver/esp32/espidf.c`, but it failed with `extmod/extmod.mk` and with 

ports/esp32/Makefile
```diff
ifeq ($(MICROPY_PY_LVGL), 1)
 LIB_SRC_C += \
+       lib/lv_bindings/driver/esp32/espidf.c \
        lib/lv_bindings/driver/esp32/modlvesp32.c \
        lib/lv_bindings/driver/esp32/modILI9341.c \
```

ports/esp32/mpconfigport.h
```diff
 extern const struct _mp_obj_module_t mp_module_xpt2046;
+extern const struct _mp_obj_module_t mp_module_espidf;
 extern const struct _mp_obj_module_t mp_module_rtch;

     { MP_OBJ_NEW_QSTR(MP_QSTR_xpt2046), (mp_obj_t)&mp_module_xpt2046 }, \
+    { MP_OBJ_NEW_QSTR(MP_QSTR_espidf), (mp_obj_t)&mp_module_espidf }, \
     { MP_OBJ_NEW_QSTR(MP_QSTR_rtch), (mp_obj_t)&mp_module_rtch }, \
```

The error is:
```build/py/objmodule.o:(.rodata.mp_builtin_module_weak_links_table+0xa4): undefined reference to `mp_module_espidf'```

Could you help me to use it with micropython?

amirgon, 2019-07-18:
Hi @sEpt0r !  

The pure python display driver for ili9341 is very much a work in progress.
It's an experiment to see whether Micropython is fast enough to handle the display without display-specific C code (it only uses esp-idf and lvgl functions).  

I've committed some fixes, please try to pull the latest changes, it should work now.  

I've also added a "hybrid" demo mode, where all management and initialization are in micropython and only the critical "flush" and ISR code are written in C. I observed significant improvement with the hybrid mode (30FPS pure micropython vs. 50FPS hybrid, ~52K pixels per frame (~70% of the screen is refreshed every frame)). With C the limiting factor is SPI, so I believe even higher frame-rates are possible (by using IOMUX pins with 80MHz instead of 40MHz) but current rates are probably good enough.

Here is an example how to build it in hybrid mode: 
```
make -C ports/esp32 -j4 ESPIDFMOD_CFLAGS="-DENABLE_ILI9341_DEMO_FLUSH=1" LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" DEBUG=0 SDKCONFIG=boards/sdkconfig.spiram deploy
```

(remove the `ESPIDFMOD_CFLAGS="-DENABLE_ILI9341_DEMO_FLUSH=1"` to build it in pure micropython mode)

sEpt0r, 2019-07-19:
It was successfully built. I'll upload and check it out later.
Thank you.

sEpt0r, 2019-07-22:
I checked it. The display driver ili9341 works well with pure python and modILI9341.c.
But both drivers hang after several touches when modxpt2046.c is used on the same SPI interface.
I tried to disable half duplex and change the clock frequency on the ILI and XPT devices, but could not achieve stable work.

amirgon, 2019-07-22:
Thanks for the update @sEpt0r!

Please have a look at #21 if you haven't yet.
I don't have xpt2046 hardware so I wrote that code "blindly" and let others test it. So I assume there may still be issues there.  

One thing you can do it to dedicate a different SPI device for the display and for the touch sensor. ESP32 has two SPI devices, HSPI and VSPI (actually more, but these two are usually available for the end user).  
Current driver assumes that both touch and display are on the same SPI device, but you can try to modify it, or you can write a pure python driver for xpt2046.  

I'll be happy to help if you have any questions!

sEpt0r, 2019-07-23:
Thank you.
I read the #21 and tried everything out of the conversation before I opened this issue.
I'll try again with different SPI hosts and write about the results.

sEpt0r, 2019-08-06:
I cannot add the xpt2046 to the VSPI, it fails on the `spi_bus_add_device`.
```    ret=spi_bus_add_device(self->spihost, &devcfg, &self->spi);
    if (ret != ESP_OK) nlr_raise(
        mp_obj_new_exception_msg(&mp_type_RuntimeError, "Failed adding SPI device"));
```

I tried to set full-duplex and 1MHz on both ili and xpt devices, but it still hangs after the first touch.
I checked signals on the oscilloscope and CS pin is low after hang.

amirgon, 2019-08-06:
@sEpt0r Did you try the [esp32_ili9341 example project](https://github.com/littlevgl/esp32_ili9341)?  
It uses HSPI for the ILI9341 display and VSPI for XPT2046 (without Micropython).  

Please try and see if that example is working for you.

sEpt0r, 2019-08-06:
Ok, I'll try it.
Thank you.

amirgon, 2019-08-11:
Let's continue this discussion on #37
Closing.

closed 2019-08-11 by amirgon