lvgl/lv_binding_micropython #37: XPT2046 input device driver in LVGL 6.x

gersch07, 2019-08-11:
@amirgon:
First of all, thanks for keeping this repo updated and sorry for opening another issue about the XPT2046 driver!

After some time of absence I pulled the most recent version and found that your code in ussue #21 (to update a label with the current touch position) doesn't work anymore. As in LVGL 5.3, before compiling I had to set `flags=0` to avoid SPI_HALF_DUPLEX errors; in contrast to LVGL 5.3, where I can get reasonable coordinates with identical wiring, I'm now only getting "-1:-1" all the time. Has there been any major change in syntax (additional to `indev_next` -> `indev_get_next`)? Thanks for your help!

sEpt0r, 2019-08-11:
Hi!
I have the same issue and am trying to solve it in #36 , but so far without success.

amirgon, 2019-08-11:
@gersch07, @sEpt0r Did you try the esp32_ili9341 with v.6 as I suggested on #36 ?
If this example works well, we can use it as a reference and try to eliminate the differences.  

I would need your help because I don't have the hardware to test this.

sEpt0r, 2019-08-12:
I tried the esp32_ili9341. I am running the demo application and slide show is working, but the screen does not respond to touch.

sEpt0r, 2019-08-12:
@amirgon give the mailbox address, I can send a display for you.

amirgon, 2019-08-12:
> slide show is working, but the screen does not respond to touch.

@sEpt0r [esp32_ili9341](https://github.com/littlevgl/esp32_ili9341) is maintained and aligned to LittlevGL v6.0.  
It is part of the LittlevGL project and is unrelated to Micropython.  
Originally I used it as a reference to create the Micropython XPT2046 driver, which is not working for you.

I suggest to first make it work on your setup, both display and touch.  
After it works well, we'll know that the problem is related to the Micropython XPT2046 driver and take it from there.

If you are having trouble with esp32_ili9341, you can either post a question on [LittlevGL forum](https://forum.littlevgl.com/), or open an issue on [esp32_ili9341](https://github.com/littlevgl/esp32_ili9341).  
I think @kisvegabor and @C47D are the maintainers of esp32_ili9341, and they'll be happy to help you out!

sEpt0r, 2019-08-12:
Thank you! I did not have time to check this in more detail. I'll continue to test the esp32_ili9341.
My display and touch are worked with the Adafruit_ILI9341 and TFT_ILI9341 libraries.

C47D, 2019-08-12:
Hi @sEpt0r,
We did some changes on the esp32_ili9341 example project so now the default kit is the ESP32 Wrover Kit v4.1, please open an issue on that repo if you need any help.

sEpt0r, 2019-08-12:
@C47D thanks for the explonation.
I have ESP32 DevKit v1 and I set the defines as described in the Readme, I also set TOUCH_SUPPORT=1.
I want to test other applications and will dig deeper to understand why it doesn't work.

gersch07, 2019-08-17:
Dear all,
here is a quick update regarding XPT2046: I've just finished testing the "touchscreen calibration" app (TPCAL) and it works perfectly with the pins defined as in the readme. Does this mean we have to use HSPI and VSPI for TFT and touchscreen, respectively?

amirgon, 2019-08-17:
> here is a quick update regarding XPT2046

Thanks for the update @gersch07 !

> By the way, is it possible to include the most recent commits from the "lv_examples" repository?

The Micorpython driver is a rewrite of the C driver, so it might not be that easy to merge commits from lv_examples.  
The main difference between the C dirver and the Micropython driver is that in Micropython we want everything to be set on runtime, while on C driver you set it on compile time by "defines".  
For example, PIN numbers and all configuration is a "define" in the C driver.  

Please feel free to send us a PR when you make this work!

gersch07, 2019-08-25:
Dear all,

sorry for the slight delay in getting back to you! Today I modified the file "modXPT2046.c": basically I removed everything related to "spihost" and added the SPI initialization just like in "modILI9341.c" (I had to turn DMA for XPT2046 off). Following the wiring in the "esp32_ili9341" repository (where everything works), I should therefore reproduce the same conditions. However I was not able to obtain any coordinates; only the interrupt seems to do something (keeping the screen pressed anywhere pauses the continuous output of "-1:-1" in the console).
I'm a bit puzzled now...

amirgon, 2019-08-31:
@gersch07 When I have some time I'll review the code again.
Another option is to start over from the working C driver and convert it to Micropython driver. Most of the work is about changing the compile-time defines to runtime inputs (the wiring, SPI parameters etc.), but there is nothing "deeper" than that.   
Actually, if you want to create an ad-hoc driver with fixed parameters for your own use, you don't even need that. Binding the driver to Micropython is very easy.  

As I mentioned before, currently I'm not using the XPT2046 for my projects, so this is not a high priority for me right now - but if you want to try it yourself, I'll do my best to help.

closed 2019-09-28 by stale[bot]

amirgon, 2019-11-01:
Hi @gersch07 @sEpt0r @timstegeman!
You all expressed interest in XPT2046 touch driver for ESP32.  
Just wanted to let you know that I've added XPT2046 driver and tested it with ILI9341, they work well together (even on the same bus).

On current settings they are both connected to the same SPI bus, 40MHZ for the ILI9341 and 5MHZ for XPT2046, everything is configurable through their constructors.

Both drivers are **pure micropython drivers**.
I found that it was much easier to develop and integrate python drivers than C drivers, and they give good enough performance on ESP32 (~30FPS easily, when the entire screen is refreshed).
Of course, they can be used as a reference if anyone wants to convert them to C and get the maximum performance / minimum power.

They are available on **dev-6.1 branch**, I've updated the [README](https://github.com/littlevgl/lv_micropython/blob/dev-6.1/README.md) with a usage example.