lvgl/lv_binding_micropython #38: event

kaiakz, 2019-08-22:


kaiakz, 2019-08-22:
OK, I find a solution for python in the [blog](https://blog.littlevgl.com/2019-02-20/micropython-bindings), use `lambda` and `closure` to bind some user data for `event_cb`, such as id. Since LittlevGL is updated to v6.0, I think the blog(v5.x) should be updated too.

amirgon, 2019-08-22:
**Events are fully supported.**
**You shouldn't use `user_data`, and you don't need to.**

> * I notice :
> 
> > Currently only single callback per object is supported.

This is not true any more. All events can be registered, just like in C.

> I wonder if `event_cb` function could pass a normal parametre(obj: which obj's event, event: what event).

It does. `event_cb` receives the same parameters the C callback receives.

https://github.com/littlevgl/lv_binding_micropython/blob/5d5d198ebe6534ea8749361f964d0ab9ee714940/gen/lv_mpy_example.c#L4862-L4875

As you can see, the C callback receives the object and an `lv_event_t` enum, and so does the Micropython callback.  

> For paramtre, I wonder [littlevgl/lv_gui_builder#1 (comment)](https://github.com/littlevgl/lv_gui_builder/issues/1#issuecomment-523672651)_ can be use for it?
> For example, if I want to catch a widget's `LV_EVENT_DRAG_END`, how can i do?

How is casting related? You already have the object, you don't need to cast it in the event handler. What do you think is missing?

> For example, if I want to catch a widget's `LV_EVENT_DRAG_END`, how can i do?

Your callback receives the event type as the second argument, just compare it to `lv.EVENT.DRAG_END`

Here is an example where we check for `CLICKED` event:

https://github.com/littlevgl/lv_binding_micropython/blob/5d5d198ebe6534ea8749361f964d0ab9ee714940/examples/advanced_demo.py#L94-L97

> * To special a widget, in C , I usually attch an id in `user_data`, when I get an obj from `event_cb`, I can look up its ID. Does it also work in lv_mpy?

In general, you are not supposed to use `user_data`.  
It is used internally by Micropython Bindings to associate the lvgl object with the micropython object.  
**But you never need to use it!** You can pass any additional argument to your callback.   
Look at this example: 

https://github.com/littlevgl/lv_binding_micropython/blob/5d5d198ebe6534ea8749361f964d0ab9ee714940/examples/advanced_demo.py#L53-L54

The event handler by default receives 2 arguments, the object and the event type.  

But did you notice that the lambda receives another argument: `name=name`?   
Here we [Closure](https://en.wikipedia.org/wiki/Closure_(computer_programming)) to bring another variable `name` from outside (the for loop) into the callback.

amirgon, 2019-08-22:
> Since LittlevGL is updated to v6.0, I think the blog(v5.x) should be updated too.

Yes, I need to go over the docs and update them... I never get to that.  
Feel free to send a PR if you want to fix something in the docs.

kaiakz, 2019-08-22:
Thank you for your help!  I am a beginner to python, and I think I need to learn more python .XD
Here is a function for get all attribute for all widget:
```python
def getobjattr(obj, id):
    d={}
    d[id] = obj
    for i in dir(obj):
        if 'get_' in i:
            try:
                ret = eval('scr.' + i + '()')
                if isinstance(ret, (int,float,str,bool)):
                    d[i] = ret
            except:
                pass
```
It will ignore some functions with more than one parametre. Before I opened the issue, I add it to a `event_cb` function, and I thought a `event_cb` will always returns a `obj` class instance, not what the object it binds actually. (`btn`'s event_cb `parametre obj` gets a `obj` not `btn`). You see, other widget has more `get_xxx` function, if the `event_cb` function always gets `obj`, we can't get more info. So I want to use the `case` to change the parametre varible `obj` to other. 
If I use `closure`.just pass more parametre to pass the id and `btn` obj :
```python
btn = lv.btn()
# get id 
btn.set_event_cb(lambda obj=None, event=-1, name=id, real_obj = btn : getattr(btn, name) if(event == xxx) else xxx)
```
Is correct?

amirgon, 2019-08-31:
> It will ignore some functions with more than one parametre

How do you plan to handle functions with more than one parameter? or functions with a non-trivial parameter/return value? (a pointer to an array or a struct, for example)  

> If I use `closure`.just pass more parametre to pass the id and `btn` obj

Yes, you can pass another parameter to get the object (or any other data you need) in the callback.

> Here is a function for get all attribute for all widget:

You can use `try .. except` block to try calling the getters, but I suggest using `getattr` instead of `eval` to get the function object, it's more efficient and clean.

closed 2019-09-02 by kaiakz