lvgl/lv_binding_micropython #39: Trying to use gen_mpy.py got the error

lewisxhe, 2019-09-10:
I am trying to port lvgl on an existing micropython platform, when I try to add `gen_mpy.py` like [example](https://github.com/littlevgl/lv_micropython/blob/bc635700e4186f39763e5edee73660fbe1a27cd4/py/py.mk#L122) The generation tool reported the following error

```
Echo -e "LVGL-GEN /mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/build/lvgl/lv_mpy.c"
-e LVGL-GEN /mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/build/lvgl/lv_mpy.c
Mkdir -p /mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/build/lvgl/
Python /mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/esp32/libs/lvgl/lv_bindings/gen/gen_mpy.py -X anim -X group -X task -I /mnt/c/Users /Lewis/Desktop/es/MicroPython_BUILD/components/micropython/esp32/libs/lvgl/lv_bindings/lvgl -I /mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/esp32/libs/lvgl/lv_bindings /pycparser/utils/fake_libc_include /mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/esp32/libs/lvgl/lv_bindings/lvgl/lvgl.h > /mnt/c/Users/Lewis/Desktop/ Es/MicroPython_BUILD/components/micropython/build/lvgl/lv_mpy.c
Usage: gen_mpy.py [-h] [-I <Include Path>] [-D <Macro Name>]
                  [-E <Preprocessed File>] [-M <Module name string>]
                  [-MP <Prefix string>] [-MD <MetaData File Name>]
                  Input [input ...]
Gen_mpy.py: error: unrecognized arguments: -X -X group -X task /mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/esp32/libs/lvgl/lv_bindings/lvgl/lvgl.h
Makefile:392: recipe for target '/mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/build/lvgl/lv_mpy.c' failed
Make: *** [/mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/build/lvgl/lv_mpy.c] Error 2
```

I saw that the allowed parameters don't contain ` -X -X group -X task`, so I removed it, and I reported another error.

```
Echo -e "LVGL-GEN /mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/build/lvgl/lv_mpy.c"
-e LVGL-GEN /mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/build/lvgl/lv_mpy.c
Mkdir -p /mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/build/lvgl/
Python /mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/esp32/libs/lvgl/lv_bindings/gen/gen_mpy.py -I /mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/ Components/micropython/esp32/libs/lvgl/lv_bindings/lvgl -I /mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/esp32/libs/lvgl/lv_bindings/pycparser/utils/fake_libc_include /mnt/ c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/esp32/libs/lvgl/lv_bindings/lvgl/lvgl.h > /mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/build /lvgl/lv_mpy.c
Traceback (most recent call last):
  File "/mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/esp32/libs/lvgl/lv_bindings/gen/gen_mpy.py", line 141, in <module>
    Lv_str_enum_pattern = re.compile('^_{prefix}_STR_(.+)'.format(prefix=module_prefix.upper()))
AttributeError: 'NoneType' object has no attribute 'upper'
Makefile:392: recipe for target '/mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/build/lvgl/lv_mpy.c' failed
Make: *** [/mnt/c/Users/Lewis/Desktop/es/MicroPython_BUILD/components/micropython/build/lvgl/lv_mpy.c] Error 1
```

How can I solve this?

Best blessing.

amirgon, 2019-09-10:
Hi @lewisxhe !

> I am trying to port lvgl on an existing micropython platform, when I try to add `gen_mpy.py` like [example](https://github.com/littlevgl/lv_micropython/blob/bc635700e4186f39763e5edee73660fbe1a27cd4/py/py.mk#L122) The generation tool reported the following error

You are not looking at the latest version.

Please have a look at the [**latest version** of py.mk](https://github.com/littlevgl/lv_micropython/blob/2940838bf6d4999050efecb29a4152ab5796d5b3/py/py.mk#L22-L41).

The `-X` option was removed and a few other options were added.

You can also look at command line on the beginning of [lv_mpy_example.c](https://github.com/littlevgl/lv_binding_micropython/blob/master/gen/lv_mpy_example.c)

```c

/*
 * Auto-Generated file, DO NOT EDIT!
 *
 * Command line:
 * gen_mpy.py -MD lv_mpy_example.json -M lvgl -MP lv -I../../berkeley-db-1.xx/PORT/include -I../../lv_bindings -I. -I../.. -Ibuild -I../../mp-readline -I ../../lv_bindings/pycparser/utils/fake_libc_include ../../lv_bindings/lvgl/lvgl.h
 *
 * Preprocessing command:
 * gcc -E -std=c99 -DPYCPARSER  -I ../../berkeley-db-1.xx/PORT/include -I ../../lv_bindings -I . -I ../.. -I build -I ../../mp-readline -I ../../lv_bindings/pycparser/utils/fake_libc_include -include ../../lv_bindings/lvgl/lvgl.h ../../lv_bindings/lvgl/lvgl.h
 *
 * Generating Objects: obj(None), cont(obj), btn(obj), imgbtn(obj), label(obj), img(obj), line(obj), page(obj), list(obj), chart(obj), table(obj), cb(obj), bar(obj), slider(obj), led(obj), btnm(obj), kb(obj), ddlist(obj), roller(obj), ta(obj), canvas(obj), win(obj), tabview(obj), tileview(obj), mbox(obj), lmeter(obj), gauge(obj), sw(obj), arc(obj), preload(obj), calendar(obj), spinbox(obj)
 */
```

> I saw that the allowed parameters don't contain ` -X -X group -X task`, so I removed it, and I reported another error.

This is because the -M and -MP options are missing on the command line.

lewisxhe, 2019-09-11:
I have noticed that unlike the latest repository [lv_micropython](https://github.com/littlevgl/lv_micropython), thanks! @amirgon

closed 2019-09-11 by lewisxhe