lvgl/lv_binding_micropython #40: Getting Super Simple example to work

jonsmirl, 2019-09-21:
I'm unable to get this basic example to work on Linux desktop.  disp_drv does not have a disp_flush member.  https://github.com/littlevgl/lv_micropython

>>> import lvgl as lv
>>> lv.init()
>>> import SDL
>>> SDL.init()
>>> disp_drv = lv.disp_drv_t()
>>> lv.disp_drv_init(disp_drv)
>>> disp_drv.disp_flush = SDL.monitor_flush
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'lv_disp_drv_t' object has no attribute 'disp_flush'
>>> dir(disp_drv)
['__class__', '__dereference__', 'antialiasing', 'buffer', 'color_chroma_key', 'flush_cb', 'gpu_blend_cb', 'gpu_fill_cb', 'hor_res', 'monitor_cb', 'rotated', 'rounder_cb', 'set_px_cb', 'user_data', 'ver_res']
>>> 

Built from git commit 2940838bf6d4999050efecb29a4152ab5796d5b3

amirgon, 2019-09-22:
Hi @jonsmirl !

Here is a self contained "Hello World" example with SDL driver:

```python
import lvgl as lv
lv.init()

import SDL
SDL.init()

# Register SDL display driver.

disp_buf1 = lv.disp_buf_t()
buf1_1 = bytearray(480*10)
lv.disp_buf_init(disp_buf1,buf1_1, None, len(buf1_1)//4)
disp_drv = lv.disp_drv_t()
lv.disp_drv_init(disp_drv)
disp_drv.buffer = disp_buf1
disp_drv.flush_cb = SDL.monitor_flush
disp_drv.hor_res = 480
disp_drv.ver_res = 320
lv.disp_drv_register(disp_drv)

# Regsiter SDL mouse driver

indev_drv = lv.indev_drv_t()
lv.indev_drv_init(indev_drv) 
indev_drv.type = lv.INDEV_TYPE.POINTER;
indev_drv.read_cb = SDL.mouse_read;
lv.indev_drv_register(indev_drv);

# Display a button with a label

btn = lv.btn(lv.scr_act())
btn.align(None, lv.ALIGN.CENTER, 0,0)
label = lv.label(btn)
label.set_text("Hello World")
```

This should work on the unix port on Linux  with "master" branch of lv_micropython.  
You can run this online here: https://littlevgl.github.io/lv_micropython/ports/javascript/lvgl_editor.html?script_direct=f834e7f5556ee10cd6092c5457712b1f1aa6525d

Please try and let me know if it works for you.

The example in the README is outdated. I'll update the README with this example.

jonsmirl, 2019-09-22:
Works great. Thanks.

closed 2019-09-22 by jonsmirl