lvgl/lv_binding_micropython #41: Trying to get the esp32 version to build

jonsmirl, 2019-09-23:
Readme in esp32 directory has wrong make command.
https://github.com/littlevgl/lv_micropython/tree/master/ports/esp32

Should be:
make -C ports/esp32/  -j4 LV_CFLAGS="-DLV_HOR_RES=240 -DLV_VER_RES=320 -DLV_COLOR_DEPTH=16" deploy

mp_lodepng.c doesn't compile....

CC build/lodepng/mp_lodepng.c
build/lvgl/lv_mpy.c: In function 'mp_lv_style_body_border_t_attr':
build/lvgl/lv_mpy.c:3701:68: error: passing argument 1 of 'mp_read_ptr_lv_color32_t' from incompatible pointer type [-Werror=incompatible-pointer-types]
 #define mp_read_byref_lv_color32_t(field) mp_read_ptr_lv_color32_t(&field)
                                                                    ^
build/lvgl/lv_mpy.c:3800:43: note: in expansion of macro 'mp_read_byref_lv_color32_t'
             case MP_QSTR_color: dest[0] = mp_read_byref_lv_color32_t(data->color); break; // converting from lv_color_t;
                                           ^
build/lvgl/lv_mpy.c:3695:24: note: expected 'lv_color32_t * {aka union <anonymous> *}' but argument is of type 'lv_color_t * {aka union <anonymous> *}'
 STATIC inline mp_obj_t mp_read_ptr_lv_color32_t(lv_color32_t *field)
                        ^
build/lvgl/lv_mpy.c:3816:49: error: incompatible types when assigning to type 'lv_color_t {aka union <anonymous>}' from type 'lv_color32_t {aka union <anonymous>}'
                 case MP_QSTR_color: data->color = mp_write_lv_color32_t(dest[1]); break; // converting to lv_color_t;
 ... for another hundred errors...

amirgon, 2019-09-23:
Thanks for reporting this.

Please try the following :

```
make -C mpy-cross
make -C ports/esp32 LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" deploy
```

I'm also using `PYTHON=python2` make parameter because of the `pyparsing` version I have installed. (see more details on[ Micropython's ESP32 README](https://github.com/micropython/micropython/blob/master/ports/esp32/README.md)).

It should build without errors, including lodepng.  
Please let me know if it works for you, and if it does I'll update the README.

jonsmirl, 2019-09-24:
Thanks. It builds now. I had to back off to esp-idf v3.3-beta to get it to build. v3.3 has been released for a while now.

The build failure in v3.3 looks like a minor link error...

GEN build/esp32.project.ld
LINK build/application.elf
build/esp-idf/esp32/libesp32.a(cpu_start.o):(.iram1.27.literal+0x4c): undefined reference to `bootloader_common_set_flash_cs_timing'
build/esp-idf/esp32/libesp32.a(cpu_start.o): In function `call_start_cpu0':
cpu_start.c:(.iram1.27+0x4d): undefined reference to `bootloader_common_set_flash_cs_timing'
Makefile:514: recipe for target 'build/application.elf' failed
make: *** [build/application.elf] Error 1

amirgon, 2019-09-24:
> Thanks. It builds now. I had to back off to esp-idf v3.3-beta to get it to build. v3.3 has been released for a while now.
> 
> The build failure in v3.3 looks like a minor link error...

Every Micropython version is supposed to work with a very specific ESP-IDF version.
If ESP-IDF version is incorrect, you'll get a warning like this:

```
** WARNING **
The git hash of ESP IDF does not match the supported version
The build may complete and the firmware may work but it is not guaranteed
ESP IDF path:       /home/amirgon/esp/esp-idf
Current git hash:   6ccb4cf5b7d1fdddb8c2492f9cbc926abaf230df
Supported git hash: 6b3da6b1882f3b72e904cc90be67e9c4e3f369a9
```

I recommend only using the supported ESP-IDF version.  

I'm planning to sync lv_micropython to upstream Micropython soon, which now supports an up to date ESP-IDF.

closed 2019-09-24 by amirgon