lvgl/lv_binding_micropython #44: ESP32 colors not right

jonsmirl, 2019-09-24:
I used this example, I see the button but the colors are all wrong. The demo button should be blue, instead I have bands of green. The text is white and purple on a black background.  Something not mapping right with 32 to 16 bit conversion?

Screenshot...
https://drive.google.com/file/d/10Q2jwXzOoF9NHsHCGu4o0sm3i9w8D0aa/view?usp=sharing

I built with this - make -C ports/esp32 LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" deploy

import lvesp32
import lvgl as lv
import ILI9341 as ili

disp = ili.display(miso=5, mosi=18, clk=19, cs=13, dc=12, rst=4, backlight=2)
disp.init()

disp_buf1 = lv.disp_buf_t()
buf1_1 = bytearray(480*10)
lv.disp_buf_init(disp_buf1,buf1_1, None, len(buf1_1)//4)
disp_drv = lv.disp_drv_t()
lv.disp_drv_init(disp_drv)
disp_drv.buffer = disp_buf1
disp_drv.flush_cb = disp.flush
disp_drv.hor_res = 240
disp_drv.ver_res = 320
lv.disp_drv_register(disp_drv)
        
scr = lv.obj()
btn = lv.btn(scr)
btn.align(lv.scr_act(), lv.ALIGN.CENTER, 0, 0)
label = lv.label(btn)
label.set_text("Hello World!")

lv.scr_load(scr)

amirgon, 2019-09-24:
Hi @jonsmirl, 

- Could you try using the [pure micropython driver](https://github.com/littlevgl/lv_binding_micropython/blob/master/driver/esp32/ili9341.py)? (it is written now as a whole program that displays a chart, but you can remove that part and put your button instead)
- Could you try your code when building Micropython without the `-DLV_COLOR_16_SWAP=1` option? It was added to save the need from the pure micropython driver to swap colors, but I never checked it was still working with the original driver.

jonsmirl, 2019-09-24:
Using the pure python driver I get a grey graph with a moving red line.  Are those the correct colors?

amirgon, 2019-09-24:
> Using the pure python driver I get a grey graph with a moving red line. Are those the correct colors?

Yes.
You can always verify this on the unix port or on the web simulation:

https://littlevgl.github.io/lv_micropython/ports/javascript/lvgl_editor.html?script_direct=f107afddc69624e0c163e18e3ef3c0aead75bef1

jonsmirl, 2019-09-24:
Removing -DLV_COLOR_16_SWAP=1 makes the button blue like it should be

amirgon, 2019-09-24:
> Removing -DLV_COLOR_16_SWAP=1 makes the button blue like it should be

Very well.
I'll update the README that `-DLV_COLOR_16_SWAP=1` is only needed when using the pure python driver.

amirgon, 2019-09-25:
If everything is resolved, I'll close it now.
Thank you for reporting this.

closed 2019-09-25 by amirgon