lvgl/lv_binding_micropython #45: How to connect touch?

ColdFireHunter, 2019-09-27:
Hello, maybe somewone now's how to connect a resistive touch to ESP32. The Touchscreen is 4Wire and has no Controller. I dont understand waht "Toch_Sense" and "Touch_Rail" mean. I have connected the 4 Wire's of the screen to the pins in my code. The code is down below.
Thank you for your help.
[image]
[image]

amirgon, 2019-09-27:
Hi @ColdFireHunter !  

The "Touch_Sense" and "Touch_Rail" pins are used on a configuration where an extra circuit allows detecting touch on the touchscreen. The idea is that when you touch you close a circuit. You can use this circuit to either detect continuity, or to detect resistance.
If you don't use this extra circuit, and connect these 4 wires directly to ESP32 ADC pins, then you can't use "Touch_Sense" and "Touch_Rail".    

I didn't try this driver in this configuration (connecting all wires directly to ADC) so I'm not sure what would be the result, but I believe there is a way to make it work.

What are you seeing when you try this?

ColdFireHunter, 2019-09-27:
Hi @amirgon!
I See only 0,0 in the screen.

amirgon, 2019-09-28:
@ColdFireHunter 

One problem I can see with your code is wiring.

**You set `yp` to 34 which is GPI**, and cannot be set as output.  
When measuring the Y axis, the driver sets `yp` to 1 and `ym` to 0, and measures the voltage on xp.  
With your wiring this is not possible.  

I use the following wiring:
`xp = 32, yp = 33, xm = 25, ym = 26`

amirgon, 2019-10-05:
Another problem that could be related:

https://github.com/littlevgl/lv_binding_micropython/blob/b1ab6ebfe510534079a143472f2535393fe3ceda/driver/esp32/modrtch.c#L481-L492

**If touch is not calibrated** you will constantly get  "Overflow measurement", and never record a touch.  
I'm working now on touchscreen calibration script, that can solve this problem.

ColdFireHunter, 2019-10-29:
@amirgon 
Ok, Thank you for making a script.

amirgon, 2019-10-29:
> Ok, Thank you for making a script.

@ColdFireHunter Did it work for you? Did you find the problem?  

I improved the rtouch driver a little and added calibration support.
As an example, you can use [this script](https://github.com/littlevgl/lv_binding_micropython/blob/master/lib/tpcal.py) to calibrate it.

ColdFireHunter, 2019-10-29:
Hi, i will try it today

ColdFireHunter, 2019-10-29:
@amirgon 
Hi, i've tested it and it dosen't work for me becuase it has some Errors. Look at the attached picture. I have changed only the Section were you import the driver because that also didn't work, but i've added some code from the example. Thank you for your help!
[image]

embeddedt, 2019-10-30:
> but no touch wat so ever.

@ColdFireHunter It might be difficult for @amirgon to help you with that limited information. Did you try the `update_label` function again? Does it behave any differently?

ColdFireHunter, 2019-10-30:
@embeddedt 
no i didn't i will try it and report.

amirgon, 2019-10-30:
> @amirgon
> Hi, now i've got the screen, but no touch wat so ever.

@ColdFireHunter To get more information, you can enable logging:

https://github.com/littlevgl/lv_binding_micropython/blob/ddf09164ee1711a61169030a7ee8bf370ee5743f/driver/esp32/modrtch.c#L6-L7

Please try to uncomment this line, build `lv_micropython`, and send us the logs that are printed when you touch the screen.

Also, on the latest version of tpcal.py, I changed the display driver to the python display driver:
https://github.com/littlevgl/lv_binding_micropython/blob/ddf09164ee1711a61169030a7ee8bf370ee5743f/lib/tpcal.py#L8-L10

You can find it here: https://github.com/littlevgl/lv_binding_micropython/blob/dev-6.1/driver/esp32/ili9341.py
I recommend using it, as it's more up to date, more configurable and easier to debug.

Don't forget to run "make" with the correct `LV_CFLAGS` argument:
https://github.com/littlevgl/lv_binding_micropython/blob/1638ca16267070309f37f7235eacbeaa9ca6a2a4/driver/esp32/ili9341.py#L4

closed 2019-11-27 by stale[bot]