lvgl/lv_binding_micropython #47: Trying to get ESP-IDF 4 working, ili9341 errors

jonsmirl, 2019-09-28:
I have merged in upstream with ESP-IDF 4 support and turned it on.  The structures have changed a little and the ili9341 driver won't build. Can you enlighten on how to fix this? Something is not getting generated right...

CC ../../lib/lv_bindings/driver/esp32/modlvesp32.c
CC ../../lib/lv_bindings/driver/esp32/modILI9341.c
CC ../../lib/lv_bindings/driver/esp32/modxpt2046.c
CC ../../lib/lv_bindings/driver/esp32/modrtch.c
CC build-TINYPICO/espidfmod/mp_espidf.c
build-TINYPICO/espidfmod/mp_espidf.c:3546:17: error: redefinition of 'mp_spicommon_bus_using_iomux'
 STATIC mp_obj_t mp_spicommon_bus_using_iomux(size_t mp_n_args, const mp_obj_t *mp_args)
                 ^
build-TINYPICO/espidfmod/mp_espidf.c:3530:17: note: previous definition of 'mp_spicommon_bus_using_iomux' was here
 STATIC mp_obj_t mp_spicommon_bus_using_iomux(size_t mp_n_args, const mp_obj_t *mp_args)
                 ^
build-TINYPICO/espidfmod/mp_espidf.c:3553:39: error: redefinition of 'mp_spicommon_bus_using_iomux_obj'
 STATIC MP_DEFINE_CONST_LV_FUN_OBJ_VAR(mp_spicommon_bus_using_iomux_obj, 1, mp_spicommon_bus_using_iomux, spicommon_bus_using_iomux);
                                       ^
build-TINYPICO/espidfmod/mp_espidf.c:76:39: note: in definition of macro 'MP_DEFINE_CONST_LV_FUN_OBJ_VAR'
     const mp_lv_obj_fun_builtin_var_t obj_name = \
                                       ^
build-TINYPICO/espidfmod/mp_espidf.c:3537:39: note: previous definition of 'mp_spicommon_bus_using_iomux_obj' was here
 STATIC MP_DEFINE_CONST_LV_FUN_OBJ_VAR(mp_spicommon_bus_using_iomux_obj, 1, mp_spicommon_bus_using_iomux, spicommon_bus_using_iomux);
                                       ^
build-TINYPICO/espidfmod/mp_espidf.c:76:39: note: in definition of macro 'MP_DEFINE_CONST_LV_FUN_OBJ_VAR'
     const mp_lv_obj_fun_builtin_var_t obj_name = \
                                       ^
build-TINYPICO/espidfmod/mp_espidf.c: In function 'mp_spi_dev_t_attr':
build-TINYPICO/espidfmod/mp_espidf.c:12173:74: error: passing argument 1 of 'mp_arr_from_uint32_t___16__' discards 'volatile' qualifier from pointer target type [-Werror=discarded-qualifiers]
             case MP_QSTR_data_buf: dest[0] = mp_arr_from_uint32_t___16__(data->data_buf); break; // converting from uint32_t [16];
                                                                          ^
build-TINYPICO/espidfmod/mp_espidf.c:10804:17: note: expected 'uint32_t * {aka unsigned int *}' but argument is of type 'volatile uint32_t * {aka volatile unsigned int *}'
 STATIC mp_obj_t mp_arr_from_uint32_t___16__(uint32_t *arr)
                 ^
build-TINYPICO/espidfmod/mp_espidf.c:12426:47: error: passing argument 1 of 'memcpy' discards 'volatile' qualifier from pointer target type [-Werror=discarded-array-qualifiers]
                 case MP_QSTR_data_buf: memcpy(&data->data_buf, mp_arr_to_uint32_t___16__(dest[1]), sizeof(uint32_t)*16); break; // converting to uint32_t [16];
                                               ^
In file included from /home/jonsmirl/aosp/esp/esp-idf/components/newlib/include/stdlib.h:11:0,
                 from build-TINYPICO/espidfmod/mp_espidf.c:18:
/home/jonsmirl/aosp/esp/esp-idf/components/newlib/include/string.h:23:15: note: expected 'void * restrict' but argument is of type 'volatile uint32_t (*)[16] {aka volatile unsigned int (*)[16]}'
 _PTR   _EXFUN(memcpy,(_PTR __restrict, const _PTR __restrict, size_t));
               ^
/home/jonsmirl/aosp/esp/esp-idf/components/newlib/include/_ansi.h:65:30: note: in definition of macro '_EXFUN'
 #define _EXFUN(name, proto)  name proto
                              ^
build-TINYPICO/espidfmod/mp_espidf.c: At top level:
build-TINYPICO/espidfmod/mp_espidf.c:12681:5: error: initialization discards 'volatile' qualifier from pointer target type [-Werror=discarded-qualifiers]
     (spi_dev_t*)&SPI0
     ^
build-TINYPICO/espidfmod/mp_espidf.c:12691:5: error: initialization discards 'volatile' qualifier from pointer target type [-Werror=discarded-qualifiers]
     (spi_dev_t*)&SPI1
     ^
build-TINYPICO/espidfmod/mp_espidf.c:12701:5: error: initialization discards 'volatile' qualifier from pointer target type [-Werror=discarded-qualifiers]
     (spi_dev_t*)&SPI2
     ^
build-TINYPICO/espidfmod/mp_espidf.c:12711:5: error: initialization discards 'volatile' qualifier from pointer target type [-Werror=discarded-qualifiers]
     (spi_dev_t*)&SPI3
     ^
build-TINYPICO/espidfmod/mp_espidf.c:17757:5: error: initialization discards 'volatile' qualifier from pointer target type [-Werror=discarded-qualifiers]
     (rtc_cntl_dev_t*)&RTCCNTL
     ^
cc1: all warnings being treated as errors
../../py/mkrules.mk:47: recipe for target 'build-TINYPICO/build-TINYPICO/espidfmod/mp_espidf.o' failed
make: *** [build-TINYPICO/build-TINYPICO/espidfmod/mp_espidf.o] Error 1
make: Leaving directory '/home/jonsmirl/aosp/foo/micropython/ports/esp32'

[mp_espidf.c.log](https://github.com/littlevgl/lv_binding_micropython/files/3665030/mp_espidf.c.log)
[mp_espidf.pp.c.log](https://github.com/littlevgl/lv_binding_micropython/files/3665031/mp_espidf.pp.c.log)

jonsmirl, 2019-09-28:
Another issue lvgl is making a second build directory inside the first one...

jonsmirl@ares:~/aosp/foo/micropython/ports/esp32/build-TINYPICO/build-TINYPICO$ find
.
./frozen_mpy.o
./lodepng
./lodepng/lodepng.P
./lodepng/mp_lodepng.P
./lodepng/lodepng.o
./lodepng/mp_lodepng.o
./lvgl
./lvgl/lv_mpy.o
./lvgl/lv_mpy.P
./espidfmod
./frozen_mpy.P

jonsmirl, 2019-09-28:
I got rid of half of them, this definition is duplicated in the esp-if header file:
bool spicommon_bus_using_iomux(spi_host_device_t host);

This is what is left, they are all about discarding volatile on a parameter.

CC ../../lib/lv_bindings/driver/esp32/modrtch.c
CC build-GENERIC_SPIRAM/espidfmod/mp_espidf.c
build-GENERIC_SPIRAM/espidfmod/mp_espidf.c: In function 'mp_spi_dev_t_attr':
build-GENERIC_SPIRAM/espidfmod/mp_espidf.c:12157:74: error: passing argument 1 of 'mp_arr_from_uint32_t___16__' discards 'volatile' qualifier from pointer target type [-Werror=discarded-qualifiers]
             case MP_QSTR_data_buf: dest[0] = mp_arr_from_uint32_t___16__(data->data_buf); break; // converting from uint32_t [16];
                                                                          ^
build-GENERIC_SPIRAM/espidfmod/mp_espidf.c:10788:17: note: expected 'uint32_t * {aka unsigned int *}' but argument is of type 'volatile uint32_t * {aka volatile unsigned int *}'
 STATIC mp_obj_t mp_arr_from_uint32_t___16__(uint32_t *arr)
                 ^
build-GENERIC_SPIRAM/espidfmod/mp_espidf.c:12410:47: error: passing argument 1 of 'memcpy' discards 'volatile' qualifier from pointer target type [-Werror=discarded-array-qualifiers]
                 case MP_QSTR_data_buf: memcpy(&data->data_buf, mp_arr_to_uint32_t___16__(dest[1]), sizeof(uint32_t)*16); break; // converting to uint32_t [16];
                                               ^
In file included from /home/jonsmirl/aosp/esp/esp-idf/components/newlib/include/stdlib.h:11:0,
                 from build-GENERIC_SPIRAM/espidfmod/mp_espidf.c:18:
/home/jonsmirl/aosp/esp/esp-idf/components/newlib/include/string.h:23:15: note: expected 'void * restrict' but argument is of type 'volatile uint32_t (*)[16] {aka volatile unsigned int (*)[16]}'
 _PTR   _EXFUN(memcpy,(_PTR __restrict, const _PTR __restrict, size_t));
               ^
/home/jonsmirl/aosp/esp/esp-idf/components/newlib/include/_ansi.h:65:30: note: in definition of macro '_EXFUN'
 #define _EXFUN(name, proto)  name proto
                              ^
build-GENERIC_SPIRAM/espidfmod/mp_espidf.c: At top level:
build-GENERIC_SPIRAM/espidfmod/mp_espidf.c:12665:5: error: initialization discards 'volatile' qualifier from pointer target type [-Werror=discarded-qualifiers]
     (spi_dev_t*)&SPI0
     ^
build-GENERIC_SPIRAM/espidfmod/mp_espidf.c:12675:5: error: initialization discards 'volatile' qualifier from pointer target type [-Werror=discarded-qualifiers]
     (spi_dev_t*)&SPI1
     ^
build-GENERIC_SPIRAM/espidfmod/mp_espidf.c:12685:5: error: initialization discards 'volatile' qualifier from pointer target type [-Werror=discarded-qualifiers]
     (spi_dev_t*)&SPI2
     ^
build-GENERIC_SPIRAM/espidfmod/mp_espidf.c:12695:5: error: initialization discards 'volatile' qualifier from pointer target type [-Werror=discarded-qualifiers]
     (spi_dev_t*)&SPI3
     ^
build-GENERIC_SPIRAM/espidfmod/mp_espidf.c:17741:5: error: initialization discards 'volatile' qualifier from pointer target type [-Werror=discarded-qualifiers]
     (rtc_cntl_dev_t*)&RTCCNTL
     ^
cc1: all warnings being treated as errors
../../py/mkrules.mk:47: recipe for target 'build-GENERIC_SPIRAM/build-GENERIC_SPIRAM/espidfmod/mp_espidf.o' failed

jonsmirl, 2019-09-28:
I shut those errors off to get past them..
	-Wno-error=unused-variable -Wno-error=deprecated-declarations \

Now I have a link error...
LINK build-GENERIC_SPIRAM/application.elf
build-GENERIC_SPIRAM/build-GENERIC_SPIRAM/espidfmod/mp_espidf.o:(.literal.mp_xt_get_interrupt_handler_arg+0x0): undefined reference to `xt_get_interrupt_handler_arg'
build-GENERIC_SPIRAM/build-GENERIC_SPIRAM/espidfmod/mp_espidf.o: In function `mp_xt_get_interrupt_handler_arg':
mp_espidf.c:(.text.mp_xt_get_interrupt_handler_arg+0x9): undefined reference to `xt_get_interrupt_handler_arg'
build-GENERIC_SPIRAM/build-GENERIC_SPIRAM/espidfmod/mp_espidf.o:(.rodata.mp_xt_get_interrupt_handler_arg_obj+0xc): undefined reference to `xt_get_interrupt_handler_arg'
Makefile:748: recipe for target 'build-GENERIC_SPIRAM/application.elf' failed
make: *** [build-GENERIC_SPIRAM/application.elf] Error 1

I stubbed that out in espidf.c, 
inline void *xt_get_interrupt_handler_arg(int n){return NULL;}

Now I can built on ESP IDF 4.0

jonsmirl, 2019-09-28:
These patches will allow it to compile with esp-idf 4...

jonsmirl@ares:~/aosp/digi-iot/micropython$ git diff
diff --git a/lib/lv_bindings b/lib/lv_bindings
--- a/lib/lv_bindings
+++ b/lib/lv_bindings
@@ -1 +1 @@
-Subproject commit 77082d701d9d67d7ff0722b2179593c96e41faf9
+Subproject commit 77082d701d9d67d7ff0722b2179593c96e41faf9-dirty
diff --git a/ports/esp32/Makefile b/ports/esp32/Makefile
index b6fd9c76c..b77ce2001 100644
--- a/ports/esp32/Makefile
+++ b/ports/esp32/Makefile
@@ -191,6 +191,7 @@ CFLAGS_COMMON = -Os -ffunction-sections -fdata-sections -fstrict-volatile-bitfie
        -mlongcalls -nostdlib \
        -Wall -Werror -Wno-error=unused-function -Wno-error=unused-but-set-variable \
        -Wno-error=unused-variable -Wno-error=deprecated-declarations \
+        -Wno-error=discarded-array-qualifiers -Wno-error=discarded-qualifiers \
        -DESP_PLATFORM
 
 CFLAGS_BASE = -std=gnu99 $(CFLAGS_COMMON) -DMBEDTLS_CONFIG_FILE='"mbedtls/esp_config.h"' -DHAVE_CONFIG_H

jonsmirl@ares:~/aosp/digi-iot/micropython/lib/lv_bindings$ git diff
diff --git a/driver/esp32/espidf.c b/driver/esp32/espidf.c
index fab6e42..0f0b49d 100644
--- a/driver/esp32/espidf.c
+++ b/driver/esp32/espidf.c
@@ -21,6 +21,7 @@ inline uint32_t gpio_intr_pending_high(void){return 0;}
 inline uint32_t gpio_intr_pending(void){return 0;}
 inline void gpio_intr_handler_register(gpio_intr_handler_fn_t fn, void *arg){}
 inline void gpio_init(void){}
+inline void *xt_get_interrupt_handler_arg(int n){return NULL;}
 
 void task_delay_ms(int ms)
 {
diff --git a/driver/esp32/espidf.h b/driver/esp32/espidf.h
index bf87cb7..7905516 100644
--- a/driver/esp32/espidf.h
+++ b/driver/esp32/espidf.h
@@ -53,7 +53,7 @@ static inline void get_ccount(int *ccount)
 #include "driver/spi_master.h"
 #include "esp_heap_caps.h"
 #include "esp_log.h"
-#include "esp_clk.h"
+#include "esp32/clk.h"
 #include "driver/adc.h"
 #include "driver/i2s.h"

jonsmirl@ares:~/aosp/esp/esp-idf$ git diff
diff --git a/components/driver/include/driver/spi_common.h b/components/driver/include/driver/spi_common.h
index b44159546..15e91cd45 100644
--- a/components/driver/include/driver/spi_common.h
+++ b/components/driver/include/driver/spi_common.h
@@ -319,7 +319,7 @@ bool spicommon_bus_using_iomux(spi_host_device_t host);
  *
  * @return false if any pins are through the GPIO matrix, otherwise true.
  */
-bool spicommon_bus_using_iomux(spi_host_device_t host);
+//bool spicommon_bus_using_iomux(spi_host_device_t host);
 
 /**
  * @brief Get the IRQ source for a specific SPI host

amirgon, 2019-09-28:
Hi @jonsmirl !

I've just update lv_micropython to support ESP IDF v4.0-beta1.
Please give it a try and let me know if you see any problems.

> Another issue lvgl is making a second build directory inside the first one...

For some reason `mpy-cross` is also building the lv modules. Apparently py.mk is processed twice - first globally and again for the specific port that is being built.  
The only implication is build time (build is just slightly longer), I don't think it affects build outputs.  
Do you see any problem with this?

jonsmirl, 2019-09-28:
I  am using commit 47de8b5d38721c5132de28efbefa2f4e2d3395fd
These patches are needed in addition to your update.

jonsmirl, 2019-09-28:
Maybe I should switch over to the pure Python LCD driver? That would avoid these issues.

amirgon, 2019-09-28:
> I am using commit 47de8b5d38721c5132de28efbefa2f4e2d3395fd

What's that commit? Is it ESP-IDF commit? I can't find it.

jonsmirl, 2019-09-28:
commit 47de8b5d38721c5132de28efbefa2f4e2d3395fd (origin/master, origin/HEAD, master)
Author: Amir Gonnen <amirgonnen@gmail.com>
Date:   Sat Sep 28 00:52:33 2019 +0300

    update README

embeddedt, 2019-09-28:
https://github.com/littlevgl/lv_micropython/commit/47de8b5d38721c5132de28efbefa2f4e2d3395fd

Second-newest commit on `lv_micropython`.

amirgon, 2019-09-28:
> [littlevgl/lv_micropython@47de8b5](https://github.com/littlevgl/lv_micropython/commit/47de8b5d38721c5132de28efbefa2f4e2d3395fd)
> 
> Second-newest commit on `lv_micropython`.

The get the newest commit. That's where I fixed this problem...

jonsmirl, 2019-09-28:
Ok, I see now there is a new commit, I will try it.

jonsmirl, 2019-09-28:
Does this need a patch in ESP-IDF to remove the duplicate definition?

jonsmirl, 2019-09-28:
I see it handle duplicate definition. Your fixes are good, I am able to build.

I want to be on IDF4 because Bluetooth support is just about to land in micropython. I am running using the PR.

closed 2019-09-28 by jonsmirl

amirgon, 2019-09-28:
> Does this need a patch in ESP-IDF to remove the duplicate definition?

No. I changed the binding generation script, to avoid generating duplicate definitions:

https://github.com/littlevgl/lv_binding_micropython/blob/736b5f0e37d9017a80b3068416c7dc49f788dbec/gen/gen_mpy.py#L1499-L1505

amirgon, 2019-09-29:
> I want to be on IDF4 because Bluetooth support is just about to land in micropython. I am running using the PR.

@jonsmirl You know that even if some ESP feature is not officially supported in Micropython (such as Bluetooth or any other feature), you can still access it in Micropython through the espidf module.  

The espidf module converts ESP-IDF API automatically to Micropython module, based on files included in [espidf.h](https://github.com/littlevgl/lv_binding_micropython/blob/master/driver/esp32/espidf.h).