lvgl/lv_binding_micropython #48: Support `lv_async_call` callback

amirgon, 2019-11-05:
`lv_async_call` doesn't obey callback conventions.
See https://forum.littlevgl.com/t/how-to-use-lv-async-call/1096

amirgon, 2019-11-05:
@kisvegabor - would it be acceptable to add a new function that does the same thing but obey the callback convention?

kisvegabor, 2019-11-06:
Can you suggest a new prototype?

amirgon, 2019-11-06:
@kisvegabor - The main problem is with the callback prototype:
```c
typedef void (*lv_async_cb_t)(void *);
```

On other cases, for example indev and display callbacks, the callback receives a struct with a `user_data` field. For example: 

```c
void (*flush_cb)(struct _disp_drv_t * disp_drv, const lv_area_t * area, lv_color_t * color_p);
```

Is it possible to change the callback prototype (or add a new one) in the same manner?
Something like:
```c
typedef struct _lv_async_info_t {
    void (*cb)(struct _lv_async_info_t * async_info);
    void *user_data;
} lv_async_info_t;

lv_res_t lv_async_call(lv_async_info_t * async_info);

```

This would also give you the flexibility to add new fields to `lv_async_info_t` for the callback use without breaking backward compatibility.

What do you think?

embeddedt, 2019-11-06:
> Is it possible to change the callback prototype

Prototypes can only be changed in major releases, so we would have to change it in 7.0.

> (or add a new one)

That can be done.

kisvegabor, 2019-11-08:
We can do the same technique as we did with animations. 
There we added `lv_anim_custom_exec_cb_t` which receives `lv_anim_t *` as first parameter (instead of `a->var`).

amirgon, 2019-11-09:
> We can do the same technique as we did with animations.
> There we added `lv_anim_custom_exec_cb_t` which receives `lv_anim_t *` as first parameter (instead of `a->var`).

I was starting to think maybe it's worth the effort to expand the calling convention on the binding script.  
Today, in case I recognize a callback registration (function that receives a function pointer) which doesn't fit into the existing calling convention, I just skip it.
Instead I could check if:

- It receives a `void *` as the last argument
- The callback also receives `void *` as the last argument

and assume that this `void *` is the `user_data` (even if it's not called that way).

It may be too broad though, and "catch" cases where that `void *` is not the `user_data`.   
Also, I'm still not sure how much work it's involved in implementing this, I will need to look into it a bit more.

amirgon, 2019-11-09:
Fixed on dev-6.1 branch on 155f748a59c975919b5b38190e51e0340e1aefc8

Usage example: 

```python
def cb(user_data):
    print(user_data.cast()['value'])

lv.async_call(cb, {'value':42})
```

Or just pass `None`:

```python
def cb(user_data):
    print('cb called!')

lv.async_call(cb, None)
```

closed 2019-11-09 by amirgon

kisvegabor, 2019-11-10:
Great! I'm sure it will be useful later too.