lvgl/lv_binding_micropython #59: Load fonts dynamically

amirgon, 2019-12-31:
Today fonts must be compiled into lvgl.
I would be useful to allow loading fonts dynamically on runtime.  

This could be implemented by creating native .mpy files from a .elf file.

---

References:
- https://github.com/micropython/micropython/pull/5083.
- https://github.com/littlevgl/lvgl/issues/1237#issuecomment-559583614

imliubo, 2020-09-10:
Have an example for this?

amirgon, 2020-09-10:
@imliubo Actually no. It was implemented as part of LVGL core and I didn't try this on Micropython yet.
You are welcome to be the first who tries this out on Micropython!

imliubo, 2020-09-10:
@amirgon I try it now! It seem I need driver the filesystem operation interface first.
But I have converted my font file.
```bash
$ lv_font_conv --size 20 --format bin  --bpp 1 --font Alibaba-PuHuiTi-Medium.subset.ttf -o Alibaba-PHT-20.bin --range 0x20-0x9f9f
```
Is this right?

amirgon, 2020-09-10:
> Is this right?

Yes you would need to implement a filesystem driver. I didn't try to do that either.

imliubo, 2020-09-10:
@amirgon It seems successful! (But the Chinese display doesn't seem to work :() English can load your own ttf font file, but you need to use the lvgl font conversion tool to convert it.

I use this filesystem driver: [file_drv -mhepp](https://forum.lvgl.io/uploads/short-url/o1wzS3zlqvu7acIEarJKNju8bin.zip)
But need change this **fs_register** function:
```python
def fs_register(fs_drv, letter):

    fs_drv.init()  # this line
    fs_drv.letter = ord(letter)
    fs_drv.open_cb = fs_open_cb
    fs_drv.read_cb = fs_read_cb
    fs_drv.write_cb = fs_write_cb
    fs_drv.seek_cb = fs_seek_cb
    fs_drv.tell_cb = fs_tell_cb
    fs_drv.close_cb = fs_close_cb

    fs_drv.register()   # this line
```
And the example code:

```python
import lvgl as lv
import file_drv

lv.init()

# display driver init...

# FS driver init.
fs_drv = lv.fs_drv_t()
file_drv.fs_register(fs_drv, 'S')

# load the font file from filesystem
myfont = lv.font_load("S:res/PHT-ASCII-20.bin")  # Refer here to convert your font file: https://github.com/lvgl/lv_font_conv

scr = lv.scr_act()
scr.clean()

style = lv.style_t()
style.init()

label = lv.label(scr)
style.set_text_font(lv.STATE.DEFAULT, myfont)
label.add_style(label.PART.MAIN, style)

label.set_text("Hi LVGL(Load fonts dynamically)")
label.align(None, lv.ALIGN.CENTER, 0, 0)
```
Result:
**Built-in font**
[image]
**My font by load**, It looks like the space character is missing.
[image]

Continue to check the Chinese font problem.

amirgon, 2020-09-10:
@imliubo Nice work!
If you want you can contribute the example under `examples/` to show how to do it.
The filesystem driver can probably go under `lib/`.
I would be great if you open a PR and add this!

embeddedt, 2020-09-10:
> It would be great if ttf font files could be parsed directly :)

You can use `lv_lib_freetype` for that.

imliubo, 2020-09-10:
@embeddedt It seems that there is no micropython interface (not sure, if there is, please let me know).

embeddedt, 2020-09-10:
That's true; I think it should be possible to initialize it in C and/or run it through the binding generator, but I'm not familiar with the details.

amirgon, 2020-09-10:
@imliubo you are right, currently there is no micropython interface for `lv_lib_freetype`.  
Today Micropython interface is generated only for core LVGL, for lodepng (PNG library) and for esp-idf.  

Do you think `lv_lib_freetype` micropython API would be useful?  
If you do please open a new issue on lv_binding_micropython to track this.