lvgl/lv_binding_micropython #64: Parameterize width/height of the SDL window.

rockindy, 2020-01-16:
I have modified SDL display driver to let user set SDL monitor resolution via SDL.init().
Could I create a pull request to merge this patch?
Thanks.

amirgon, 2020-01-16:
Hi @rockindy!

> Could I create a pull request to merge this patch?

Of course. Thank you.  

But please make it backwards compatible - instead of 2 required arguments to `init`, make them optional kwargs.  
You can use `MP_DEFINE_CONST_FUN_OBJ_KW` instead of `MP_DEFINE_CONST_FUN_OBJ_2`, and use `mp_arg_parse_all` to parse the kw arguments.

amirgon, 2020-01-17:
Merged.
Thank you!

closed 2020-01-17 by amirgon

rockindy, 2020-02-03:
Hi @amirgon,

I found there will has a segmentation fault caused by `free(tft_fb)` when calling `SDL.deinit()`
And if I move the `free(tft_fb)` to here(SDL_monitor.c:212):
```
    if (sdl_quit_qry) {
        monitor_deinit();
        free(tft_fb)
    }
```
The fault will be fixed.

I do not know if this is the correct fix, do you have any idea?
Thanks.

amirgon, 2020-02-03:
> I do not know if this is the correct fix, do you have any idea?

@rockindy - Moving `free(tft_fb)` up from `monitor_deinit` to `monitor_sdl_refr_core` doesn't make any difference logically, so I think that the problem is still there, and just randomly didn't cause a crash.

My guess is that `monitor_flush` is called after `monitor_deinit` was called, which is illegal because `monitor_flush` writes to `tft_fb` which is already freed by `monitor_deinit`.

Could you check this? (by a breakpoint or by printing).  

If this really is the problem, try to add this as the first line of `monitor_flush`:
```c
if (!monitor_active()) return;
```

rockindy, 2020-02-03:
@amirgon Yes, the fault is caused by monitor_flush. I will try your suggestion.
Thanks.