lvgl/lv_binding_micropython #66: ESP32 Micropython version is not printed correctly

amirgon, 2020-01-26:
While lv_micropython is aligned to v1.12, the welcome message still refers to v1.9.4:
```
MicroPython v1.9.4-2155-g627ed9545-dirty on 2020-01-25; ESP32 module (spiram) with ESP32
Type "help()" for more information.
>>> import sys
>>> sys.implementation
(name='micropython', version=(1, 12, 0), mpy=10757)
>>>
```

embeddedt, 2020-02-09:
The JavaScript port is currently displaying as `v1.11` for me.

closed 2020-03-09 by stale[bot]

embeddedt, 2020-07-08:
I believe this issue occurs because the git tags from upstream are not being brought across to `lv_micropython` (i.e. the JS port is still showing `1.9.4-dirty` despite using post-1.12 code). It doesn't appear to affect any functionality so I probably won't spend any time fixing it.

amirgon, 2020-07-08:
@embeddedt maybe we can change the message to display lvgl version instead....

amirgon, 2021-06-02:
Related: https://github.com/micropython/micropython/issues/7286

embeddedt, 2021-06-02:
I adjusted the GitHub action for the JS port to pull tags and now it shows `v1.14-775-g0347aa162`. That looks correct, as there is no 1.15 tag in our repository.

amirgon, 2021-06-02:
> That looks correct, as there is no 1.15 tag in our repository.

I think I aligned dev-v8 to 1.15: https://github.com/lvgl/lv_micropython/commit/070da978e2f2cfa8c160f2295948f0ef7ae56fa0
Maybe I didn't pull the tag itself.

amirgon, 2021-06-02:
I've just pushed [tag v1.15 to lv_micropython](https://github.com/lvgl/lv_micropython/tree/v1.15), hopefully it will print the version correctly now.