lvgl/lv_binding_micropython #69: Can't mount SD card and display at the same time when sharing the same SPI bus

ryanfobel, 2020-02-14:
# Summary

I'm trying to use LittlevGL with an [m5stack](https://m5stack.com/collections/m5-core) development board (esp32-based with an ili9341 display). Both the display and SD card interface share the same SPI bus, and it's not currently possible to initialize either device on an existing SPI bus.

Based on the work of @miketeachman, I was able to produce a workaround using the ILI9341 C module (details [here](https://github.com/sci-bots/m5-lvgl/issues/1)). Basically, this involves removing the SPI initialization code in `disp_spi_init()` and setting the display to communicate in full-duplex instead of half-duplex mode. I have tried to port these changes to the [Pure/Hybrid ili9341 driver](https://github.com/littlevgl/lv_binding_micropython/blob/master/driver/esp32/ili9341.py), but I can't seem to get it work.

# boot.py (v1)

The following `boot.py` is a minimal example of loading a LittlevGL button on the m5stack:

```python
import machine
import os
import ili9341
import lvgl as lv

lv.init()

disp = ili9341.ili9341(spihost=1, miso=19, mosi=23, clk=18, cs=14,
                       dc=27, rst=33, backlight=32, backlight_on=1,
                       hybrid=True, mhz=25, width=320, height=240,
                       colormode=ili9341.ili9341.COLOR_MODE_BGR,
                       rot=ili9341.ili9341.MADCTL_ML, invert=True)

scr = lv.obj()
btn = lv.btn(scr)
btn.align(lv.scr_act(), lv.ALIGN.CENTER, 0, 0)
label = lv.label(btn)
label.set_text("Button")
lv.scr_load(scr)
````

This produces a blue button on a white screen:

[image]

# boot.py (v2): mount SD card before initializing display

This is the same `boot.py`, but it tries to mount an SD card prior to initializing the display:

```python
import machine
import os
import ili9341
import lvgl as lv

# Mount the SD card (note that this initializes the SPI bus)
os.mount(machine.SDCard(slot=3, sck=18, mosi=23, miso=19, cs=4), '/sd')

lv.init()

disp = ili9341.ili9341(spihost=1, miso=19, mosi=23, clk=18, cs=14,
                       dc=27, rst=33, backlight=32, backlight_on=1,
                       hybrid=True, mhz=25, width=320, height=240,
                       colormode=ili9341.ili9341.COLOR_MODE_BGR,
                       rot=ili9341.ili9341.MADCTL_ML, invert=True)

scr = lv.obj()
btn = lv.btn(scr)
btn.align(lv.scr_act(), lv.ALIGN.CENTER, 0, 0)
label = lv.label(btn)
label.set_text("Button")
lv.scr_load(scr)
````

As expected, this fails with the error:

```python
E (420) spi: SPI2 already claimed by spi master.
E (420) spi_master: spi_bus_initialize(242): host already in use
Traceback (most recent call last):
  File "boot.py", line 15, in <module>
  File "ili9341.py", line 113, in __init__
  File "ili9341.py", line 326, in init
  File "ili9341.py", line 196, in disp_spi_init
RuntimeError: Failed initializing SPI bus
```

# Attempted workaround

I tried modifying the pure/hybrid ili9341 driver by removing the SPI initialization code in `disp_spi_init()` and setting the display to communicate in full-duplex mode instead of half-duplex (the things that seemed to resolve the issue for the [ILI9341 C module](https://github.com/sci-bots/m5-lvgl/issues/1)). The changes can be [seen here](https://github.com/sci-bots/lv_binding_micropython/commit/686e42b9ba20307163a4b89208d70efed5fac8a2). When I tried testing this modified driver with the `boot.py (v2)` from above, it resulted in the following error:

```python
E (930) spi_master: check_trans_valid(1103): txdata transfer > host maximum
```

I'm not really sure what else to try. I'm happy to have a workaround using the C module, but it looks like the C module is being deprecated in favor of the pure/hybrid python driver, so if possible, I'd like to find a fix for that too. It would also be nice to be able to decide at run time whether you want to attach to a new or pre-initialized SPI bus.

amirgon, 2020-02-14:
Hi @ryanfobel !

> it looks like the C module is being deprecated in favor of the pure/hybrid python driver

That's true. I'm currently using and supporting only the micropython driver. It provides good performance (when using it in the hybrid mode), it's more flexible than the C driver and easier to maintain.

> setting the display to communicate in full-duplex instead of half-duplex mode

The reason I chose half duplex mode - is performance.
[According to the docs](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#known-issues), Full-duplex transactions are not compatible with the dummy bit workaround, hence the frequency is limited.  
Half duplex allows 80MHz on the dedicated SPI pins and 40MHz on GPIO-matrix-routed pins while full-duplex transfers only support speeds up to 26MHz. (see [spi_bus_add_device](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#_CPPv418spi_bus_add_device17spi_host_device_tPK29spi_device_interface_config_tP19spi_device_handle_t)).

This is important for high enough frame-rates, especially when using animations. In case of ili9341 we only care about writing, so half duplex makes sense.

> It would also be nice to be able to decide at run time whether you want to attach to a new or pre-initialized SPI bus

This is already supported.  
The ili9341 driver is [built to work with shared SPI bus](
https://github.com/littlevgl/lv_binding_micropython/blob/49e5b6562f566995030d241655be2c02f4154a54/driver/esp32/ili9341.py#L182-L196). If either `miso`, `mosi` or `clk` are set to `-1`, the bus is not initialized and the driver assumes it was already initialized by another driver.
I'm using this technique today to use ili9341 with [xpt2046 touchscreen driver](https://github.com/littlevgl/lv_binding_micropython/blob/master/driver/esp32/xpt2046.py) on the same bus. You can see that default miso/mosi/clk values of xpt2046 [are -1](https://github.com/littlevgl/lv_binding_micropython/blob/49e5b6562f566995030d241655be2c02f4154a54/driver/esp32/xpt2046.py#L19), so ili9341 initializes the SPI bus and xpt2046 uses the pre-initialized bus, but it can also work the other way around.  

All drivers that share the same SPI bus should be either half-duplex or full-duplex.  
Can you use your SD card driver in half-duplex mode?  
If you can, try to pass `-1` to `miso`/`mosi`/`clk` on the ili9341 driver.

> When I tried testing this modified driver with the `boot.py (v2)` from above, it resulted in the following error:
> 
> ```python
> E (930) spi_master: check_trans_valid(1103): txdata transfer > host maximum
> ```

As I mentioned above, full duplex limits performance.  
Try to set a lower SPI freq. Set `mhz=20` on ili9341 constructor, for example.
If it works for you, we can add another parameter to ili9341 driver to let the user select half/full duplex.  
Feel free to send a PR!

ryanfobel, 2020-02-14:
Hi @amirgon. Thanks for your reply and for all of your work on the micropython bindings for LittlevGL; I'm finding them super useful!

I tried running the following code based on your hint to set `mosi=-1` and `mhz=20`:

```python
import os
import machine

import lvgl as lv
import ili9341

os.mount(machine.SDCard(slot=3, sck=18, mosi=23, miso=19, cs=4), '/sd')

lv.init()
disp = ili9341.ili9341(spihost=1, mosi=-1, cs=14, dc=27,
              rst=33, backlight=32, mhz=20, backlight_on=1,
              hybrid=True, width=320, height=240,
              colormode=ili9341.ili9341.COLOR_MODE_BGR,
              rot=ili9341.ili9341.MADCTL_ML)
disp.send_cmd(0x21)

scr = lv.obj()
label = lv.label(scr)
label.set_text("Hello World!")
label.align(lv.scr_act(), lv.ALIGN.CENTER, 0, 0)
lv.scr_load(scr)

print(os.listdir('/sd'))
```

When I run the above code against the default micropython build (i.e., with the ili9341 driver initialized using half-duplex SPI), I get the following errors:

```
E (990) sdmmc_cmd: sdmmc_read_sectors_dma: sdmmc_send_cmd returned 0x107
['RRaA']
E (1000) spi_master: check_trans_valid(1103): txdata transfer > host maximum
```

If I rebuild micropython with the ili9341 driver initialized using full-duplex mode, the first error disappears, but I still get the second one:

```
E (1000) spi_master: check_trans_valid(1103): txdata transfer > host maximum
```

I couldn't find any way to mount the SD card on a pre-initialized SPI bus, but maybe that's something I should ask about on the micropython forums.

amirgon, 2020-02-14:
> If I rebuild micropython with the ili9341 driver initialized using full-duplex mode, the first error disappears, but I still get the second one:
> 
> ```
> E (1000) spi_master: check_trans_valid(1103): txdata transfer > host maximum
> ```

It looks like [DMA size is too large](https://github.com/espressif/esp-idf/blob/605da33c3d4fa93c6c98da2622f053ecbe5cfbf4/components/driver/spi_master.c#L809).
Try decreasing it by setting `factor=8` or `factor=16` etc. on ili9341 constructor.  

The problem is that the SD card driver initializes the bus with small DMA buffer size. If you can't configure the buffer size on the SD card driver you'd have to use smaller buffers on the ili9341 driver (= higher factors). This will impact performance but save some RAM.

ryanfobel, 2020-02-14:
I tried `factor=8`, `factor=16` and `factor=32` and I get the same error message in every case.

amirgon, 2020-02-15:
> I tried `factor=8`, `factor=16` and `factor=32` and I get the same error message in every case.

I'm not really sure why this happens.
If you have the time and resources, you could try building lv_micropython in DEBUG configuration and debugging it ([with JTAG](https://github.com/amirgon/ESP32-JTAG)).

Did you see this issue: https://github.com/espressif/esp-idf/issues/1597 ?

> Due to the limited amount of IO pins on ESP32, it is desirable to share a SPI bus used for SD cards with other SPI devices such as a LCD display. This is currently not possible with esp-idf since the sdspi_host driver assumes the SD card will be the only device on the SPI bus.

This problem was fixed in https://github.com/espressif/esp-idf/commit/067f3d21c9f1f3bc76361a0e3cb656216f2583ff, but that commit is **not** included yet in Micropython (at least not on the last release v1.12, which lv_micropython is aligned to).  
If you want, you can try building lv_micropython with a more up to date esp-idf that includes this fix, but you'll probably need to change the SD driver to use  the new `sdspi_host_init_device`, `sdspi_host_remove_device` functions.  

Another thing to try on the ili9341 driver is sending SPI transactions exclusively by [Bus Acquiring](https://docs.espressif.com/projects/esp-idf/en/latest/api-reference/peripherals/spi_master.html#bus-acquiring).

I'm not sure why this is working at all with the C driver, but not with the micropython driver. Did you make sure it is working stably?  
It might be related to the fact the the C driver is not really doing DMA in the background, like the micropython driver does. Instead, [it sends the request and waits for it to complete](https://github.com/littlevgl/lv_binding_micropython/blob/49e5b6562f566995030d241655be2c02f4154a54/driver/esp32/modILI9341.c#L208-L211). While very inefficient, it might prevent other devices from accessing the SPI bus while in progress.

closed 2020-03-14 by stale[bot]

favnec5, 2021-08-28:
Hello guys,
Do you have a pure micropython solution to resolve with 1.14 ?
Same problem to my Core2...

amirgon, 2021-08-28:
> Do you have a pure micropython solution to resolve with 1.14 ?
> Same problem to my Core2...

Hi @favnec5 !

I don't think there was any progress on this.  
Personally I don't use SD card on my projects so I was never blocked by this problem.  

Since I can't debug this, I was hoping someone from the LVGL community would face it and would want to invest some time exploring it.

I don't think we fully understand the problem yet, specifically why the C driver works where the Python driver doesn't.  
Here are some related forum threads regarding this problem:
- https://forum.lvgl.io/t/lvgl-and-sd-card-wont-share-spi-bus/2268
- https://forum.lvgl.io/t/not-able-to-pre-initialize-spi-bus-for-ili9341/2474
- https://forum.lvgl.io/t/once-more-esp32-ili9341-xpt2046-sdcard-on-same-spi/3924

This might also be related: https://github.com/lvgl/lv_binding_micropython/issues/80

So if you (or anyone else) would like to explore this - I'll be happy to re-open this issue.

favnec5, 2021-08-30:
Hi amirgon,
I think we have to rewrite the sdCard driver but added into illixxx.py with esp idf module functions to resolve this problem. I don’t Know if it’s possible. I’m searching.

favnec5, 2021-08-31:
Sorry, i don't think it's possible.
On my 1.14 lv_micropython, the esp-idf api don't have sdmmc functions :

```
espidf module
ADC1_CHANNEL
 | MAX
 | _0
 | _1
 | _2
 | _3
 | _4
 | _5
 | _6
 | _7
ADC2_CHANNEL
 | MAX
 | _0
 | _1
 | _2
 | _3
 | _4
 | _5
 | _6
 | _7
 | _8
 | _9
ADC_ATTEN
 | DB_0
 | DB_11
 | DB_2_5
 | DB_6
 | MAX
ADC_CHANNEL
 | MAX
 | _0
 | _1
 | _2
 | _3
 | _4
 | _5
 | _6
 | _7
 | _8
 | _9
ADC_ENCODE
 | MAX
 | _11BIT
 | _12BIT
ADC_I2S_DATA_SRC
 | ADC
 | IO_SIG
 | MAX
ADC_UNIT
 | ALTER
 | BOTH
 | MAX
 | _1
 | _2
ADC_WIDTH
 | BIT_10
 | BIT_11
 | BIT_12
 | BIT_9
 | MAX
BUSY
CANCEL
CHIP
 | ESP32
C_Pointer
[... 1139 lines cut ...]
 | SIZE
 | cast
 | cast_instance
tcpip_adapter_sta_input
tcpip_adapter_sta_list_t
 | SIZE
 | cast
 | cast_instance
tcpip_adapter_sta_start
tcpip_adapter_stop
tcpip_adapter_test_start
tcpip_adapter_up
uxTaskPriorityGet
vPortCleanUpTCB
wifi_sta_info_t
 | SIZE
 | cast
 | cast_instance
wifi_sta_list_t
 | SIZE
 | cast
 | cast_instance
xPortGetCoreID
xt_clock_freq
```

amirgon, 2021-08-31:
@embeddedt  - I didn't know this `<details>   <summary>` Markdown trick - cool!

@favnec5 - I'm not sure I understand. Doesn't Micropython [has its own implementation for sdcard driver](https://github.com/micropython/micropython/blob/master/drivers/sdcard/sdcard.py)?  
Why are you looking for esp-idf implementation?

favnec5, 2021-08-31:
Sure, but sdcard.py use internal micropython machine.SPI function. it do not support -1 value into mosi/miso/cl.
I think re-create sdcard.py driver with native espidf module to implement device addition on the same SPI like the XPT2046.py driver could be a solution. But it's not easy for me and sdspi functions are not implemented.

amirgon, 2021-09-01:
> On my 1.14 lv_micropython, the esp-idf api don't have sdmmc functions :

espidf is a generated module.
It's possible to add new APIs from ESP-IDF into it by editing [espidf.h](https://github.com/lvgl/lv_binding_micropython/blob/f0407d8f446a45b9e1809dc94a4c4fb4587f2378/driver/esp32/espidf.h#L100-L130).  
If you think sdmmc functions in espidf module could help, you can try adding the relevant includes there.

The tricky part is how to **not** include the whole ESP-IDF because of dependent includes. This is done by [defining include guards ](https://github.com/lvgl/lv_binding_micropython/blob/f0407d8f446a45b9e1809dc94a4c4fb4587f2378/driver/esp32/espidf.h#L11-L39) and [filtering out include files explicitly in CMake](https://github.com/lvgl/lv_binding_micropython/blob/f0407d8f446a45b9e1809dc94a4c4fb4587f2378/mkrules.cmake#L137-L145).

favnec5, 2021-10-03:
Hi amirgon,
i'm happy to report you that i found the solution to resolve this problem.
In fact, it was possible to create pure SPI conection for the sdcard.py driver to work at the same time the SPI ILIXXX driver.
I writed a specific sdcard.py version to do that :
```
from ili9XXX import ili9341
lcd = ili9341(mosi=23, miso=38, clk=18, dc=15, cs=5, invert=True, rot=0x10, width=320, height=240,
              rst=-1, power=-1, backlight=-1, half_duplex=False)

import sdcard
spi=sdcard.SPI(mosi=-1, miso=-1, clk=-1)
sd = sdcard.SDCard(spi, Pin(4))
os.mount(sd, "/sd")
print(os.listdir('/sd'))
```
maybe you would be interested in testing this?

reguard,
Thomas

amirgon, 2021-10-03:
> i'm happy to report you that i found the solution to resolve this problem.

That's great! 

Could you open a PR on lv_binding_micropython to add your `sdcard.py` under `driver/esp32/`?

favnec5, 2021-10-04:
Sure ! :-)