lvgl/lv_binding_micropython #70: gen_mpy.py and IDF v4.0

brotherdust, 2020-02-24:
Hi there! First of all, thanks for making this sweet library! Very useful!
I've been having some problems getting the MP bindings to work with IDF 4.0 and MP 1.12.

Using the following command in the MP directory:
```
make -C ports/esp32 LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" BOARD=GENERIC PYTHON=python
```
I get the following result:
```
...
CC ../../lib/lv_bindings/driver/generic/modlvindev.c
CC build-GENERIC/lvgl/lv_mpy.c
build-GENERIC/lvgl/lv_mpy.c: In function 'get_native_obj':
build-GENERIC/lvgl/lv_mpy.c:98:12: error: implicit declaration of function 'mp_instance_cast_to_native_base'; did you mean 'mp_obj_cast_to_native_base'? [-Werror=implicit-function-declaration]
     return mp_instance_cast_to_native_base(mp_obj, MP_OBJ_FROM_PTR(native_type));
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
            mp_obj_cast_to_native_base
build-GENERIC/lvgl/lv_mpy.c:98:12: error: returning 'int' from a function with return type 'mp_obj_t' {aka 'void *'} makes pointer from integer without a cast [-Werror=int-conversion]
     return mp_instance_cast_to_native_base(mp_obj, MP_OBJ_FROM_PTR(native_type));
            ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
../../py/mkrules.mk:47: recipe for target 'build-GENERIC/build-GENERIC/lvgl/lv_mpy.o' failed
make: *** [build-GENERIC/build-GENERIC/lvgl/lv_mpy.o] Error 1
make: Leaving directory '/home/jarad/mp-m5stack/micropython/ports/esp32'
```
Since the compiler suggested this:
```
did you mean 'mp_obj_cast_to_native_base'?
```
I figured I'd give it a try. So, modified gen_mpy.py on line 520:
```
return mp_obj_cast_to_native_base(mp_obj, MP_OBJ_FROM_PTR(native_type));
```
Ran 
```
make -C ports/esp32 BOARD=GENERIC clean
make -C ports/esp32 LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" BOARD=GENERIC PYTHON=python
```
Yay! Compiles! I reboot the ESP32 (an M5Stack):
```
rst:0x1 (POWERON_RESET),boot:0x17 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:5108
load:0x40078000,len:12856
load:0x40080400,len:3480
entry 0x40080638
I (602) cpu_start: Pro cpu up.
I (602) cpu_start: Application information:
I (602) cpu_start: Compile time:     Feb 23 2020 20:00:47
I (606) cpu_start: ELF file SHA256:  0000000000000000...
I (612) cpu_start: ESP-IDF:          v4.0
I (616) cpu_start: Starting app cpu, entry point is 0x40083538
I (0) cpu_start: App cpu up.
I (627) heap_init: Initializing. RAM available for dynamic allocation:
I (634) heap_init: At 3FFAFF10 len 000000F0 (0 KiB): DRAM
I (640) heap_init: At 3FFB6388 len 00001C78 (7 KiB): DRAM
I (646) heap_init: At 3FFB9A20 len 00004108 (16 KiB): DRAM
I (652) heap_init: At 3FFBDB5C len 00000004 (0 KiB): DRAM
I (658) heap_init: At 3FFD39C0 len 0000C640 (49 KiB): DRAM
I (664) heap_init: At 3FFE0440 len 00003AE0 (14 KiB): D/IRAM
I (670) heap_init: At 3FFE4350 len 0001BCB0 (111 KiB): D/IRAM
I (677) heap_init: At 4009F784 len 0000087C (2 KiB): IRAM
I (683) cpu_start: Pro cpu start user code
I (701) spi_flash: detected chip: gd
I (701) spi_flash: flash io: dio
I (701) cpu_start: Starting scheduler on PRO CPU.
I (0) cpu_start: Starting scheduler on APP CPU.
MicroPython v1.12-196-g5349e0cde-dirty on 2020-02-23; ESP32 module with ESP32
Type "help()" for more information.
>>>
```
And load the following code:
```
from ili9341 import ili9341
import lvgl as lv

class M5ili9341(ili9341):
    def __init__(
            self, mosi=23, miso=19, clk=18, cs=14, dc=27, rst=33, backlight=32,
            backlight_on=1, hybrid=False, width=320, height=240, factor=2,
            colormode=ili9341.COLOR_MODE_BGR, rot=ili9341.MADCTL_ML, invert=True, **kwargs):
        super().__init__(
            mosi=mosi, miso=miso, clk=clk, cs=cs, dc=dc, rst=rst,
            backlight=backlight, backlight_on=backlight_on, hybrid=hybrid,
            width=width, height=height, colormode=colormode, rot=rot,
            invert=False, **kwargs)
        if invert:
            # Invert colors (work around issue with `invert` kwarg in stock
            # class).
            self.send_cmd(0x21)

lv.init()
disp = M5ili9341()
```
And... nope!
```
ILI9341 initialization completed
Enable backlight
Traceback (most recent call last):
  File "<stdin>", line 20, in <module>
  File "<stdin>", line 9, in __init__
  File "ili9341.py", line 125, in __init__
RuntimeError: Not enough DMA-able memory to allocate display buffer
```

Any thoughts on how this can be resolved? Any help would be appreciated!

amirgon, 2020-02-25:
Hi @brotherdust !

> get the following result:
>
> ```
> ...
> CC ../../lib/lv_bindings/driver/generic/modlvindev.c
> CC build-GENERIC/lvgl/lv_mpy.c
> build-GENERIC/lvgl/lv_mpy.c: In function 'get_native_obj': [...]

Are you using [lv_micropython](https://github.com/littlevgl/lv_micropython) or did you port lv_binding_micropython to your own micropython fork?

Have a look at lv_micropython's [py/obj.h](https://github.com/littlevgl/lv_micropython/blob/6b32fae730fec8aacb28c2797bd7c542aba60cbb/py/obj.h#L675):
```c
mp_obj_t mp_instance_cast_to_native_base(mp_const_obj_t self_in, mp_const_obj_t native_type);
```

lv_micropython is aligned to latest micropython **release**, which is v1.12.  
Your code is not aligned to it.  

Among many changes done after v1.12 was released, `mp_instance_cast_to_native_base` changed to `mp_obj_cast_to_native_base` on micropython master **head**:
```
$ git diff v1.12 micropython/master -- py/obj.h | grep mp_instance_cast_to_native_base --context=1
 bool mp_obj_is_subclass_fast(mp_const_obj_t object, mp_const_obj_t classinfo); // arguments should be type objects
-mp_obj_t mp_instance_cast_to_native_base(mp_const_obj_t self_in, mp_const_obj_t native_type);
+mp_obj_t mp_obj_cast_to_native_base(mp_obj_t self_in, mp_const_obj_t native_type);
```

I recommend sticking to micropython versions. When a new version comes out I'm aligning lv_micropython to it.  
You can, of course, take a different version of micropython and build it with lvgl, but then you will need to integrate it by your own, and handle yourself problems like this one and possibly others.

> And... nope!
> 
> ```
> ILI9341 initialization completed
> Enable backlight
> Traceback (most recent call last):
>   File "<stdin>", line 20, in <module>
>   File "<stdin>", line 9, in __init__
>   File "ili9341.py", line 125, in __init__
> RuntimeError: Not enough DMA-able memory to allocate display buffer
> ```

This is unrelated to your previous problem.
Do you have enough RAM? Perhaps you really don't have enough DMA-able RAM?
Also, what's `M5ili9341`?  

The supported driver is [ili9341.py](https://github.com/littlevgl/lv_binding_micropython/blob/master/driver/esp32/ili9341.py), it can be configured to use single buffer or double buffer, and buffer size can be controlled.  
You might want to consider switching to this driver to get the best control and performance.

brotherdust, 2020-02-26:
@amirgon , thank you for your considered reply. I was using vanilla Micropython master-HEAD. The device is an M5Stack Core; bottom line: not enough memory. It doesn't have SPIRAM module, so just the 560Kb it comes with. Not enough for LV and WiFi, even with single-buffer and trying different driver options. Sad. That's OK, though. Thanks again!

closed 2020-02-26 by brotherdust

embeddedt, 2020-02-26:
>  It doesn't have SPIRAM module, so just the 560Kb it comes with. Not enough for LV and WiFi

That's odd. For LittlevGL (without MicroPython) our minimum is 16KB of RAM. How much RAM is the WiFi code using?

brotherdust, 2020-02-26:
Honestly I don’t know how to figure that out. If you can show me how, I’ll gladly retrieve it for you. What I can tell you is that it’s the same issue whether I use the provided lv_micropython binaries or something I’ve compiled myself (including IDF 3.3 and 4.0).

brotherdust, 2020-02-26:
Apologies. I thought I remembered trying a lv_micropython binary, but was mistaken.

amirgon, 2020-02-26:
> bottom line: not enough memory. It doesn't have SPIRAM module, so just the 560Kb it comes with. Not enough for LV and WiFi, even with single-buffer and trying different driver options. Sad. That's OK, though.

@brotherdust I think you gave up too fast.  
560KB is **a lot**. It's more than enough for Micropython+lvgl+display driver+wifi.
It's possible, of course, to consume all this RAM by misconfiguring any of these components or by consuming it in other ways.

There are several people (@ryanfobel, @cfobel, @endtwist and others) who already have some experience working with micropython+lvgl on M5Stack, perhaps you could consult with them, or study their forks.

brotherdust, 2020-02-26:
Challenge accepted! Thank you!