lvgl/lv_binding_micropython #71: ESP32 WiFi + LV: "Not enough DMA-able memory to allocate display buffer"

brotherdust, 2020-02-26:
@amirgon 
In reference to issue #70 , I did as you asked and reviewed the available ports I could find for M5Stack. Unfortunately, I couldn't find any particular IDF or MP optimizations or configuration changes that would allow me to have enough stack space for both WiFi and LV.

Please have a look at [this commit](https://github.com/sci-bots/micropython/commit/47de8b5d38721c5132de28efbefa2f4e2d3395fd#diff-04c6e90faac2675aa89e2176d2eec7d8) specifically. According to the build command, it would seem that SPIRAM is implicitly expected (the change in questions was even authored by you).

At this point, I have compiled and tested the 'vanilla' lv_micropython and the one authored by @ryanfobel. Same result: `not enough DMA-able memory to allocate display buffer.`

I have also attempted to initialize the driver with `double_buffer=False` as well as `hybrid=False`.

Do you have _any_ recommendations on available IDF SDK or MP optimizations that would make this work? According to you, this should theoretically work, but I don't have enough experience with MP internals to get a good idea as to what is hogging so much memory.

Thank you kindly!

brotherdust, 2020-02-27:
OK!! I think... I think I got it working! I ended up adapting [sdkconfig.defaults from 
MicroPython_ESP32_psRAM_LoBo](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/sdkconfig.defaults). Here's the final `sdkconfig`:
```
#
# Default and recommended options for MicroPython build
#

# Application manager
CONFIG_APP_EXCLUDE_PROJECT_VER_VAR=y
CONFIG_APP_EXCLUDE_PROJECT_NAME_VAR=y
#
# ESP Tool & Flash
#
CONFIG_ESPTOOLPY_BAUD_921600B=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_AFTER_NORESET=y
CONFIG_ESPTOOLPY_FLASHMODE="dio"
CONFIG_ESPTOOLPY_FLASHFREQ_40M=y
CONFIG_ESPTOOLPY_FLASHFREQ="40m"
CONFIG_FLASHMODE_DIO=y

#
# Bootloader config
#
CONFIG_LOG_BOOTLOADER_LEVEL_NONE=
CONFIG_LOG_BOOTLOADER_LEVEL_ERROR=
CONFIG_LOG_BOOTLOADER_LEVEL_WARN=y
CONFIG_LOG_BOOTLOADER_LEVEL_INFO=y
CONFIG_LOG_BOOTLOADER_LEVEL_DEBUG=
CONFIG_LOG_BOOTLOADER_LEVEL_VERBOSE=
CONFIG_LOG_BOOTLOADER_LEVEL=2

#
# ESP32 specific
#

CONFIG_ESP32_DEFAULT_CPU_FREQ_240=y
CONFIG_ESP32_XTAL_FREQ_AUTO=y
CONFIG_ESP32_XTAL_FREQ=0
CONFIG_ESP32_PANIC_PRINT_HALT=y
CONFIG_ESP32_PHY_INIT_DATA_IN_PARTITION=n
CONFIG_TASK_WDT_TIMEOUT_S=15

#
# SPI RAM config
#
# CONFIG_SPIRAM_BOOT_INIT=y
# CONFIG_SPIRAM_IGNORE_NOTFOUND=y
# CONFIG_SPIRAM_USE_MEMMAP=
# CONFIG_SPIRAM_USE_CAPS_ALLOC=y
# CONFIG_SPIRAM_USE_MALLOC=
# CONFIG_SPIRAM_TYPE_ESPPSRAM32=y
# CONFIG_SPIRAM_SIZE=4194304
# CONFIG_SPIRAM_SPEED_40M=y
# CONFIG_SPIRAM_SPEED_80M=
# CONFIG_SPIRAM_MEMTEST=y
# CONFIG_SPIRAM_CACHE_WORKAROUND=y
# CONFIG_WIFI_LWIP_ALLOCATION_FROM_SPIRAM_FIRST=y

#
# Log output
#
CONFIG_LOG_DEFAULT_LEVEL_NONE=
CONFIG_LOG_DEFAULT_LEVEL_ERROR=
CONFIG_LOG_DEFAULT_LEVEL_WARN=y
CONFIG_LOG_DEFAULT_LEVEL_INFO=y
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=
CONFIG_LOG_DEFAULT_LEVEL_VERBOSE=
CONFIG_LOG_DEFAULT_LEVEL=2
CONFIG_LOG_COLORS=y

#
# FreeRTOS
#
CONFIG_FREERTOS_ISR_STACKSIZE=4096
CONFIG_FREERTOS_CHECK_STACKOVERFLOW_CANARY=y
CONFIG_FREERTOS_THREAD_LOCAL_STORAGE_POINTERS=4
CONFIG_FREERTOS_ASSERT_ON_UNTESTED_FUNCTION=n
CONFIG_FREERTOS_HZ=1000
CONFIG_TIMER_TASK_STACK_DEPTH=2048
CONFIG_SUPPORT_STATIC_ALLOCATION=y
CONFIG_ENABLE_STATIC_TASK_CLEAN_UP_HOOK=y

#
# lwip
#
CONFIG_LWIP_SO_REUSE=y
CONFIG_LWIP_MAX_SOCKETS=4
CONFIG_TCPIP_TASK_STACK_SIZE=4096
CONFIG_PPP_SUPPORT=y
CONFIG_PPP_PAP_SUPPORT=y
CONFIG_PPP_CHAP_SUPPORT=y

#
# WiFi
#
CONFIG_ESP32_WIFI_STATIC_RX_BUFFER_NUM=8
CONFIG_ESP32_WIFI_DYNAMIC_RX_BUFFER_NUM=16
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER=y
CONFIG_ESP32_WIFI_DYNAMIC_TX_BUFFER_NUM=16

#
# mbedTLS
#
CONFIG_MBEDTLS_SSL_MAX_CONTENT_LEN=8192

#
# FatFS
#
CONFIG_FATFS_LFN_HEAP=y
CONFIG_FATFS_MAX_LFN=127
CONFIG_FATFS_CODEPAGE_437=y
CONFIG_FATFS_CODEPAGE=437
CONFIG_FATFS_PER_FILE_CACHE=n

#
# Wear Levelling
#
CONFIG_WL_SECTOR_SIZE_512=y
CONFIG_WL_SECTOR_SIZE_4096=
CONFIG_WL_SECTOR_SIZE=512
CONFIG_WL_SECTOR_MODE_PERF=
CONFIG_WL_SECTOR_MODE_SAFE=y
CONFIG_WL_SECTOR_MODE=1

#
# SPIFFS Configuration
#
CONFIG_SPIFFS_MAX_PARTITIONS=1
CONFIG_SPIFFS_OBJ_NAME_LEN=64

#
# Partition Table
#
# CONFIG_PARTITION_TABLE_CUSTOM=y
# CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions.csv"
# CONFIG_PARTITION_TABLE_CUSTOM_APP_BIN_OFFSET=0x10000
# CONFIG_PARTITION_TABLE_CUSTOM_PHY_DATA_OFFSET=0xf000
# CONFIG_PARTITION_TABLE_FILENAME="partitions.csv"
# CONFIG_APP_OFFSET=0x10000
# CONFIG_PHY_DATA_OFFSET=0xf000
```
Hopefully this helps out anyone out there trying to do the same with an M5Stack.

@amirgon , thanks again for your assistance in this matter and for challenging me to keep digging!

closed 2020-02-27 by brotherdust

amirgon, 2020-02-27:
> Please have a look at [this commit](https://github.com/sci-bots/micropython/commit/47de8b5d38721c5132de28efbefa2f4e2d3395fd#diff-04c6e90faac2675aa89e2176d2eec7d8) specifically. According to the [...]

That's just how I build lv_micropython. It was not meant to imply SPIRAM is a must.

> OK!! I think... I think I got it working! I ended up adapting [sdkconfig.defaults from MicroPython_ESP32_psRAM_LoBo](https://github.com/loboris/MicroPython_ESP32_psRAM_LoBo/blob/master/MicroPython_BUILD/sdkconfig.defaults). Here's the final `sdkconfig`

Well done. Thanks for the update!  

A few questions:

- Do you have an idea what in this configuration helped?
I would guess it's something in the lwip/WiFi/mbedTLS configuration.

- Which display driver did you end up using eventually?
Do you use double buffering? Did you change [other parameters](https://github.com/littlevgl/lv_binding_micropython/blob/49e5b6562f566995030d241655be2c02f4154a54/driver/esp32/ili9341.py#L48-L52), such as `factor` which controls the display buffer size?

- Another thing - how much RAM are you have with left after all this?
It's easy to check like this:
```python
>>> import gc
>>> gc.collect();gc.mem_free()
1032048
```

ryanfobel, 2020-03-06:
I can confirm that [this binary](https://github.com/sci-bots/micropython/releases/tag/v1.11-877-gbf7699eee-m5stack) release works with the m5stack black and gray cores.

```python
>>> import gc
>>> gc.collect()
>>> gc.mem_free()
106000
```