lvgl/lv_binding_micropython #73: More detailed info from help()

kisvegabor, 2020-04-06:
Hi,

I remember that we have talked about improving help() to tell more info about functions but I don't remember if it was implemented or not. E.g. tell the type of the parameters and return value for `help(lv.btn)`.

If implemented can you tell how it can be used?

amirgon, 2020-04-06:
Extraction of information (meta-data) about a program is called Introspection or Reflection.
In Micropython, Introspection is only partially supported.  

Since Micropython is designed for embedded devices with limited resources, it does not make sense to always keep full metadata about the program (such as parameter names, types etc.). That would consume precious memory (either RAM or Flash).

Some introspection is supported though. You can, for example, get a list of attributes of an object, or a list of functions in a module, because this is something Micropython has to maintain anyway, being a dynamic language. (see [`dir`](https://docs.python.org/3/library/functions.html#dir), which is supported in Micropython)

**But you are right - we discussed this in another context: lvgl GUI builder.**

https://github.com/littlevgl/lv_gui_builder/issues/1#issuecomment-528845543

A Micropython GUI builder would need introspection so it could automatically display what are all the setters and getters of a widget, and call them with the right parameters (for example).  
For that purpose I added an option to generate metadata (in the form of a JSON file) that contains information missing from Micropython's reflection.

kisvegabor, 2020-04-07:
Thank you very much!
The JSON file is perfect!

closed 2020-04-07 by kisvegabor