lvgl/lv_binding_micropython #74: Support for ILI9488 driver

albertomercurio, 2020-04-11:
Hello,
i need the support for the ILI9488 driver. How can i proceed?

amirgon, 2020-04-11:
Hi @albertomercurio !

> i need the support for the ILI9488 driver. How can i proceed?

- Start from [ili9341 driver](https://github.com/littlevgl/lv_binding_micropython/blob/master/driver/esp32/ili9341.py).
- Go over the datasheets and identify the differences between IL9341 and ILI9488.  
Resolution is different, initialization is probably different. I'm not sure if there are other differences.
- If differences are small, you can integrate the changes into existing ili9341 driver, and let the user select between IL9341 and ILI9488. If changes are significant, you can create a new driver.  
Another approach is to build it in an object oriented way: common parent class for both, and inherit ili9341 and ili9488 from it.
- After you test it - send us a PR 😉

embeddedt, 2020-04-11:
Our ESP32 project has [some drivers for the ILI displays](https://github.com/littlevgl/lv_port_esp32/tree/master/components/lvgl_esp32_drivers/lvgl_tft). It may be a good starting point.

te-bachi, 2020-04-12:
I don't use MicroPython at the moment, but work with C/C++ (HW: ESP32, ILI9488 over SPI; SW: Arduino, TFT_eSPI, LittlevGL).
If you haven't found everything in @embeddedt post, look at [Bodmer's TFT_eSPI Library](https://github.com/Bodmer/TFT_eSPI)

closed 2020-05-10 by stale[bot]