lvgl/lv_binding_micropython #75: Test with dev-7.0

kisvegabor, 2020-04-17:
Hi,

I've checked if there any Micropython related errors in  dev-7.0. The example command in the README runs normally. 
Here is the result: [gen.c.zip](https://github.com/littlevgl/lv_binding_micropython/files/4491332/gen.c.zip)

Is there anything else to check?

amirgon, 2020-04-17:
I usually search the generated file for `NOT generated` string.  

For every function or struct that fails to generate, the script prints a comment that includes the words "NOT generated".  
If a function was not generated it could hint that something is wrong in its prototype, or there might be some other problem.  

One more thing I think worth to check is the new style system.
Maybe worth writing an example in Micropython that uses it.

I can see, for example, that there are problems with `lv_style_prop_cb_t`, `lv_theme_t_apply_cb_callback`, `lv_draw_mask_common_dsc_t_cb_callback` and some others.

kisvegabor, 2020-04-17:
I see, thank you. I've reviewed the `NOT generated` cases and added `..._xcb_t` where it was required.

> One more thing I think worth to check is the new style system.
Maybe worth writing an example in Micropython that uses it.

I agree! However, I'm not sure one of the cases with the new styles will work correctly. You can store function pointers in the styles (now only for one property). It looks like this:
```c
  lv_style_t style;
  lv_style_init(&style);
  lv_style_set_transition_path(&style, LV_STATE_DEFAULT, lv_anim_path_linear);
```  
```c
  /**
   * Calculate the current value of an animation applying linear characteristic
   * @param a pointer to an animation
   * @return the current value to set
   */
  lv_anim_value_t lv_anim_path_linear(const lv_anim_t * a);
```

So `lv_style_set_transition_path` gets the style as the first parameter which has no user data, but the function pointer gets the animation which has user data. What do you think about this case?

amirgon, 2020-04-20:
> So `lv_style_set_transition_path` gets the style as the first parameter which has no user data, but the function pointer gets the animation which has user data. What do you think about this case?

This cannot work correctly the way it's defined.

The callback registration function `lv_style_set_transition_path` receives `lv_style_t`, and tries to store the python callback on `user_data` of `lv_style_t`.  
If `lv_style_t` has no `user_data`, **the compilation will fail**.  

Even if `lv_style_t` had a `user_data` field, the callback receives **a different object** as its parameter - `lv_anim_t`. It will look for the python callback object on `user_data` of  `lv_anim_t`, and will not find it there.

- Did you try to compile the generated `gen.c` in lv_micropython?
- Could you explain why `lv_style_set_transition_path` was defined that way?
- Can this be changed to obey the callback conventions we agreed upon?

I suspect there are some other problems with the new styles, I'll need to find some time to get into this myself.  
At the moment I'm a bit busy, but eventually I'll find some time for it. Do you already have some "user" code examples that show how to use the new styles?

embeddedt, 2020-04-20:
@amirgon There are some preliminary examples [here](https://docs.littlevgl.com/v7/en/html/overview/style.html). Search for `lv_ex_style`.

kisvegabor, 2020-04-21:
> If lv_style_t has no user_data, the compilation will fail.

This function is not super-critical. In the worst case, it can be simply ignored. (Check if the type of the first parameter is not the same as the callback's first parameter?)

> Did you try to compile the generated gen.c in lv_micropython?

Not yet.

> Could you explain why lv_style_set_transition_path was defined that way?

It's no like an event callback. This callback is used e.g. when the you press a button and configure its style to animate the background color between the released and pressed color. In this case, lvgl creates an animation and uses the `transition_path` as `anim_path`. It describes the "curve" of the animation. So the `transition_path` is saved in the style, read by the library when required and passed to an animation.

> Can this be changed to obey the callback conventions we agreed upon?

I don't think so, because it works differently. It's not really important to allow custom functions here, "just" pass some lvgl built-in functions (e.g. `lv_anim_path_linear`) back to lvgl. Although I don't know if it's easier or not.

amirgon, 2020-04-21:
> This function is not super-critical. In the worst case, it can be simply ignored. (Check if the type of the first parameter is not the same as the callback's first parameter?)

Ignoring it is possible, but let's first see how hard it would be to support it.

> lvgl creates an animation and uses the `transition_path` as `anim_path`. It describes the "curve" of the animation. So the `transition_path` is saved in the style, read by the library when required and passed to an animation.

How about adding `user_data` to the style object? and then adding the style as the first parameter of the callback? This can allow it to work with our callback conventions.

> "just" pass some lvgl built-in functions (e.g. `lv_anim_path_linear`) back to lvgl. Although I don't know if it's easier or not.

Passing lvgl C function pointers as callback is already supported.  
The binding keeps a pointer to every lvgl function, and it can be used with callbacks.  
For example, today it's possible to do something like this:

```c
lv.anim_set_path_cb(anim, lv.anim_path_ease_in)
```

here `anim_path_ease_in` is a C function pointer which is passed as a callback.

kisvegabor, 2020-04-21:
> How about adding user_data to the style object? and then adding the style as the first parameter of the callback? This can allow it to work with our callback conventions.

Adding user data to styles doesn't make sense in general because the user never gets them in functions as a parameter or never should use them for anything then adding them to an object. However, if it'd help in the case of Micropython we can have a define for it. 

Basically the animations and styles are unrelated (it's just possible to animate styles too), so adding a style pointer to the path callback would be quite messy. 

> lv.anim_set_path_cb(anim, lv.anim_path_ease_in)

Is it still expects to have a user data field in `anim` or just passes the pointer to `lv_anim_set_path_cb` C function?

amirgon, 2020-04-21:
> Basically the animations and styles are unrelated (it's just possible to animate styles too), so adding a style pointer to the path callback would be quite messy.

Another option is to pass `user_data` directly, on both `lv_style_set_transition_path` and `lv_anim_path_linear` (and the other transition function).

> Is it still expects to have a user data field in `anim` or just passes the pointer to `lv_anim_set_path_cb` C function?

In general it expects `user_data`, because it supports both options - python callback or C callback.
If `user_data` is missing, we can change the script to support only C callback.

kisvegabor, 2020-04-21:
> Another option is to pass user_data directly, on both lv_style_set_transition_path and lv_anim_path_linear (and the other transition function).

How would the prototype of the functions look like in this case?

> In general it expects user_data, because it supports both options - python callback or C callback.
If user_data is missing, we can change the script to support only C callback.

There are builtin functions for [some common path types](https://github.com/littlevgl/lvgl/blob/dev-7.0/src/lv_misc/lv_anim.h#L331-L379) so it seems acceptable if we end up with this solution. However, I agree to try to solve it in a really Micropython compatible way.

amirgon, 2020-04-21:
> > Another option is to pass user_data directly, on both lv_style_set_transition_path and lv_anim_path_linear (and the other transition function).
> 
> How would the prototype of the functions look like in this case?

- A parameter called `void * user_data` should be provided to the registration function as the last argument.
- The callback itself should receive `void *` as the last argument
- The registration function should save the user_data next to the callback function pointer, and provide it to the callback every time it is called.

kisvegabor, 2020-04-21:
I see, thank you. 
Adding a `void * user_data` to the path callback might make sense because this way user could pass some config. (E.g. the overshoot percentage in case of `lv_anim_path_overshoot`).

We can also have `user_data` in the style which is not enabled by default but enabled for MicroPython. So `lv.style_set_transition_path` can use the style's `user_data`.

When the transition animation is created lvgl could pass the style's user data to the animation, which can pass it to the `path_cb`. However, lvgl can't see from which style does the `path_cb` comes from. There is logic to decide which property should be picked according to the order of styles, state of the object, the inheritance of the property, etc. Lvgl only sees a `path_cb` in the end which was "somehow" figured out.

Besides, objects might have local styles too. In this case, the function looks like this: 
```c
lv_obj_set_style_local_transition_path(obj, LV_BTN_PART_MAIN, LV_STATE_DEFAULT, path_cb);
``` 
So here an `lv_obj_t *` is the first parameter whose `user_data` can be seen by MicroPython. 

Although, it's a different use case of callbacks, it's really far from the callback conventions we have agreed on.  In light of the above-mentioned difficulties, I think the best we can do is allow only the built-in C functions here.

amirgon, 2020-04-21:
> However, lvgl can't see from which style does the `path_cb` comes from. There is logic to decide which property should be picked according to the order of styles, state of the object, the inheritance of the property, etc. Lvgl only sees a `path_cb` in the end which was "somehow" figured out.

@kisvegabor there is something I still don't understand.
You keep the function pointer of the callback somewhere right? You have a way to pass the function pointer from the registering function until the point you call that callback.    

My suggestion was - keep the `user_data` next to it. Instead of keeping the function pointer alone, every time you save it somewhere, or move it somewhere, do the same to the `user_data`.  
Just think of `user_data` as an extension of the function pointer, that goes along with it wherever it goes.  

Could you explain the difficulties with this?

kisvegabor, 2020-04-21:
> Could you explain the difficulties with this?

Now the style properties have 4 simple possible data types:
1. int (border width, padding, etc)
2. opacity (border opacity, bg opacity, etc)
3. color (border color, bg color, etc)
4. pointer (font, path_cb, etc)

To attach the `user_data` to the `path_cb` and save it as a "pointer" two things would be required:
1. Add an extra parameter to `lv_style_set_transition_path` and `lv_obj_set_style_local_transition_path` which can't be used by the user in C. Or it could a config parameter as I mentioned above but then it'd work differently in C and MP. Besides it'd break the similarity of these function.
2. A struct would be required to keep the `user_data` and `cb` together. E.g.
```c
typedef struct {
   void (*cb)(void);
   void * user_data;
}foo_t;
```
To save this struct in the style it needs to be dynamically allocated and its pointer should be saved instead. (And should be freed when the style is deleted). It's certainly possible but I think it can be considered a difficulty.

Maybe, would it be possible to manage it on MP's side? I don't know the internals to give ideas just want to be sure it can't be solved on the "other side". In this case, in C, I'd think in a global list where the `user_data`s and `callback`s are connected. However, I'm don't know if it makes sens in MP.

amirgon, 2020-04-22:
> 1. Add an extra parameter to `lv_style_set_transition_path` and `lv_obj_set_style_local_transition_path` which can't be used by the user in C. Or it could a config parameter as I mentioned above but then it'd work differently in C and MP. Besides it'd break the similarity of these function.

Why can't it be used by the user in C? I think it could be useful in C as well.
For example, suppose the user has a parametric transitions, where the transition itself is configurable (how fast it goes, how many times it blinks etc.).  
Instead of writing a separate function for every possible transition+parameter, why not provide the parameter to the transition, when registering the transition callback function?

> To save this struct in the style it needs to be dynamically allocated and its pointer should be saved instead. (And should be freed when the style is deleted). It's certainly possible but I think it can be considered a difficulty.

Why do you need to allocate it dynamically?  
Instead of keeping a pointer, keep a struct with pointer+user_data, as part of your style or whatever parent struct you are using.

> Maybe, would it be possible to manage it on MP's side? I don't know the internals to give ideas just want to be sure it can't be solved on the "other side". In this case, in C, I'd think in a global [...]

Micropython bindings eventually calls lvgl function.  
What should it pass to lvgl as the callback function pointer parameter? It must be a function, but when the function is called we no longer have the context.  

Suppose the Micropython user registered 3 different transitions by calling the same registration function 3 times. The binding can keep each of the transitions the user has registered.
But when the transition callback is called, which of the 3 registered transitions should we call? That is the context we are losing when passing only a function pointer.  

In cases we must pass only a function pointer without its context, we actually need to **generate a new C function on runtime** which calls our function with its context.
There is a way to achieve this is in C with [FFI](en.wikipedia.org/wiki/Foreign_function_interface), where a function pointer can be generated on runtime effectively creating a [closure](en.wikipedia.org/wiki/Closure_(computer_programming)).  
Unfortunately, embedded system don't always support FFI so we can't rely on it.
For example, [Micropython ESP32 doesn't support FFI](github.com/micropython/micropython-esp32/issues/230#issuecomment-454703767).

kisvegabor, 2020-04-22:
> Why do you need to allocate it dynamically?
Instead of keeping a pointer, keep a struct with pointer+user_data, as part of your style or whatever parent struct you are using.

I'd like to avoid it to keep the number of style parameter types minimal. It's an important goal because the properties have a strict order which allows getting some extra data from their numerical value. See the list [here](https://github.com/littlevgl/lvgl/blob/dev-7.0/src/lv_core/lv_style.h#L91-L200). Seeing the lower 4 bits and checking in which range it is, tells the type. It allows quite compact data packing.

> Micropython bindings eventually calls lvgl function.
What should it pass to lvgl as the callback function pointer parameter? It must be a function, but when the function is called we no longer have the context.

Anyway, I'm sure that you are right with it just I don't know all the difficulties in the binding. So I accept that it's not a good direction. 

---

Actually using function pointers in styles causes issues with C++ too. So it'd be great to get rid of them. So let's try a new approach. What if we had a struct like 
```c
typedef struct {
  void(*cb)(void);
  // more params if required
  void * user_data; // Does it help here?
}lv_style_transition_path_t
```

And use it like this:
```c
  static lv_style_transition_path_t p1;
  p1.cb = lv_anim_path_linear;
  lv_style_set_transition_path(&style, LV_STATE_DEFAULT, &p1);
```

Could it work better with Micropython somehow?

embeddedt, 2020-04-22:
> What if we had a struct like `lv_style_transition_path_t`

I like the idea. I think we can make it more generic (and solve our C++ problems at the same time).

```c
typedef struct {
  void * user_data; // Does it help here?
  void(*cb)(void);
} lv_style_cb_obj_t;
typedef struct {
  lv_style_cb_obj_t path_cb;
  // extra parameters for transitions
} lv_style_transition_path_t;
```

We can then re-use this design in the style system anywhere where we want a function callback.

amirgon, 2020-04-22:
> Could it work better with Micropython somehow?

Almost.  

A struct with the callback and `user_data` is good. It's very much like `user_data` in `lv_anim_t` for example.

But the one thing that is missing is **receiving `lv_style_transition_path_t` pointer as a callback parameter**. This is a must, so that the callback implementation would have access to the callback context. This would be needed for C++ as well.

Something like this:

```c
// Forward declaration
struct _lv_style_transition_path_t;
typedef struct _lv_style_transition_path_t lv_style_transition_path_t;

// Callback prototype
typedef void (*lv_style_transition_path_callback_t)(lv_style_transition_path_t*);

// Struct definition with callbacks and user_data
typedef struct _lv_style_transition_path_t{
  lv_style_transition_path_callback_t *cb;
  // more params if required
  void * user_data;
}lv_style_transition_path_t;
```

In  `lv_anim_t` struct we have several callbacks fields (`exec_cb`, `path_cb`, etc.) and `user_data` field.
The Micropython Bindings script generates C functions for these callbacks, which extract the Python callable objects from `user_data`.

This works well today, and if I understand correctly you suggest doing something similar with `lv_style_transition_path_t`.
If you pass a pointer to the struct that contains `user_data` as the callback parameter, Micropython bindings would support this without any change, it's actually one of the callback conventions we agreed upon.

kisvegabor, 2020-04-22:
@embeddedt I agree, it really looks like a good pattern.

Just to have a more complete example: 
```c
// I've changed the names because 
// it's related to animation in general not only the styles

// Forward declaration
struct _lv_anim_path_t;
typedef struct _lv_anim_path_t lv_anim_path_t;

// Callback prototype (lv_anim_t is added as parameter)
typedef void (*lv_anim_path_cb_t)(lv_anim_path_t*, lv_anim_t*);

// Struct definition with callbacks and user_data
typedef struct _lv_anim_path_t{
  lv_anim_path_cb_t *cb;
  int param;
  void * user_data;
}lv_anim_path_t;

...

//Create a new transition. (Can be resued)
static lv_anim_path_t path1;
lv_anim_path_init(&path1);
lv_anim_path_set_cb(&path1, lv_anim_path_linear)
lv_anim_path_set_param(&path1, 35);

//Use the transition in an animation
lv_anim_t a;
lv_anim_init(&a);
lv_anim_set_path(&a, &path1);

//Use the transition in a style
static lv_style_t style;
lv_style_init(&style);
lv_style_set_transition_path(&style, LV_STATE_DEFAULT, &path1);

//Use the transition in a local style
lv_obj_set_style_local_transition_path(obj, LV_OBJ_PART_MAIN, LV_STATE_DEFAULT,  &path1);
```

If you like this then it's also okay from my side.

embeddedt, 2020-04-22:
Looks good to me. @amirgon?

amirgon, 2020-04-22:
> Looks good to me. @amirgon?

Why do you have special functions to update `lv_anim_path_t` struct?
(`lv_anim_path_init`, `lv_anim_path_set_*`)
You can just let the user initialize it explicitly.  

Advantages:
- Can be done statically, instead of in code (`static lv_anim_path_t path1 = {...};`)
- A few less functions in the Micropython binding. Each function in the binding cost program ram (or flash), and there are already lots of functions. Why not eliminate the trivial ones?

Another question - 
What's the `param` field? Why do you need both `param` and `user_data`? Is `param` used by lvgl or only by the callback?

embeddedt, 2020-04-22:
I think `param` is for a required parameter that is used by the path function. `user_data` is for extra custom data you might want to pass to the path function.

amirgon, 2020-04-22:
> I think `param` is for a required parameter that is used by the path function. `user_data` is for extra custom data you might want to pass to the path function.

Is `param` updated/used by lvgl, or only by the callback?  
If it's just data that you want to pass to the callback, then it's very much like `user_data`, but for specific purpose. Why not hide it in `user_data` and save some ram?

kisvegabor, 2020-04-23:
> Why do you have special functions to update lv_anim_path_t struct?

In C I did the same with `lv_anim_...` using inline functions. So there is no overhead in C but it's more elegant and more future proof (e.g. if the assignment becomes more complex for any reason)

I really haven't considered the memory overhead in case of Micropyton. However, there will be only 2-3 functions here, so I think the overhead is negligible. Of course, if you think it's an important thing, we can discuss it in more detail. 
 
> What's the param field? Why do you need both param and user_data? Is param used by lvgl or only by the callback?

I just added `param` as an example. I still don't know how should it work. In C `user_data` really can be used, but in Micropython it is used by the binding. But you already have a solution for it because it'd cause a problem in other cases too.

amirgon, 2020-04-23:
> I really haven't considered the memory overhead in case of Micropyton. However, there will be only 2-3 functions here, so I think the overhead is negligible. Of course, if you think it's an important thing, we can discuss it in more detail.

A possible solution could be to prevent the creation of some functions, using some naming convention, or just by listing the excluded functions somewhere.

It's not limited to styles I guess, there are probably many other inline functions that are "redundant".
The cost is not RAM but ROM/Flash which I always considered a cheaper resource, but I think someone already mentioned that Micropython bindings is quite large already on Flash. I'm not sure how important this is vs other things, we should probably get some feedback on this from the users.

> In C `user_data` really can be used, but in Micropython it is used by the binding. But you already have a solution for it because it'd cause a problem in other cases too.

The difference is that in C the user explicitly uses `user_data`, while in Micropython it's "under the hood". But eventually it's the same thing.  
When the user creates a lambda function in Micropython and uses it to pass arguments to his callback, under the hoods the `user_data` is used to pass the callable Python object including the user's parameters.  
So eventually, no matter what the language is, `user_data` is the vehicle to pass arguments to a callback function on runtime.

kisvegabor, 2020-04-24:
> A possible solution could be to prevent the creation of some functions, using some naming convention, or just by listing the excluded functions somewhere.

If we choose the naming convention than it shouldn't make the usage in C less intuitive. So maybe it should be marked in the parameter names somehow instead of the function name.

> So eventually, no matter what the language is, user_data is the vehicle to pass arguments to a callback function on runtime.

Okay, let's use user_data in C as well. I'll implement it soon.

kisvegabor, 2020-04-27:
I've implemented the new "path" handling. See [here](https://github.com/littlevgl/lvgl/blob/dev-7.0/src/lv_misc/lv_anim.h#L47-L55).

And an example usage [here](https://docs.littlevgl.com/v7/en/html/get-started/quick-overview.html#styling-buttons).

amirgon, 2020-04-28:
@kisvegabor I tried to build dev-7.0.
There seem to be many functions that are **declared** but are not **defined**.
Since the binding wraps each of them, I get many unresolved symbols:
```
|| LINK micropython
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_theme_get_style':
|| lv_mpy.c:(.text.mp_lv_theme_get_style+0xc): undefined reference to `lv_theme_get_style'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_obj_get_opa_scale':
|| lv_mpy.c:(.text.mp_lv_obj_get_opa_scale+0xd): undefined reference to `lv_obj_get_opa_scale'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_obj_get_opa_scale_enable':
|| lv_mpy.c:(.text.mp_lv_obj_get_opa_scale_enable+0xd): undefined reference to `lv_obj_get_opa_scale_enable'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_spinner_get_style':
|| lv_mpy.c:(.text.mp_lv_spinner_get_style+0x21): undefined reference to `lv_spinner_get_style'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_switch_set_knob_on_img':
|| lv_mpy.c:(.text.mp_lv_switch_set_knob_on_img+0x21): undefined reference to `lv_switch_set_knob_on_img'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_switch_set_knob_img':
|| lv_mpy.c:(.text.mp_lv_switch_set_knob_img+0x21): undefined reference to `lv_switch_set_knob_img'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_win_get_btn_size':
|| lv_mpy.c:(.text.mp_lv_win_get_btn_size+0xd): undefined reference to `lv_win_get_btn_size'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_slider_get_knob_on_img':
|| lv_mpy.c:(.text.mp_lv_slider_get_knob_on_img+0x21): undefined reference to `lv_slider_get_knob_on_img'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_slider_get_knob_off_img':
|| lv_mpy.c:(.text.mp_lv_slider_get_knob_off_img+0x21): undefined reference to `lv_slider_get_knob_off_img'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_cpicker_get_preview':
|| lv_mpy.c:(.text.mp_lv_cpicker_get_preview+0xd): undefined reference to `lv_cpicker_get_preview'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_group_get_style_mod_edit_cb':
|| lv_mpy.c:(.text.mp_lv_group_get_style_mod_edit_cb+0xd): undefined reference to `lv_group_get_style_mod_edit_cb'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_group_get_style_mod_cb':
|| lv_mpy.c:(.text.mp_lv_group_get_style_mod_cb+0xd): undefined reference to `lv_group_get_style_mod_cb'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_group_set_style_mod_edit_cb':
|| lv_mpy.c:(.text.mp_lv_group_set_style_mod_edit_cb+0x2f): undefined reference to `lv_group_set_style_mod_edit_cb'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_group_set_style_mod_cb':
|| lv_mpy.c:(.text.mp_lv_group_set_style_mod_cb+0x2f): undefined reference to `lv_group_set_style_mod_cb'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_group_mod_style':
|| lv_mpy.c:(.text.mp_lv_group_mod_style+0x21): undefined reference to `lv_group_mod_style'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_spinner_set_style':
|| lv_mpy.c:(.text.mp_lv_spinner_set_style+0x31): undefined reference to `lv_spinner_set_style'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_msgbox_set_style':
|| lv_mpy.c:(.text.mp_lv_msgbox_set_style+0x31): undefined reference to `lv_msgbox_set_style'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_btnmatrix_set_style':
|| lv_mpy.c:(.text.mp_lv_btnmatrix_set_style+0x31): undefined reference to `lv_btnmatrix_set_style'
|| build/build/lvgl/lv_mpy.o: In function `mp_lv_checkbox_set_style':
|| lv_mpy.c:(.text.mp_lv_checkbox_set_style+0x31): undefined reference to `lv_checkbox_set_style'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_theme_get_style_obj+0x18): undefined reference to `lv_theme_get_style'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_group_get_style_mod_edit_cb_obj+0x18): undefined reference to `lv_group_get_style_mod_edit_cb'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_group_get_style_mod_cb_obj+0x18): undefined reference to `lv_group_get_style_mod_cb'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_group_mod_style_obj+0x18): undefined reference to `lv_group_mod_style'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_group_set_style_mod_edit_cb_obj+0x18): undefined reference to `lv_group_set_style_mod_edit_cb'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_group_set_style_mod_cb_obj+0x18): undefined reference to `lv_group_set_style_mod_cb'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_spinner_get_style_obj+0x18): undefined reference to `lv_spinner_get_style'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_spinner_set_style_obj+0x18): undefined reference to `lv_spinner_set_style'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_switch_set_knob_on_img_obj+0x18): undefined reference to `lv_switch_set_knob_on_img'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_switch_set_knob_img_obj+0x18): undefined reference to `lv_switch_set_knob_img'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_msgbox_set_style_obj+0x18): undefined reference to `lv_msgbox_set_style'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_win_get_btn_size_obj+0x18): undefined reference to `lv_win_get_btn_size'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_btnmatrix_set_style_obj+0x18): undefined reference to `lv_btnmatrix_set_style'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_slider_get_knob_on_img_obj+0x18): undefined reference to `lv_slider_get_knob_on_img'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_slider_get_knob_off_img_obj+0x18): undefined reference to `lv_slider_get_knob_off_img'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_cpicker_get_preview_obj+0x18): undefined reference to `lv_cpicker_get_preview'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_checkbox_set_style_obj+0x18): undefined reference to `lv_checkbox_set_style'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_obj_get_opa_scale_obj+0x18): undefined reference to `lv_obj_get_opa_scale'
|| build/build/lvgl/lv_mpy.o:(.rodata.mp_lv_obj_get_opa_scale_enable_obj+0x18): undefined reference to `lv_obj_get_opa_scale_enable'
|| collect2: error: ld returned 1 exit status
```

Please either define those functions, or remove their declarations.

kisvegabor, 2020-04-29:
Thank you, I've just removed them.

amirgon, 2020-04-29:
@kisvegabor , @embeddedt - When converting user code from v.6 to v.7, what do you suggest doing with `lv_style_plain`/ `lv_style_plain_color` / `lv_style_pretty` / `lv_style_pretty_color`?   
What are the equivalents on dev-7.0?

embeddedt, 2020-04-29:
I don't think there are really any equivalents, as the whole theming scheme has changed. I'd personally suggest simply using the default styles and tweaking individual properties as necessary.

amirgon, 2020-04-30:
> I don't think there are really any equivalents, as the whole theming scheme has changed. I'd personally suggest simply using the default styles and tweaking individual properties as necessary.

Do you have some guidelines for migrating user code from v.6 to v.7?
I think it would be a good idea to formulate something like this before releasing v.7.
Or do you expect v.7 to be used only for new code?

kisvegabor, 2020-04-30:
> I don't think there are really any equivalents, as the whole theming scheme has changed. I'd personally suggest simply using the default styles and tweaking individual properties as necessary.

I agree.

> Do you have some guidelines for migrating user code from v.6 to v.7?
I think it would be a good idea to formulate something like this before releasing v.7.
Or do you expect v.7 to be used only for new code?

I will write about it today, but there is no easy way to migrate the style part because it needs to be completely rewritten.

amirgon, 2020-05-02:
I'm a little worried about the size of the `lvgl` namespace.  

On dev-7.0 there are 654 child elements to the lvgl module, most of them are global functions.  
This means, for example, that when someone types in the REPL "`lv.` + <kbd>TAB</kbd>" he would get 654 completion options. That's a lot!  
(And that's not including the different methods of each widget, only "global" functions on the top of the hierarchy)

To mitigate this, I'm planning to move functions into their corresponding struct.  
For example, instead of writing `lv.style_get_radius(style, radius)` the user would write:
```python
style = lv.style_t()
style.set_radius(radius)
```
Given the correct naming conventions, the binding script can identify the relation between `lv_style_t` and `lv_style_get_radius(lv_style_t *, ...)`, and declare it as a member method of `lv_style_t`.
This would move many global functions into their related struct.

This is not backward compatible, but I think dev-7.0 is a good opportunity to make this change.  It provides a more hierarchical structure and the new syntax is clearer.  
But it does not actually reduce the total number of function.

@kisvegabor - 
- How strictly does lvgl follow the naming convention of struct methods?
 (`StructName_FunctionName(StructName_t *, ...)`). I'm relying on that naming convention for this change.
- The large number of functions, as I mentioned in previous post, costs ROM/Flash. We should think of some way to mitigate this for trivial functions that can be omitted (setters / getters).
- In case of styles, maybe something like a single `set/get_property(PROP_ID, prop_value)` instead of a function for each and every property? What do you think?
- Another idea to decrease program size is to identify functions with the exact same prototype and merge them together. This could give the same functionality with smaller code size. I still need to think about how to implement something like that.

embeddedt, 2020-05-02:
@amirgon I assume most of the new functions come from the style system. What if we added some special casing for the styling functions in order to make them more Pythonic? It might help reduce the code size since we won't need to generate so many functions.

For example, in JS, we can modify the style of an element using `element.style.backgroundColor = 'red'`. What do you think about adding something similar (i.e. make styles a Python object that appears as a property of a LittlevGL object)?

kisvegabor, 2020-05-02:
@amirgon 
I like the idea!

> How strictly does lvgl follow the naming convention of struct methods?
(StructName_FunctionName(StructName_t *, ...)). I'm relying on that naming convention for this change.

My intention was to fully follow this convention, but there might be exceptions. For example, the new `anim_path` functions look like this:
```c
static inline void lv_anim_path_set_cb(lv_anim_path_t * path, lv_anim_path_cb_t cb)
```
So if the "base name" is compound then it is used. 
Will it be possible to list the incompatible places with `gen.py`?

> The large number of functions, as I mentioned in previous post, costs ROM/Flash. We should think of some way to mitigate this for trivial functions that can be omitted (setters / getters).

I think the majority of the functions come from 2 sources:
- styles: both set and get are complex
- widgets: there might be a few trivial cases but the user still needed to set or get using the `lv_xxx_ext_t`. E.g. `layout = cont->ext.layout`. Besides, this way, some features would be available via API function and some via direct set which makes it not easy to use.

> In case of styles, maybe something like a single set/get_property(PROP_ID, prop_value) instead of a function for each and every property? What do you think?

Actually this is the underlying logic. However, there 4 functions for 4 types: int, opa, color, pointer.
There could be a common function because from `PROP_ID` the type can be calculated, but there is no common parameter type: 
```c
void lv_style_set(int ptop_id, lv_state_t state, ??? int/opa/color/ptr)
```

> Another idea to decrease program size is to identify functions with the exact same prototype and merge them together. This could give the same functionality with smaller code size. I still need to think about how to implement something like that.

Do the more functions really made the program so larger? Can you tell how large is it with v6 and v7?

@embeddedt 
> For example, in JS, we can modify the style of an element using element.style.backgroundColor = 'red'. What do you think about adding something similar (i.e. make styles a Python object that appears as a property of a LittlevGL object)?

Is it for local styles, right? How could it look like when you set/get `class`es?

embeddedt, 2020-05-02:
> Is it for local styles, right? How could it look like when you set/get classes?

On the web, classes are declared using CSS, and then added using `element.classList.add("classname")`.

For Python/LittlevGL, it would essentially be the same as the autogenerated script (I'm guessing a bit at the syntax here, so bear with me):

```py
# Create a style class
style = lv.style_t()

style.radius = 30
style.bg_color = lv_color_make(0xff, 0xff, 0xff)

obj.add_style(lv.obj.PART_MAIN, style)
```

There are basically two changes I'm proposing here:

*  Make `lv.style_t` appear like an object with value properties instead of requiring functions to be called for everything.
* Create a "virtual" `lv.style_t` as a property of LittlevGL objects, for local styles to be set/retrieved.

I'm not familiar at all with how MicroPython binds to C code, so it's quite possible my changes do nothing to reduce code size. Still, I just wanted to put the idea out there. :slightly_smiling_face:

amirgon, 2020-05-02:
>I assume most of the new functions come from the style system.

Many are, but there are a many others unrelated to the style system.

> ```c
> static inline void lv_anim_path_set_cb(lv_anim_path_t * path, lv_anim_path_cb_t cb)
> ```
> 
> 
> So if the "base name" is compound then it is used.
> Will it be possible to list the incompatible places with gen.py?

@kisvegabor I may be missing something here but why is `lv_anim_path_set_cb` incompatible?
- The struct is `lv_anim_path_t`
- The function name starts with `lv_anim_path`
- The first argument is `lv_anim_path_t *`

What is incompatible?

> Do the more functions really made the program so larger? Can you tell how large is it with v6 and v7?

Already in v.6 we got some feedback that the program size is very large (I can't remember if it was on some Github issue or some forum post). 

Here are the numbers when building for Linux (-Os):

dev-7.0:
```
||    text	   data	    bss	    dec	    hex	filename
|| 1553490	   7964	   7552	1569006	 17f0ee	micropython

2953760 May  3 00:09 lv_mpy.o
```

v.6 (main branch):
```
||    text	   data	    bss	    dec	    hex	filename
|| 1171961	   9300	  45664	1226925	 12b8ad	micropython

1559672 May  3 00:23 lv_mpy.o
```

That's around **30% increase** in total code size (lvgl+binding), and the binding object file itself `lv_mpy.o` grew by **90% of its original size**!

There can be two approaches to decrease the binding size:
- Bind less functions.  Currently all functions are bound. 
  - We would need some naming convention to let the script know what not to bind. 
  - We would need to decide on some coherent set of guidelines that say which functions should/should not be bound, and what should the user do instead, if a function in missing.
- Optimize the binding itself. It can be done at different levels of complexity. Merging bound functions with the same prototype is one idea. There may be other more radical ideas like fully dynamic FFI or some way to use Micropython's dynamic C module loading for this purpose.

amirgon, 2020-05-02:
> There are basically two changes I'm proposing here:
> 
> * Make `lv.style_t` appear like an object with value properties instead of requiring functions to be called for everything.
> * Create a "virtual" `lv.style_t` as a property of LittlevGL objects, for local styles to be set/retrieved.

@embeddedt -

"`style_t` as an object" is my suggestion above.  
It would make things prettied and more "Pythonic", but it would not save any code space.

A "virtual" `lv.style_t` as a property of LittlevGL objects would need to be created on the C side.
Instead of all these `get/set_style_local` functions you could instead provide a function to get the widget's style, and use the standard style getter/setters on it.  
In my opinion that would be cleaner also on the C side.

kisvegabor, 2020-05-04:
> Instead of all these get/set_style_local functions you could instead provide a function to get the widget's style, and use the standard style getter/setters on it.

It works like this [under the hood](https://github.com/littlevgl/lvgl/blob/1633b2e6dbb9b81050ea73c9541bc19bc4b6f404/src/lv_core/lv_obj.c#L1200-L1208) but as you can see it's a little bit more than simply setting a style's properties. Besides, I found it very long and error prone to write something like:
```c
lv_style_t * style = lv_obj_get_style_list(bar, LV_BAR_PART_INDIC);
lv_style_set_bg_color(style, LV_STATE_DEFAULT, LV_COLOR_RED);
lv_obj_refresh_style(obj, LV_STYLE_BG_COLOR);

/*Compared to*/
lv_obj_set_style_local_bg_color(bar, LV_BAR_PART_INDIC, LV_STATE_DEFAULT, LV_COLOR_RED)
```

amirgon, 2020-05-04:
> it's a little bit more than simply setting a style's properties.

Aren't you worried about the fast growth of the API?
A C user using his IDE for code completion and browsing, would see a flat list of hundreds of functions, many of them with similar and repeating names, and needs to browse them to find the specific correct one...

> Besides, I found it very long and error prone to write something like:
>
> ```c
> lv_style_t * style = lv_obj_get_style_list(bar, LV_BAR_PART_INDIC);
> lv_style_set_bg_color(style, LV_STATE_DEFAULT, [...]

How about something like:

```c
lv_style_set_bg_color(
    lv_obj_get_style_list(bar, LV_BAR_PART_INDIC), LV_STATE_DEFAULT, LV_COLOR_RED));
```

I'm assuming refresh can be done automatically in this case (for example by `lv_style_set_bg_color`)

kisvegabor, 2020-05-05:
> Aren't you worried about the fast growth of the API?

I see your point but in my opinion, with proper naming conventions auto-completion is not a problem as you can quickly eliminate most of the options. 

However, I also see that flash usage is an issue. As you suggested earlier we can reduce the number of functions to bind. I have 2 ideas:
1. Add an option to `lv_conf.h` to eliminate functions that are very unlikely to be used. These are typically `get` function. For example, it's not probably that anybody needs to get the number of horizontal division lines of a chart, or the animation time of the drop-down list. IMO `style_get` and `style_local_get` are also unlikely. I've tested how much could it help in [feat/minimal_api](https://github.com/littlevgl/lvgl/commit/97f048394447358e18ec408e1bf2883c636480a6). I've eliminated the `style_get`s and functions from the widget and it reduced the number of lines in `gen.c` from 56k to 49k (~12%). Not that much. I think we could reach 15% in the end. It will also slightly reduce the memory usage in C. (many of these functions are inline)
2. As I see `gen.c` there are a lot of types and functions that are required for LVGL but probably not required for someone who uses the binding.  For example `lv_anim_init_core()` is only called in `lv_init()` or `lv_area_align()` is used internally in `lv_obj_align()`. Regarding the types, I'm not sure if a binding to `lv_anim_t` (`mp_lv_anim_t_attr`?) is required or not as it is not intended to set/get directly. I can imagine 2 ways here: use and `#if BUILD_WRAPPER` guard around these functions and types (would be easy to use in other bindings too) and/or add an `_` prefix and skip them during generation (will also help with auto competition). 

In `gen.c` I saw a lot of `.print`. I'm not sure how many ROM do they use but if they are only for `help`, maybe a config option to disable them could help.

What do you think of these?

I made some experiments with [SWIG](http://www.swig.org/) to generate Lua binding and it created 150k lines which were built to ~900 kB. LVGL+Lua+Binding left only 10% space in a 2MB flash MCU.

> How about something like:
>lv_style_set_bg_color(
    lv_obj_get_style_list(bar, LV_BAR_PART_INDIC), LV_STATE_DEFAULT, LV_COLOR_RED));
> I'm assuming refresh can be done automatically in this case (for example by lv_style_set_bg_color) 

The refresh can happen in `lv_style_set_bg_color` in a very ineffective way because it needs to search for every object who possibly uses that style and invalidate it. `lv_obj_set_style_local_...` invalidates only the affected object. 

In `feat/minimal_api` I disabled two types of `style_get` functions + some widget functions and gained 12%. Skipping and another bunch of style functions would result in ~5% gain I guess.

embeddedt, 2020-05-05:
I think the API size isn't a problem for C programs.

```
  addr     size  type   name
-----------------------------------------
080315dc 0000000c T lv_style_init
080315e8 0000000c T lv_style_list_init
08039d3a 00000012 t lv_cont_get_style
0803ac8c 00000012 t lv_img_get_style
08046f18 00000012 T lv_label_get_style
0803181c 00000012 T lv_style_reset
080355ac 00000014 T lv_obj_reset_style_list
0803241c 00000016 T lv_style_list_set_local_int
08032432 00000016 T lv_style_list_set_local_opa
08032448 00000016 T lv_style_list_set_local_ptr
08032252 0000001a T lv_style_list_get_transition_style
0802e298 0000001c t style_init_reset
0804a96c 00000026 t lv_checkbox_get_style
0803222c 00000026 T lv_style_list_get_local_style
0803186e 00000028 T lv_style_copy
08034a00 0000002a T lv_obj_get_style_list
080355c0 00000034 T _lv_obj_set_style_local_int
080355f4 00000034 T _lv_obj_set_style_local_opa
08035628 00000034 T _lv_obj_set_style_local_ptr
080353cc 0000003c t refresh_children_style
0803182e 00000040 T lv_style_get_mem_size
08034aa4 00000044 T lv_obj_clean_style_list
080356ea 00000048 T lv_obj_report_style_mod
08049dec 00000054 t lv_arc_get_style
08041cbc 00000054 t lv_bar_get_style
0804ac50 00000054 t lv_btnmatrix_get_style
0803b650 00000054 t lv_roller_get_style
0804c690 00000054 t lv_spinbox_get_style
08035554 00000058 T lv_obj_add_style
080485f4 0000005c t lv_chart_get_style
0803d994 0000005c t lv_gauge_get_style
0804714c 0000005c t lv_slider_get_style
08038718 0000005c t lv_switch_get_style
0803226c 00000060 T lv_style_list_reset
0803cb94 00000064 t lv_calendar_get_style
08038c1c 00000068 t lv_table_get_style
08043ab4 00000068 t lv_textarea_get_style
0803ff74 0000006c t lv_dropdown_get_style
08041180 0000006c t lv_list_get_style
0804cfa8 00000070 t lv_msgbox_get_style
0803e728 00000070 t lv_page_get_style
08036370 00000080 T _lv_obj_get_style_color
080363f0 0000008a T _lv_obj_get_style_opa
0803565c 0000008e t report_style_mod_core
0803647c 00000094 T _lv_obj_get_style_ptr
08032388 00000094 T lv_style_list_add_trans_style
08042908 00000094 t lv_tabview_get_style
080360f0 000000a0 T _lv_obj_get_style_int
08031778 000000a4 t get_alloc_local_style
080316c4 000000b4 T lv_style_list_add_style
08032698 000000ba T lv_style_list_get_ptr
080322cc 000000bc T lv_style_list_copy
08032520 000000bc T lv_style_list_get_color
080325dc 000000bc T lv_style_list_get_opa
0803245e 000000c2 T lv_style_list_get_int
08032090 000000ce T _lv_style_get_color
08031ef4 000000ce T _lv_style_get_int
08031fc2 000000ce T _lv_style_get_opa
0803215e 000000ce T _lv_style_get_ptr
080315f4 000000d0 T lv_style_list_remove_style
08031896 000000e8 T lv_style_remove_prop
08035408 0000014c T lv_obj_refresh_style
08031adc 0000015c T _lv_style_set_color
08031980 0000015c T _lv_style_set_int
08031c38 0000015c T _lv_style_set_opa
08031d94 00000160 T _lv_style_set_ptr
```

This is the output of `nm` on a LittlevGL application filtered to only show the APIs dealing with styles. The specific APIs like `set_bg_color`, etc. are replaced in the final binary with direct calls to `_lv_style_set_int`, etc. That means we pay a one-time cost for each *type* of style property, but we can add as many properties as we want with little to no Flash overhead (but there will be RAM overhead for storage, obviously).

It sounds like MicroPython can't take advantage of this optimization, and has to generate a new function for every style property. This plus the extra internal types and functions that @kisvegabor mentioned probably makes the binding way larger than it needs to be.

amirgon, 2020-05-07:
@kisvegabor I'm trying to implement struct functions, and I've seen some cases where [the naming convention](https://github.com/littlevgl/lv_binding_micropython/issues/75#issuecomment-622629812) does not apply.    

For example:
```c
void lv_disp_flush_ready(lv_disp_drv_t * disp_drv);
```

If the function was called `lv_disp_drv_flush_ready`, or the struct was called `lv_disp_t`, that would have worked. But the script expects that the name of the struct would really match the beginning of the function name.

Here are some ideas how to handle this situation:
- Fix the function name or the struct name to match, such that the naming convention would apply.
- Change the convention such that the function name is completely ignored, and any function that receives struct `S` as its first argument would belong to struct `S`.  
This might be too broad and include in a struct unrelated functions, I'm not sure.

Which do you think is better?

embeddedt, 2020-05-07:
> Why do some of the functions start with _?

They're private functions which shouldn't be used externally. If they can be hidden that's fine.

amirgon, 2020-05-07:
> They're private functions which shouldn't be used externally. If they can be hidden that's fine.

@embeddedt If so why are they reachable by code that includes `lvgl.h`?
Is it possible to move them such that user code would not see them at all?

embeddedt, 2020-05-07:
They are wrapped by inline functions in public headers. The inline functions need to be able to reference them.

amirgon, 2020-05-07:
> They are wrapped by inline functions in public headers. The inline functions need to be able to reference them.

I see.  
I'll add a rule in the script to ignore (hide) any function that starts with an underscore.

kisvegabor, 2020-05-08:
> If the function was called lv_disp_drv_flush_ready, or the struct was called lv_disp_t, that would have worked. But the script expects that the name of the struct would really match the beginning of the function name.

It needs to use `drv` because `lv_disp_t` has no `user_data`. We can change the name to `lv_disp_drv_flush_ready` as you suggested but needs to keep the current version too for a while. 

FYI, in v7 we have created [lv_api_map.h](https://github.com/littlevgl/lvgl/blob/dev-7.0/src/lv_api_map.h) which contains wrappers for minor name changes. 

> Change the convention such that the function name is completely ignored, and any function that receives struct S as its first argument would belong to struct S.

What do you think about removing the common parts of the struct and function name from the function name in the script? E.g.
```
lv_draw_mask_line_points_init(lv_draw_mask_line_param_t * param, ...)

lv_draw_mask_line_points_init
lv_draw_mask_line_param_t
---------------------------------------
lv_draw_mask_line  #The common part
```
So in Python:
```python
param.points_init(...)
file.read(...)
```

> Why do some of the functions start with _?

I agree with gnoring the functions starting with an underscore. The purpose of using the underscore was to hide these names from autocomplete. 

> I suspect there are some functions that should not be exposed to the user, only used internally.

I agree and removed them here: 9dd387b2

>> Add an option to lv_conf.h to eliminate functions that are very unlikely to be used. These are typically get function. ... . I've eliminated the style_gets and functions from the widget and it reduced the number of lines in gen.c from 56k to 49k (~12%). Not that much. I think we could reach 15% in the end.

> Relatively small gain.
Also it may be hard to maintain this, eventually we might eliminate some functions which are actually needed.
I'm not sure it's worth it for 15%, I'm looking for some way to reach around 50% at least.

I agree, let's drop this direction.

>> As I see gen.c there are a lot of types and functions that are required for LVGL but probably not required for someone who uses the binding. ... I can imagine 2 ways here: use and #if BUILD_WRAPPER guard around these functions and types (would be easy to use in other bindings too) and/or add an _ prefix and skip them during generation ...

> The Micropython binding processes all functions/structs that are reachable from lvgl.h.
If you have some internal functions/struct, just make sure they are not exposed to the user through lvgl.h either directly or indirectly.

I meant for example `lv_obj_t` is obviously required, but I'm not sure the user needs to be able to see and access the members directly but via only API functions. So I don't know if `mp_lv_obj_t_attr` is required.

amirgon, 2020-05-08:
> FYI, in v7 we have created [lv_api_map.h](https://github.com/littlevgl/lvgl/blob/dev-7.0/src/lv_api_map.h) which contains wrappers for minor name changes.

Adding more functions ("C wrappers") is cheap on C but costs more on Micropython.  

How about not including `lv_api_map.h` from `lvgl.h`?
If a C user wants the old names via wrappers, he could include `lv_api_map.h` explicitly. But the Micropython binding will not process these wrappers as long as they are inaccessible from `lvgl.h`.  

That would also give the C users a simple way to git rid of the old names in their code - just don't include `lv_api_map.h` and fix the compilation errors.

> What do you think about removing the common parts of the struct and function name from the function name in the script?

We can do that, but what should be the shortest length of the common part?
Is a zero-length common part valid? If it is, any function that receives struct `S` as the first argument would become a member function of struct `S`.  

I'm not sure if all this makes sense or not.

> As I see gen.c there are a lot of types and functions that are required for LVGL but probably not required for someone who uses the binding. ... I can imagine 2 ways here: use and #if BUILD_WRAPPER [...]

> I meant for example `lv_obj_t` is obviously required, but I'm not sure the user needs to be able to see and access the members directly but via only API functions. So I don't know if `mp_lv_obj_t_attr` is required.

When deciding whether to provide or hide a function from the user, I think we should choose the simplest and most straight forward approach.
I also think this should be a general approach that applies to both C and Micropython (or any other binding).  

Using `#if BUILD_WRAPPER` is not so straight forward  in my opinion. It doesn't hint anything to the C user and it's easy to define it the wrong way - either include unwanted functions or exclude required functions, especially when the code changes over time.

Currently, a function is available in Micropython **only** if the following conditions are all true:
- The function is accessible when including `lvgl.h` in a C file.
- The function does not start with an underscore.
- The script can figure out how to create the Micropython wrapper.

`lv_obj_t` is a special case, being the "base class" of all objects.  
The script does not create a Micropython struct object to wrap `lv_obj_t` like it does with all other structs. `lv_obj_t` internal state is not exposed to the user.
- Functions that receive `lv_obj_t` as its first argument are considered Object Member Functions.
- Function that don't are either global functions or Struct Member Functions.

**So, my suggestion is -** if there are any functions that are included in the binding but should not be there, let's either making them inaccessible from `lvgl.h`, or rename them with underscore prefix.

What do you think?

kisvegabor, 2020-05-11:
> How about not including lv_api_map.h from lvgl.h?

IMO, an option in `lv_conf.h` would be better to enable backward compatibility or not.

> But does it really make sense to include all the functions above as point_t members?

Ah, that's true. There are cases that are clearly not intended to happen.

> Using #if BUILD_WRAPPER is not so straight forward in my opinion. It doesn't hint anything to the C user and it's easy to define it the wrong way - either include unwanted functions or exclude required functions, especially when the code changes over time.

The `_` would be really cleaner.

I counted the number of API functions with `cat gen.c | grep MP_DEFINE_CONST_LV_FUN_OBJ_VAR | wc` (is it correct?) and it says there are 1311 functions. To have a significant gain ~500 functions should be removed either by no including them or by marking them with an underscore. 

To no include them most of the header files should be duplicated to separate e.g. `lv_area_t` (which is required in some API functions) and `lv_area_set_pos()` (which should be used only internally). 

Underscores can be added "freely". 

> lv_obj_t is a special case, being the "base class" of all objects.
The script does not create a Micropython struct object to wrap lv_obj_t like it does with all other structs. lv_obj_t internal state is not exposed to the user.

It was a bad example then. What's the case with `mp_lv_anim_t_attr`?

kisvegabor, 2020-05-11:
I got an idea! I've never seen in other projects but I think it could work. 
We can define `LV_API_RESTRICTED` in `lv_conf.h` and wrap some functions in the beader files in `#ifndef LV_API_RESTRICTED ... #endif`. This way if the header file is included from `lvgl.h` the function won't be visible but they will be visible when used internally via "normal include". What do you think?

embeddedt, 2020-05-11:
@kisvegabor Don't we (possibly indirectly) include `lv_conf.h` inside `lvgl.h`?

kisvegabor, 2020-05-11:
Ah sorry, I meant to write `lvgl.h`. (Updated)

amirgon, 2020-05-11:
> IMO, an option in `lv_conf.h` would be better to enable backward compatibility or not.

That's fine too. `lv_conf.h` of lv_micropython is already different, with all the Micropython-gc definitions.  
So backward compatibility should be disabled by default in lvgl under lv_micropython.

> > But does it really make sense to include all the functions above as point_t members?
> 
> Ah, that's true. There are cases that are clearly not intended to happen.

I'm trying to implement your idea of common part, but making sure there is at least one "word" (delimited by underscore) at the beginning, common to both the struct name and function name.

> I counted the number of API functions with `cat gen.c | grep MP_DEFINE_CONST_LV_FUN_OBJ_VAR | wc` (is it correct?) and it says there are 1311 functions.

That's `wc -l` to count lines not words, but I got even higher number: 1506.

> To have a significant gain ~500 functions should be removed either by no including them or by marking them with an underscore.

584 functions are specific to the new style system.
It could be nice to find a way to reduce those.

(`grep "MP_DEFINE_CONST_LV_FUN_OBJ_VAR.*_style_" lv_mpy_example.c | wc -l`)

> Underscores can be added "freely".

I think underscore are better than ifdefs in all cases of internal functions.

> It was a bad example then. What's the case with `mp_lv_anim_t_attr`?

I'm not sure what's the "attr" in `mp_lv_anim_t_attr`, I don't have such struct.
In case of `anim_t` struct, Micropython has access to everything that the C user can access, both functions and data members (btw data members cost less than functions).
Here are `anim_t` members:

```
>>> a = lv.anim_t()
>>> print("\n".join(dir(a)))
__class__
end
start
__dereference__
SIZE
act_time
cast
cast_instance
custom_del
early_apply
exec_cb
get_delay
has_run
init
path
playback_delay
playback_now
playback_time
ready_cb
repeat_cnt
repeat_delay
set_custom_exec_cb
set_delay
set_exec_cb
set_path
set_playback_delay
set_playback_time
set_ready_cb
set_repeat_count
set_repeat_delay
set_start_cb
set_time
set_values
set_var
start_cb
time
time_orig
user_data
var
```

If you think any of them should be "private" and not exposed to the user, let's just prefix the private ones with underscore.

> We can define `LV_API_RESTRICTED` in the top of `lvgl.h` and wrap some functions in the beader files in `#ifndef LV_API_RESTRICTED ... #endif`. This way if the header file is included from `lvgl.h` the function won't be visible but they will be visible when used internally via "normal include". What do you think?

In general, I prefer underscoring some function instead of ifdefing entire ranges.
When you ifdef entire range it becomes much easier to miss, and harder to maintain.

Also - I think private functions should be marked as such not only for Micropython but also for the C user. It is much more obvious for a C user to identify a function as private when it's prefixed by underscore, vs. some obscure ifdefs.

kisvegabor, 2020-05-11:
I noticed that when counted the functions above lvgl was on a test branch, In the current `dev-7.0` I counted 1521 API functions that fall into these categories:
- style related: ~600. I have an idea about cutting it into half
- lv_obj related: ~100. Not so much to do here I think
- widgets: ~500. Disabling widgets would help in C but not in Micropython
- Others: ~400. 100-150 might be eliminated.

So we can save ~500 functions.

amirgon, 2020-05-11:
> * widgets: ~500. Disabling widgets would help in C but not in Micropython

Actually when you are disabling a widget in `lv_conf.h` and building Micropython again, that would also reduce the binding size. The script parses lvgl.h after preprocessing, taking lv_conf.h into account.

> * style related: ~600. I have an idea about cutting it into half

How about providing a more generic mechanism for applying styles, instead of a different function for every style option? Something more like the underlying mechanism that these function call.
Is it really that inconvenient for a user to call a generic function and provide an enum as an argument?

embeddedt, 2020-05-11:
> Is it really that inconvenient for a user to call a generic function and provide an enum as an argument?

I think that type information is the issue, not the fact that you have to provide an extra argument. As far as I know, the generic functions use non-specific types like `uint32_t` instead of types like `lv_coord_t` or `lv_border_side_t` that are more intuitive.

amirgon, 2020-05-11:
> As far as I know, the generic functions use non-specific types like `uint32_t` instead of types like `lv_coord_t` or `lv_border_side_t` that are more intuitive.

What is the added value of `lv_border_side_t` vs `uint8_t`?
It's really not clear to me.

Why is :
```c
lv_border_side_t my_border_size = lv_obj_get_style_border_size(o);
```

more intuitive than this?:
```c
uint8_t my_border_size = lv_obj_get_style_attr(o, BORDER_SIZE);
```

kisvegabor, 2020-05-11:
> I'm trying to implement your idea of common part, but making sure there is at least one "word" (delimited by underscore) at the beginning, common to both the struct name and function name.

Great!

> I think underscore are better than ifdefs in all cases of internal functions.

As I imagine it can make some difficulties if functions which are frequently used internally begin with an underscore and others are not. We will ask: was it `lv_area_get_size()` or `_lv_area_get_size`? 

Now underscore is used for functions which are really rarely used during development too.

> In general, I prefer underscoring some function instead of ifdefing entire ranges.
When you ifdef entire range it becomes much easier to miss, and harder to maintain.

We can add the guard around each function instead of a range too. What do you think?

> Also - I think private functions should be marked as such not only for Micropython but also for the C user. It is much more obvious for a C user to identify a function as private when it's prefixed by underscore, vs. some obscure ifdefs.

I tested it Eclipse and its autocomplete still sees the disabled function. So `_` really better in this regard.

@embeddedt what do you think?

> Is it really that inconvenient for a user to call a generic function and provide an enum as an argument?

It's mainly what @embeddedt described about the types. The user needs to know that `lv_style_set_int` is required to set border width (`int`), border side (`lv_bordr_side_t`), border post (draw border after children) (`bool`), but for border opa `lv_style_set_opa` should be used. 
 
Now we have this kind of functions:
- `lv_obj_get_style_<property>`: Used internally during drawing. It can be marked with `_`.
- `lv_obj_set_style_local_<property>` Required for the user
- `lv_obj_get_style_local_<property>` Can replaced by getting the local style and getting its property: `lv_style_get_<property>(lv_obj_get_local_style(obj), ...)` 
- `lv_obj_remove_style_local_<property>` No confusing type here so the internal `_lv_obj_remove_style_local_prop` can be made public.
- `lv_style_get_<property>` Difficult to use and not really required on the user level. Can be marked with `_`
- `lv_style_set_<property>` Required for the user

So 4 out of the 6 can be eliminated from the binding and 3 from C.

kisvegabor, 2020-05-11:
> What is the added value of lv_border_side_t vs uint8_t?

It would look like this:
```c
uint8_t my_border_width      = lv_obj_get_style_int(o, BORDER_WIDTH);
uint8_t my_border_side       = lv_obj_get_style_int(o, BORDER_SIDE);
uint8_t my_border_blend_mode = lv_obj_get_style_int(o, BORDER_BLEND_MODE);
uint8_t my_border_post       = lv_obj_get_style_int(o, BORDER_POST);
lv_opa_t my_border_opa       = lv_obj_get_style_opa(o, BORDER_OPA);
lv_color_t my_border_color   = lv_obj_get_style_color(o, BORDER_COLOR);
```

There are only 4 types in styles: opa, color, int pointer. And it's not intuitive that e.g. bool, all enums, signed and unsigned integers are `_int`.

[108 more replies]

closed 2020-07-06 by amirgon