lvgl/lv_binding_micropython #77: ST7789

jdtsmith, 2020-05-30:
[This project ](https://github.com/lbthomsen/esp-idf-littlevgl) implements a LittlevGL compatible display driver for the popular ST7789 display.  Haven't tested it in C, but it would be wonderful to have that ported for use with the lvgl Micropython bindings.

amirgon, 2020-05-30:
@jdtsmith would you like to port it and contribute it to lvgl?  
That would be greatly appreciated!
Feel free to open a PR and let me know if you need any help.

btw, the difference between st7789 and ili9341 might be small enough such that it would make sense to have the same driver for both.
Currently supported driver for ili9341 is the hybrid python driver [ili9341.py](https://github.com/lvgl/lv_binding_micropython/blob/master/driver/esp32/ili9341.py).

jdtsmith, 2020-05-31:
Not in my ready skill set; is there a guide anywhere?  BTW, now that MP on ESP32 [supports viper](https://github.com/micropython/micropython/pull/5082), are high speed pure python drivers possible?

embeddedt, 2020-05-31:
From my limited understanding it should be possible, it's just not implemented yet in the existing drivers. I believe @amirgon has a note for it in #56.

amirgon, 2020-05-31:
> now that MP on ESP32 [supports viper](https://github.com/micropython/micropython/pull/5082), are high speed pure python drivers possible?

Viper is most useful when making some transformation on RAM by looping over many bytes.  
[A good example is imagetools.py](https://github.com/lvgl/lv_binding_micropython/blob/34719041093486ccb2d5c6b6c8619ecea77d6e2a/lib/imagetools.py#L64) which helps decoding PNG files. When decoding PNG file, we usually need to convert all pixels to another RGB format by looping over them and doing some arithmetic. This is where Viper really shines.
But in other cases, such as in the case of a display driver, we want to do something a bit different. We want to prepare DMA message and call architecture specific functions to drive DMA. Viper can help to some extent, but best results would be [to do that in C](https://github.com/lvgl/lv_binding_micropython/blob/34719041093486ccb2d5c6b6c8619ecea77d6e2a/driver/esp32/espidf.c#L164).

[ili9341.py](https://github.com/lvgl/lv_binding_micropython/blob/master/driver/esp32/ili9341.py) works by default in "hybrid" mode, where the "slowpath" flows are implemented in Python (initializing the display controller, configuring and initializating SPI, DMA etc.), and the "fastpath" flows are in C (the "flush" function implementation).  

My guess is that if anyone ever wants to use ili9341.py for other display controllers, in many cases this can be achieved without touching the C side, only by changing the Python side.

> Not in my ready skill set; is there a guide anywhere?

A guide for porting the display driver? Not really.
Basically, you need to read the datasheet of your display driver and modify ili9341.py to support it. The skills you would need are: reading datasheets + writing python.

closed 2020-06-28 by stale[bot]