lvgl/lv_binding_micropython #78: Complete documentation

TimonPeng, 2020-06-16:
I did the following work, but it was not included:

```
WORKDIR /root
RUN git clone --recurse-submodules https://github.com/micropython/micropython.git

WORKDIR /root/micropython
RUN git submodule update --init

WORKDIR /root/micropython/lib
RUN git clone --recurse-submodules https://github.com/lvgl/lv_binding_micropython.git

WORKDIR /root/micropython/lib/lv_binding_micropython/gen
RUN python gen_mpy.py -MD lv_mpy_example.json -M lvgl -MP lv -I../../berkeley-db-1.xx/PORT/include -I../../lv_binding_micropython -I. -I../.. -Ibuild -I../../mp-readline -I ../../lv_binding_micropython/pycparser/utils/fake_libc_include ../../lv_binding_micropython/lvgl/lvgl.h

WORKDIR /root/micropython
RUN make -j4 -C mpy-cross

WORKDIR /root/micropython/ports/esp32
CMD make -j4 submodules && make -j4 &
```

```
>>> help('modules')
__main__          framebuf          uasyncio/funcs    upip
_boot             gc                uasyncio/lock     upip_utarfile
_onewire          inisetup          uasyncio/stream   urandom
_thread           machine           ubinascii         ure
_uasyncio         math              ubluetooth        uselect
_webrepl          micropython       ucollections      usocket
apa106            neopixel          ucryptolib        ussl
btree             network           uctypes           ustruct
builtins          ntptime           uerrno            utime
cmath             onewire           uhashlib          utimeq
dht               sys               uhashlib          uwebsocket
ds18x20           uarray            uheapq            uzlib
esp               uasyncio/__init__ uio               webrepl
esp32             uasyncio/core     ujson             webrepl_setup
flashbdev         uasyncio/event    uos               websocket_helper
Plus any modules on the filesystem
>>> import lvgl
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: no module named 'lvgl'
```

What's the problem? please

embeddedt, 2020-06-16:
@TimonPeng You also have to [patch several upstream files](https://github.com/lvgl/lv_binding_micropython/blob/master/README.md#adding-micropython-bindings-to-a-project) in order for the lvgl module to be built and linked. Take a look at the [difference](https://github.com/micropython/micropython/compare/master...lvgl:master#diff-8cb873f8d91be202cc15633f10a62f49) between `lv_micropython` and upstream MicroPython.

amirgon, 2020-06-16:
To add to @embeddedt answer, the simplest way to use lvgl+micropython is to clone and build [lv_micropython](https://github.com/lvgl/lv_micropython), which is a fork of Micropython v1.12 that incorporates lvgl.

TimonPeng, 2020-06-17:
@embeddedt thanks for your reply, but i can't found `lv_conf.h` in https://github.com/lvgl/lv_micropython/tree/master/lib

TimonPeng, 2020-06-17:
@amirgon thanks for your reply, but i used the same process to compile lv_micropython, but something wrong

```
Building with ESP IDF v3
find: ‘/root/esp-idf/components/json/include’: No such file or directory
find: ‘/root/esp-idf/components/json/port/include’: No such file or directory
```

TimonPeng, 2020-06-17:
> @amirgon thanks for your reply, but i used the same process to compile lv_micropython, but something wrong
> 
> ```
> Building with ESP IDF v3
> find: ‘/root/esp-idf/components/json/include’: No such file or directory
> find: ‘/root/esp-idf/components/json/port/include’: No such file or directory
> ```

Use the same process as micropython

amirgon, 2020-06-17:
> but i can't found `lv_conf.h` in https://github.com/lvgl/lv_micropython/tree/master/lib

`lv_conf.h` [is part of lv_binding_micropython](https://github.com/lvgl/lv_binding_micropython/blob/master/lv_conf.h)

> i used the same process to compile lv_micropython, but something wrong
> 
> ```
> Building with ESP IDF v3
> find: ‘/root/esp-idf/components/json/include’: No such file or directory
> find: ‘/root/esp-idf/components/json/port/include’: No such file or directory
> ```

Did you checkout the correct esp-idf version as described in the [README](https://github.com/lvgl/lv_micropython/tree/master/ports/esp32#setting-up-the-toolchain-and-esp-idf)?

Did you get any other errors?  
I believe that the two warnings you showed above can be ignored.

In general, to build lv_micropython you need to [follow the instructions here](https://github.com/lvgl/lv_micropython#build-instructions).

TimonPeng, 2020-06-17:
@amirgon  yes, i used correct esp-idf version, my dockerfile here:

```
FROM python:3

RUN apt-get update && apt-get -y install \
    git wget curl \
    build-essential libreadline-dev libffi-dev pkg-config gcc-arm-none-eabi libnewlib-arm-none-eabi \
    libncurses-dev flex bison gperf cmake ninja-build ccache libffi-dev libssl-dev \
    gawk gperf grep gettext automake texinfo help2man libtool libtool-bin make

ENV PATH="/root/esp/xtensa-esp32-elf/bin:$PATH"
ENV IDF_PATH="/root/esp-idf"

USER root

WORKDIR /root
RUN mkdir esp

WORKDIR /root/esp
RUN wget https://dl.espressif.com/dl/xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz
RUN tar -xzf xtensa-esp32-elf-gcc8_2_0-esp-2020r1-linux-amd64.tar.gz

WORKDIR /root
RUN git clone https://github.com/espressif/esp-idf.git

WORKDIR /root/esp-idf
# V3
RUN git checkout 6ccb4cf5b7d1fdddb8c2492f9cbc926abaf230df
# V4
# RUN git checkout 310beae373446ceb9a4ad9b36b5428d7fdf2705f
RUN git submodule update --init --recursive
RUN pip install -r requirements.txt

WORKDIR /root
RUN git clone --recurse-submodules https://github.com/lvgl/lv_micropython.git

WORKDIR /root/lv_micropython
RUN git submodule update --init
RUN make -j4 -C mpy-cross

WORKDIR /root/lv_micropython/ports/esp32

CMD make -j4 LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1" submodules \
    && make -j4 LV_CFLAGS="-DLV_COLOR_DEPTH=16 -DLV_COLOR_16_SWAP=1"
```

but many warning and error logs here (5000+ lines):

```
In file included from /root/esp-idf/components/cxx/cxx_exception_stubs.cpp:2:
/root/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdio:101:11: error: '::clearerr' has not been declared
   using ::clearerr;
           ^~~~~~~~
/root/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdio:102:11: error: '::fclose' has not been declared
   using ::fclose;
           ^~~~~~
/root/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdio:103:11: error: '::feof' has not been declared
   using ::feof;
           ^~~~
/root/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdio:104:11: error: '::ferror' has not been declared
   using ::ferror;
           ^~~~~~
/root/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdio:105:11: error: '::fflush' has not been declared
   using ::fflush;
           ^~~~~~
/root/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdio:106:11: error: '::fgetc' has not been declared
   using ::fgetc;
           ^~~~~
/root/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdio:107:11: error: '::fgetpos' has not been declared
[... 125 lines cut ...]
/root/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdio:188:22: error: '__gnu_cxx::vsnprintf' has not been declared
   using ::__gnu_cxx::vsnprintf;
                      ^~~~~~~~~
/root/esp/xtensa-esp32-elf/xtensa-esp32-elf/include/c++/8.2.0/cstdio:189:22: error: '__gnu_cxx::vsscanf' has not been declared
   using ::__gnu_cxx::vsscanf;
                      ^~~~~~~
/root/esp-idf/components/cxx/cxx_exception_stubs.cpp: In function 'void __cxx_fatal_exception_message(const char*)':
/root/esp-idf/components/cxx/cxx_exception_stubs.cpp:24:5: error: 'printf' was not declared in this scope
     printf("%s%s\n", FATAL_EXCEPTION, msg);
     ^~~~~~
/root/esp-idf/components/cxx/cxx_exception_stubs.cpp:24:5: note: 'printf' is defined in header '<cstdio>'; did you forget to '#include <cstdio>'?
/root/esp-idf/components/cxx/cxx_exception_stubs.cpp:6:1:
+#include <cstdio>

/root/esp-idf/components/cxx/cxx_exception_stubs.cpp:24:5:
     printf("%s%s\n", FATAL_EXCEPTION, msg);
     ^~~~~~
/root/esp-idf/components/cxx/cxx_exception_stubs.cpp: In function 'void __cxx_fatal_exception_int(int)':
/root/esp-idf/components/cxx/cxx_exception_stubs.cpp:35:5: error: 'printf' was not declared in this scope
     printf("%s (%d)\n", FATAL_EXCEPTION, i);
     ^~~~~~
/root/esp-idf/components/cxx/cxx_exception_stubs.cpp:35:5: note: 'printf' is defined in header '<cstdio>'; did you forget to '#include <cstdio>'?
make: *** [Makefile:770: build-GENERIC//root/esp-idf/components/cxx/cxx_exception_stubs.o] Error 1
make: *** Waiting for unfinished jobs....
```

amirgon, 2020-06-17:
I don't think the errors you are getting are specific to lv_micropython.
I suggeest you first try your dockerfile with upstream Micropython: instead of cloning lv_micropython.git, clone micropython.git, checkout v1.12, and see if you get the same errors.

There are many unusual things I noticed on your dockerfile:
- You don't always use `--recursive` when updating submodules
- You run esp32  `make` with target `submodules` (??)
- You run esp32  `make`  twice
- You don't specify `BOARD` parameter, as explained in the README.

closed 2020-07-18 by stale[bot]