lvgl/lv_binding_micropython #82: Current STM32 port issues

embeddedt, 2020-06-28:
These were reproduced on STM32F7DISCO, but may also apply to other boards.

* `advanced_demo` crashes if run after a soft reset (the board must be hard reset). I suspect that the display driver isn't deinitializing/reinitializing something correctly.
* Running the following code after `advanced_demo` causes the entire UI to freeze and stop responding to input (though the REPL continues working):

```py
import lvgl as lv
obj = lv.obj()
lv.scr_load(obj)
btn = lv.btn(obj)
```

Entire UI freezes also happened with v6, along with animations ceasing to run.

I don't think we checked whether `advanced_demo` worked with a soft reset on v6. This isn't that major because a hard reset only takes about a second on these STM32 boards, so it's more of an inconvenience.

prolomb, 2020-06-28:
@amirgon @embeddedt 

> advanced_demo crashes if run after a soft reset (the board must be hard reset).

Also on v6 the issue is there. However after soft reset the driver does not provide any error which sounds strange to me. I will try to debug it asap.

> Running the following code after advanced_demo causes the entire UI to freeze and stop responding to input (though the REPL continues working):
import lvgl as lv
obj = lv.obj()
lv.scr_load(obj)
btn = lv.btn(obj)

On v6 it seems to work correctly to me. The screen is purged, the button appears and is responsive.

embeddedt, 2020-06-28:
Are you using the internal flash or an SD card? I'm checking because that changed the behavior in the past.

prolomb, 2020-06-28:
> Are you using the internal flash or an SD card? I'm checking because that changed the behavior in the past.
Both works the same way!

prolomb, 2020-07-02:
@embeddedt 
I tried to debug soft reset issue with a cleaned version of example3.py for V7.
[example3forV7.zip](https://github.com/lvgl/lv_binding_micropython/files/4865236/example3forV7.zip)

After Ctrl-D  I get:
> MPY: sync filesystems
MPY: soft reboot
Traceback (most recent call last):
  File "main.py", line 32, in <module>
MemoryError: memory allocation failed, allocating 3221423020 bytes
MicroPython v1.9.4-2171-gefe39abd6-dirty on 2020-06-29; F7DISC with STM32F746
Type "help()" for more information.

As you can see we have an ubnormal allocation at line 32 that is `scr1 = lv.obj()`
so apparently the issue seems not related to display driver.
Have you evidence about 
> I suspect that the display driver isn't deinitializing/reinitializing something correctly.

?

embeddedt, 2020-07-02:
No; it was just a guess based on the main difference between this and other boards like ESP32 (which presumably work with soft reset).

embeddedt, 2020-07-02:
The behavior you're describing sounds like memory corruption, as an `lv_obj_t` obviously doesn't need 3 gigabytes of memory. Is there a chance that the stack is being overflowed?

One of us will have to debug it to figure out where the bogus value is coming from. I'm currently a bit busy but I can look into it at some point if you'd like.

prolomb, 2020-08-15:
Master branch does not compile now as reported by @TA1DB in https://github.com/lvgl/lv_binding_micropython/pull/72
I just fixed it but need to submit a PR on both bindings and mp.

For a fast fix:

- in  file `lv_micropython/lib/lv_bindings/driver/stm32/STM32F7DISC/modrk043fn48h.c` comment all occurrences of  `nlr_raise()`

- in file `lv_micropython/ports/stm32/Makefile` replace  `SRC_LIB` with `LIB_SRC_C`

embeddedt, 2020-08-15:
I have the board so I'll make the changes and confirm it works.

embeddedt, 2020-08-15:
Your changes appear to work, however, the workaround with commenting out `nlr_raise` means that a lot of errors are now being ignored. I've commented out the `nlr_raise` calls in the interim, but it would be great if you could send a PR to `lv_binding_micropython` that updates the `nlr_raise` calls to use the new syntax and restores exception functionality.

@TA1DB I've tested the build and it appears to be successful now.

TA1DB, 2020-08-15:
> @TA1DB I've tested the build and it appears to be successful now.

Yeah, I'm affirming that it's working on my side too..

prolomb, 2020-08-15:
> Your changes appear to work, however, the workaround with commenting out nlr_raise means that a lot of errors are now being ignored. I've commented out the nlr_raise calls in the interim, but it [...]

@embeddedt 
Sent PR https://github.com/lvgl/lv_binding_micropython/pull/91

embeddedt, 2021-02-20:
The freezing problem still exists as of https://github.com/lvgl/lv_micropython/commit/a66af07d48a7e575712da13c8578b31e89875a37. Interestingly it printed a (fairly useless) message:

```
uncaught exception in Timer(4) interrupt handler
RuntimeError: 
```

amirgon, 2021-02-20:
I wonder if this might be related to [lvstm32.py](https://github.com/lvgl/lv_binding_micropython/blob/master/driver/stm32/lvstm32.py).   
Another option is using [lv_async](https://github.com/lvgl/lv_binding_micropython/blob/55de7683ba6aed3f88e61570e4441db57077b23b/lib/async_utils.py#L17) instead.  
It could be interesting to know if the problems are reproducible without lvstm32.py.

embeddedt, 2021-02-20:
Thanks for that suggestion. The issue does not occur when I modify `example3` to use `lv_async`. Should I push that change to `example3`? I see no reason to use the timer approach when `lv_async` does the same thing and works across all platforms.

amirgon, 2021-02-21:
> Should I push that change to `example3`? I see no reason to use the timer approach when `lv_async` does the same thing and works across all platforms.

Unfortunately, it might not be that simple.
Using `lv_async` requires the whole application to work with uasyncio and to avoid "real" blocking calls. It requires a single event loop and correct usage of `async` and `await` across all code where blocking is needed.  
In many cases Python applications are not designed to work with uasyncio. Sometimes users prefer to avoid it for some reasons (For example, because they use libraries that are incompatible with uasyncio, or simple because they don't understand uasyncio).

I think it's best to make LVGL available in all cases, both when uasyncio is on use and when it's not.  

If the problem never happen with `lv_async`, it gives us a good hint where to look for the problem - [lvstm32.py](https://github.com/lvgl/lv_binding_micropython/blob/master/driver/stm32/lvstm32.py).
It should be ok to call `lv.tick_inc` and `micropython.schedule` from within interrupt context, so I'm not sure what is the problem.  
Maybe we should avoid overflowing the schedule queue and not call `micropython.schedule` if the previous call to `task_handler` did not complete.

embeddedt, 2021-03-29:
> Maybe we should avoid overflowing the schedule queue and not call micropython.schedule if the previous call to task_handler did not complete.

I think you may have found the reason for the odd backtrace. `micropython.schedule` throws a RuntimeError if the queue is full ([reference](https://docs.micropython.org/en/latest/library/micropython.html?highlight=schedule#micropython.schedule)).

I am going to quickly try this tweak and see whether it fixes the issue.

embeddedt, 2021-03-29:
@amirgon Am I doing something wrong with this code?

```py
    def timer_cb(self, t):
        lv.tick_inc(self.delay)
        try:
            micropython.schedule(self.task_handler_ref, 0)
        except RuntimeError:
            pass
```

I would expect this to catch the RuntimeError but I still see the `uncaught exception` message.

The other issue I noted is that we are calling `lv_tick_inc` every 1ms; this is a bit overkill so I reduced the timer frequency to 10ms. Unfortunately the queue still seems to get overrun as the RuntimeError keeps happening.

amirgon, 2021-03-29:
> I would expect this to catch the RuntimeError but I still see the `uncaught exception` message.

Are you sure that the uncaught exception is from your code?
The schedule queue is global. Maybe the queue gets full, and as a result some other code throws this runtime exception?  

Is there a way to get a stack trace? Or connect a debugger?
I know how to do this on ESP32 but I'm not familiar with STM32.

embeddedt, 2021-03-29:
The error does mention that it is from the Timer(4) interrupt handler. I think that is `timer_cb` since lvstm32 is what references Timer 4. You could be right, though.

I'm not aware of an easy way to get a stack trace but connecting a debugger is easy. What would I be looking for, exactly? (Sorry to ask but I'm not very familiar with MicroPython internals.)

amirgon, 2021-03-29:
> I'm not aware of an easy way to get a stack trace but connecting a debugger is easy. What would I be looking for, exactly?

That's a good question. It's not easy debugging Micropython C code.  
You can try stepping through the C code that implements the Timer for stm32 which (as we suspect) throws this exception.
[This line](https://github.com/micropython/micropython/blob/9fef1c0bde2f9642d383bd56aa112447384a84ba/ports/stm32/timer.c#L1611) seems to throw the exception you are seeing:

```c
mp_printf(MICROPY_ERROR_PRINTER, "uncaught exception in Timer(%u) interrupt handler\n", tim->tim_id);
```

But it's not clear how you get there if you are catching exceptions in the callback.

- Is it really this line that throws the exception you are seeing?
- Is the exception printed on [`mp_obj_print_exception`](https://github.com/micropython/micropython/blob/9fef1c0bde2f9642d383bd56aa112447384a84ba/ports/stm32/timer.c#L1615) later on?
- Can you tell which function or python script is running when [`mp_call_function_1`](https://github.com/micropython/micropython/blob/9fef1c0bde2f9642d383bd56aa112447384a84ba/ports/stm32/timer.c#L1604) is called?
- Maybe it's not a "RuntimeException" at all? Did you try to catch all exceptions?

prolomb, 2021-03-30:
@embeddedt 

I suspect the issue could be related to GPU.
In particular the following lines could lead to starvation

https://github.com/lvgl/lv_binding_micropython/blob/e0c783b57e8070dc37e82705cf7913da6c7f2867/driver/stm32/STM32F7DISC/modrk043fn48h.c#L148

https://github.com/lvgl/lv_binding_micropython/blob/e0c783b57e8070dc37e82705cf7913da6c7f2867/driver/stm32/STM32F7DISC/modrk043fn48h.c#L165

Try to disable GPU and see what happens.

embeddedt, 2021-04-11:
Finally got a chance to sit down and take a look at this issue again. I think I must have missed something last time because I could not make the behavior I described above happen again. I even tried scheduling 30 task handler calls inside the `try` handler, but never got a message about an uncaught exception.

What I think happened is that I was editing the raw .py file and didn't realize that lvstm32 now gets frozen at build time, so I was using an old version of the file to "debug". Catching RuntimeError does appear to be a working fix, so I've opened #131.

amirgon, 2021-04-12:
That's great @embeddedt !
I've merged the PR.

Do you think we can close this issue?

closed 2021-04-12 by amirgon

embeddedt, 2021-04-12:
Yes. If we get reports about it happening again, I'll take another look, but I think this is the right solution.

PR #131 (2021-04-12): fix(stm32) skip lv_task_handler if scheduler queue is full
This means that GUI ticks may be skipped on a busy system, but that is better than hanging the GUI forever.

Fixes #82.