lvgl/lv_binding_micropython #83: what does the class btnm.get_map_array return?

WalkerWolf, 2020-06-29:
when i use it in[ LittlevGL/MicroPython simulator](https://sim.lvgl.io/v6/micropython/ports/javascript/bundle_out/index.html), .i hope to get a  list in python,but it just return <class 'BJOB'>. And i can't find this class on mircopython. 
```
        self.btnm_map = ["0","0","0","0","0","0","0","0",""]
        print(self.btnm_map)
        self.button_matrix = lv.btnm(page)
        self.button_matrix.set_size(240,30)
        self.button_matrix.set_map(self.btnm_map)
        map_array = self.button_matrix.get_map_array()
        print(type(map_array))
```
[image]

amirgon, 2020-06-29:
Hi @WalkerWolf !

The lvgl Micropython binding code supports only writing arrays (python lists), but not reading them.
In general, it's usually uncommon for anyone to read array in lvgl, since when you build a UI you usually set values, not read values.
It's worth noting that in C, array does not come encapsulated with its size. Array can be just a pointer. So even if we wanted to convert a C array to a python iterable object, what would be its size? How can we make sure it doesn't overflow?

Regarding `Blob`, it's a python object that encapsulates a C pointer. It can be used when you want access the underlying memory that the pointer points to. So in principle it's possible to access the array data, but only as raw memory and not as a Python object. I don't recommend this.

Could you explain your use case and your need for calling `get_map_array`? Perhaps there is some other way to achieve what you need.

Also, please send further questions to [the forum](https://forum.lvgl.io/c/micropython). We use GitHub issues for reporting bugs or discussing future development, while questions and general discussions go to the forum.

WalkerWolf, 2020-06-30:
> Hi @WalkerWolf !
>
> The lvgl Micropython binding code supports only writing arrays (python lists), but not reading them.
> In general, it's usually uncommon for anyone to read array in lvgl, since when [...]

this array map is end of '\0', it just  string ,and i think we can calc the size by strlen function in anic c.
i want to make something like this. 
[image]. ok , i think it's usually reported this problem whic not implemented by issue,i will handle it by  forum.

amirgon, 2020-06-30:
> this array map is end of '\0', it just string ,and i think we can calc the size by strlen function in anic c.

In this case `\0` specifies the array size, but this is not reflected in C API.
In the general case a C function that reads an array could return an array of any size. The binding code is generic and does not assume different assumptions for specific lvgl C functions.

> i want to make something like this.

When setting the map array, also keep it as a member of your Python object.
lvgl doesn't change it so you can always access it on your Python object instead of reading it by `get_map_array`.

> . ok , i think it's usually reported this problem whic not implemented by issue,i will handle it by forum.

It's not implemented by design and not because of a bug.
I'm closing the issue for now. You are welcome to continue this discussion (or any other discussion) on [the forum](https://forum.lvgl.io/c/micropython).

closed 2020-06-30 by amirgon