lvgl/lv_binding_micropython #86: Update online simulator for v.7

amirgon, 2020-07-06:
Now when Micropython binding is officially available for v.7, we should update the online js simulator with up-to-date lv_micropython.

embeddedt, 2020-07-08:
It's been updated to the latest version available on `master`. Let me know if anything needs changing.

closed 2020-07-08 by embeddedt

amirgon, 2020-07-08:
Works fine. Thanks @embeddedt !

amirgon, 2020-07-10:
@embeddedt actually the simulator is already not up to date (missing `lv.dpx` function).
Maybe worth setting up a GitHub action to regenerate the simulator every time `master` is updated?

embeddedt, 2020-07-11:
It will take me some time to figure out how to do that, as it involves multiple repositories. Which master branch are you thinking of?

It may be easier for me to just have a local script which does the updating. In the meantime I'll update it manually tomorrow.

amirgon, 2020-07-12:
> Which master branch are you thinking of?

@embeddedt Good question.  
Ideally, it would be useful to build and have some tests run when the `master` of either of these is updated:

- lvgl
- lv_binding_micropython
- micropython upstream
- lv_micropython

That would provide several benefits:  

- When tests pass the action could automatically merge the updated lvgl/binding/micropython upstream into lv_micropython to keep it always up to date with all its components.  
- It would flag us when something is not compatible.
- The simulator would always be up to date and aligned with `master`.

embeddedt, 2020-07-12:
It seems that we need a chain of GitHub actions then.

* An action that updates `lv_binding_micropython` when the `master` branch of `lvgl` is updated.
* An action that updates `lv_micropython` when `lv_binding_micropython` is updated.
* An action that updates `lv_micropython` when upstream MicroPython is updated.
* An action that rebuilds the simulator when the master branch of `lv_micropython` is updated.

Does that sound right?

amirgon, 2020-07-12:
> Does that sound right?

Yes, but I wouldn't automatically update any repo before building and running some basic test.
I would rather have a non-up-to-date repo than a broken repo.

embeddedt, 2020-09-21:
@amirgon Do you think the CI you have in place on `lv_micropython` now is reliable enough to update the online simulator with? I have some time over the next few days to set this automation up.

amirgon, 2020-09-21:
> Do you think the CI you have in place on `lv_micropython` now is reliable enough to update the online simulator with?

@embeddedt Yes I think it's stable, I've seen it working correctly for both push and pull requests.  

It's not really running any tests now. It only builds the project and runs something to make sure it doesn't crash in the most basic way.  
Next step would be to think about tests, simulate input device and display and verify some UI sequences, but that would require a bit more work.