lvgl/lv_binding_micropython #87: Linemeter style changes in v7

kdunn926, 2020-07-08:
I want to begin by saying thank you all for this incredibly powerful tool. I'd never have guessed prototyping a UI for an embedded device could be possible in such short iteration cycles (I'm using the Python bindings).

I was previously using a build from master (unhelpfully vague, I realize) and recently pulled down lv_micropython:release/v7 and subsequently ran `git submodule update`. There seems to be various API changes (I'll happily produce docs fixes), along with some functional changes. For example, now when I create a linemeter it looks like this (ignore label):
[image]

It used to (should?) look like this:
[image]

I'm happy to poke around a determine the issue myself but any clues on where to start would be appreciated.

Thanks!

amirgon, 2020-07-09:
Hi @kdunn926 !
Indeed API changed between v6 and v7. Major releases usually contain some API changes.
This question is unrelated to Micropython, the same change would appear on C code that uses LVGL.

We are using GitHub issues mainly to report bugs, for question we have a forum: https://forum.lvgl.io.
**Please post this question on the forum**. Probably the "how-to" category would be appropriate.

closed 2020-07-09 by amirgon

kdunn926, 2020-07-10:
Thanks for the helpful suggestion @amirgon . For posterity, I've created a forum topic here: https://forum.lvgl.io/t/line-meter-material-style-broken-in-v7/2687