lvgl/lv_binding_micropython #88: Example for STM32

kisvegabor, 2020-07-16:
I've played with STM32 port and there were some points which weren't obvious to me. I'm wondering where is the right place to add a minimal init example and some notes.

The init example:
```py
import lvgl as lv
lv.init()

import lvstm32 as st
st.lvstm32()

import rk043fn48h as rk
rk.init()

disp_buf1 = lv.disp_buf_t()
buf1_1 = bytes(480 * 80)
disp_buf1.init(buf1_1, None, len(buf1_1) // 4)
disp_drv = lv.disp_drv_t()
disp_drv.init()
disp_drv.buffer = disp_buf1
disp_drv.flush_cb = rk.flush
disp_drv.hor_res = 480
disp_drv.ver_res = 272
disp_drv.register()

indev_drv = lv.indev_drv_t()
indev_drv.init()
indev_drv.type = lv.INDEV_TYPE.POINTER
indev_drv.read_cb = rk.ts_read
indev_drv.register()

```

amirgon, 2020-07-16:
@kisvegabor why not add it under `examples` directory?  
Perhaps with "stm32" in its name and some comments about the things you found weren't obvious? (what were they btw?)

kisvegabor, 2020-07-16:
>  the things you found weren't obvious? (what were they btw?)

- I figured out that the STM32F746 board is supported but to build `BOARD=STM32F7DISC` needed to be used. (instead of `STM32F7DISC`).
- `MICROPY_PY_LVGL=1` was required to pass to make. The final build command was: 
`make -j8  BOARD=STM32F7DISC MICROPY_PY_LVGL=1 deploy-stlink`
- Baudrate needs to be set to 115200: `picocom /dev/ttyACM0 -b 115200`
- The rk043fn48h driver needs to be used like:
 ```py
import rk043fn48h as rk
rk.init()  # init similarly to lvgl 
```
- `rk.frambuffer` is public but doesn't need to be used as `init` calls it as required. 
- To have tick and task handling running you need to do this:
```py
import lvstm32 as st
st.lvstm32()  #no init function but instantiation
```
- Dump thing: the board needs to be reset after flushed. 

I figured out most of these things by reading the source code.
Probably some of them weren't obvious only because of my inexperience with (Micro)Python but others can be in the same shoes as well.

amirgon, 2020-07-16:
@kisvegabor Yes that's a lot of nuances. 
How about a blog post that goes through all of these?

embeddedt, 2020-07-16:
> `MICROPY_PY_LVGL` was required to pass to make.

It is? When I last tested it, I don't recall needing to pass that flag. `boards/STM32F7DISC/mpconfigboard.mk` appears to set it to `1` already.

kisvegabor, 2020-07-17:
> How about a blog post that goes through all of these?

Sounds good. I'll work on it.

>> MICROPY_PY_LVGL was required to pass to make.

> It was? When I last tested it, I don't recall needing to pass that flag. boards/STM32F7DISC/mpconfigboard.mk appears to set it to 1 already.

I've tried again and it really works without it. Probably I concluded that it's required while using `STM32F746DISC` board.

kisvegabor, 2020-08-10:
I've added the post so this issue can be closed.
https://blog.lvgl.io/2020-08-10/stm32-microptyothon-and-lvgl

closed 2020-08-10 by kisvegabor